MarlinFirmware/Marlin #6590: UBL Not Properly Leveling

houseofbugs, 2017-05-05:
SO I have UBL setup and went through the Phases up to Phase 3. My mesh is saving and loading and looks correct based on my bed. However it is running my head into the bed on multiple sides.

Anyone else having this issue? UBL sound amazing in theory but in practice it is not.

Roxy-3D, 2017-05-05:
> Anyone else having this issue? UBL sound amazing in theory but in practice it is not.

In practice not???   First...  Why did you do a G29 P2 ?    Use G29 P3 to fill the unprobed regions unless you have a good reason not to.      And if it is running the nozzle into the side of the bed....  THAT IS NOT UBL doing that.   You have your bed dimensions and Z_PROBE_OFFSET numbers wrong.

Turn UBL off.   And see if you can get your printer to print a 20^3mm calibration cube at the center of your bed.    Do that....  And then turn on UBL.

houseofbugs, 2017-05-05:
Cube prints correctly and bilinear is working and reports similar offsets for the bed when it probes.

This is for a 220x220 bed. Probe offsets are X47 and Y-19

`#elif ENABLED(AUTO_BED_LEVELING_UBL)

  //===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  #define UBL_MESH_INSET 10          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
  #define UBL_PROBE_PT_1_X 35       // These set the probe locations for when UBL does a 3-Point leveling
  #define UBL_PROBE_PT_1_Y 35      // of the mesh.
  #define UBL_PROBE_PT_2_X 35
  #define UBL_PROBE_PT_2_Y 195
  #define UBL_PROBE_PT_3_X 195
  #define UBL_PROBE_PT_3_Y 110
  //#define UBL_G26_MESH_EDITING    // Enable G26 mesh editing

#elif ENABLED(MESH_BED_LEVELING)`

Roxy-3D, 2017-05-05:
Great!!!!    So...   Please describe exactly what happens when you do a P1.

We will get this figured out and help you bring up a fully tuned system!

houseofbugs, 2017-05-06:
So I did the P1 probe and it mapped out the bed and did have the little dots. I filled those in with a P2 and manual leveling. I have a 47 offset because that is where it is on the carriage.

The mapped region is accurate but when it prints it is either too high or too low.

Roxy-3D, 2017-05-06:
> The mapped region is accurate but when it prints it is either too high or too low.

That is OK...   Presumably you have the mesh saved...   Now you want to do a G26 C P O3.0
That will put a Mesh validation pattern on the bed.   You will see where things are good...  And where things are less than perfect.

Print the pattern...  and then put the Nozzle in the center of any area that is bad.   You do a G29 P4 R O to edit the bad locations.   Just as a rule of thumb...  If you have adhesion...  Don't shift the mesh point by more than .2mm      If you have adheasion...   look at how squished the line is...  and make a determination of whether you want to pull the mesh point up or down and how much.

Edit the bad areas... save the mesh, and do another G26.   It will only take a few iterations to have the mesh perfect.    And from then on...   You just home the printer... and print.    (I like having Z-BabyStep-Double-Click turned on...    I deliberately home a little bit too high (.15mm) and use the Z BabyStepping on every print to dial things in. )   

But I digress...  Each G26 will be noticeably better...   Until there is no reason to keep doing it.

closed 2017-11-01 by thinkyhead