MarlinFirmware/Marlin #6592: build failing with Z_DUAL_STEPPER_DRIVERS enabled

alekseigolikov, 2017-05-05:
in version 1.1.x
having defition for extra Z stepper in DEFAULT_AXIS_STEPS_PER_UNIT, in case of dual Z stepper configuration.
caused by sanitycheck
static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.");

james94jeans2, 2017-05-05:
I think there's no intention to define seperate steps per unit values for dual Z steppers.

Bob-the-Kuhn, 2017-05-05:
All dual steppers have to use the same steps per unit.

It sounds like you tried the following:
#define DEFAULT_AXIS_STEPS_PER_UNIT   {X, Y, Z, Z2, E0}

What Marlin interpreted it as is:
#define DEFAULT_AXIS_STEPS_PER_UNIT   {X, Y, Z, E0, E1}

closed 2017-05-05 by alekseigolikov

ceseuron, 2017-05-24:
This should not be closed without further clarification. In my case, I have dual Z steppers driven off _seperate_ drivers on the Rumba board _and_ I have dual extruders. My configuration.h line for AXIS_STEPS_PER_UNIT is:
`
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 159.57, 160.50, 2117.25, 200, 200 }
`
In the order they are presented above, it is supposed to be X, Y, Z, E0, E1.

This compiled fine on my last version of Marlin but as of 1.1, I get the error mentioned here.

Bob-the-Kuhn, 2017-05-24:
I think I know what's going on.

With DISTINCT_E_FACTORS enabled then it would be
`       #define DEFAULT_AXIS_STEPS_PER_UNIT { 159.57, 160.50, 2117.25, 200, 200 }`

With DISTINCT_E_FACTORS disabled then it would be
`       #define DEFAULT_AXIS_STEPS_PER_UNIT { 159.57, 160.50, 2117.25, 200}`

Sometime after RC8 checks were put in to make sure the number of entries matched the number that Marlin was going to use.

Older versions were OK with DISTINCT_E_FACTORS disabled and having more entries than needed.

ceseuron, 2017-05-24:
I'll check it out as soon as I get home. If we're OK with it, I'll put up a different issue with a solution that we can just close but have available for others to review. Thoughts?