MarlinFirmware/Marlin #6594: USB communication errors with version 1.1.0

timofverberg, 2017-05-05:
After upgrading from version 1.0.0 to the latest version I can't get the USB communication working.
All the configuration settings have been changed to match the known working settings, but I still get tons of errors before Repetier finally shuts down the connection. As soon as I revert back to version 1.0.0 all of the errors are gone, so the problem should be software related.

```
  14:24:54: Attempting to connect to printer
  14:24:54: Connection opened
< 14:24:55: start
< 14:24:55: echo:Marlin 1.1.0
< 14:24:55: echo: Last Updated: 2017-05-04 12:00 | Author: (none, default config)
< 14:24:55: Compiled: May  5 2017
< 14:24:55: echo: Free Memory: 3552  PlannerBufferBytes: 1232
< 14:24:55: echo:Hardcoded Default Settings Loaded
< 14:24:55: echo:  G21 ; Units in mm
< 14:24:55: echo:Filament settings: Disabled
< 14:24:55: echo:  M200 D3.00
< 14:24:55: echo:  M200 D0
< 14:24:55: echo:Steps per unit:
< 14:24:55: echo:  M92 X100.00 Y100.00 Z400.00 E600.00
< 14:24:55: echo:Maximum feedrates (units/s):
< 14:24:55: echo:  M203 X500.00 Y500.00 Z5.00 E500.00
< 14:24:55: echo:Maximum Acceleration (units/s2):
< 14:24:55: echo:  M201 X9000 Y9000 Z100 E10000
< 14:24:55: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
< 14:24:55: echo:  M204 P1000.00 R1000.00 T3000.00
< 14:24:55: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
< 14:24:55: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
< 14:24:55: echo:Home offset:
< 14:24:55: echo:  M206 X0.00 Y0.00 Z0.00
< 14:24:55: echo:Material heatup parameters:
< 14:24:55: echo:  M145 S0 H180 B70 F0
< 14:24:55:   M145 S1 H240 B110 F0
< 14:24:55: echo:PID settings:
< 14:24:55: echo:  M301 P35.65 I4.32 D73.57
< 14:25:00: echo:SD init fail
< 14:25:00: Error:checksum mismatch, Last Line: 5
< 14:25:00: Resend: 6
  14:25:00: Resend: N6 M111 S6 *65
< 14:25:00: FIRMWARE_NAME:Marlin 1.1.0 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
< 14:25:06: echo:Marlin 1.1.0
< 14:25:06: echo: Last Updated: 2017-05-04 12:00 | Author: (none, default config)
< 14:25:06: Compiled: May  5 2017
< 14:25:06: echo: Free Memory: 3552  PlannerBufferBytes: 1232
< 14:25:06: echo:Hardcoded Default Settings Loaded
< 14:25:06: echo:  G21 ; Units in mm
< 14:25:06: echo:Filament settings: Disabled
< 14:25:06: echo:  M200 D3.00
< 14:25:06: echo:  M200 D0
< 14:25:06: echo:Steps per unit:
< 14:25:06: echo:  M92 X100.00 Y100.00 Z400.00 E600.00
< 14:25:06: echo:Maximum feedrates (units/s):
< 14:25:06: echo:  M203 X500.00 Y500.00 Z5.00 E500.00
< 14:25:06: echo:Maximum Acceleration (units/s2):
< 14:25:06: echo:  M201 X9000 Y9000 Z100 E10000
< 14:25:06: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
< 14:25:06: echo:  M204 P1000.00 R1000.00 T3000.00
< 14:25:06: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
< 14:25:06: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
< 14:25:06: echo:Home offset:
< 14:25:06: echo:  M206 X0.00 Y0.00 Z0.00
< 14:25:06: echo:Material heatup parameters:
< 14:25:06: echo:  M145 S0 H180 B70 F0
< 14:25:06:   M145 S1 H240 B110 F0
< 14:25:06: echo:PID settings:
< 14:25:06: echo:  M301 P35.65 I4.32 D73.57
< 14:25:09: Error:Line Number is not Last Line Number+1, Last Line: 0
< 14:25:09: Resend: 1
  14:25:09: Resend: N1 M110 *2
  14:25:09: Resend: N2 M110 *1
  14:25:09: Resend: N3 M115 *5
< 14:25:11: echo:SD init fail
< 14:25:11: Error:Line Number is not Last Line Number+1, Last Line: 0
  14:25:11: Resend: N5 M111 S6 *66
< 14:25:11: Resend: 1
  14:25:11: Resend: N1 M110 *2
  14:25:11: Resend: N2 M110 *1
  14:25:11: Resend: N3 M115 *5
< 14:25:11: Error:Line Number is not Last Line Number+1, Last Line: 0
< 14:25:11: Resend: 1
  14:25:11: Resend: N1 M110 *2
  14:25:11: Resend: N2 M110 *1
  14:25:11: Resend: N3 M115 *5
< 14:25:11: Error:Line Number is not Last Line Number+1, Last Line: 0
< 14:25:11: Resend: 1
  14:25:11: Resend: N1 M110 *2
  14:25:11: Resend: N2 M110 *1
  14:25:11: Resend: N3 M115 *5
< 14:25:11: Error:Line Number is not Last Line Number+1, Last Line: 0
< 14:25:11: Resend: 1
  14:25:11: Resend: N1 M110 *2
  14:25:11: Resend: N2 M110 *1
  14:25:11: Resend: N3 M115 *5
< 14:25:11: Error:Line Number is not Last Line Number+1, Last Line: 0
< 14:25:11: Resend: 1
  14:25:11: Receiving only error messages. Stopped communication.
  14:25:11: Connection closed
< 14:25:11: Error:Line Number is not Last Line Number+1, Last Line: 0
```

equake, 2017-05-05:
@timofverberg take a look here: https://github.com/MarlinFirmware/Marlin/issues/6586

timofverberg, 2017-05-05:
Correct! With the bugfix version everything seems OK.
Looks like it's a waiting game until the next stable release is out :)
Thank you!

closed 2017-05-05 by timofverberg