MarlinFirmware/Marlin #6595: Manual levelling gets stuck at 2nd probe point

paulr4, 2017-05-05:
I'm running the latest version (1.1.x) on a RAMPS 1.4 single extruder setup.

I'm trying the manual bed levelling feature via the LCD, and have the following #defines enabled:

MESH_BED_LEVELING
LCD_BED_LEVELING
PROBE_MANUALLY
EEPROM_SETTINGS

I follow the instructions, but after moving to the second calibration point, the LCD continues to display 'Moving...', and gets stuck. Clicking the control knob produces a beep, but nothing happens.

james94jeans2, 2017-05-05:
Disable PROBE_MANUALLY, then it will work. Had the same issue.

paulr4, 2017-05-05:
Thanks I'll try that.

inahas, 2017-05-08:
Disabling PROBE_MANUALLY allows the procedure to continue, but no mesh data is saved or used. Doing G29 S0 says : 
Send: G29 S0
Recv: Mesh bed leveling has no data.
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0

kylerwinters, 2017-05-08:
Can you post your config?

paulr4, 2017-05-08:
Who?

inahas, 2017-05-08:
This is mine:

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/984279/Configuration.txt)

Fixx, 2017-05-08:
I can confirm Mesh Data is not saved to EEPROM, but works for that session.

inahas, 2017-05-09:
With PROBE_MANUALLY commented out the procedure completes, but no bed leveling is applied. I checked multiple times, no z-axis movement whatsoever.

With PROBE_MANUALLY enabled, the procedure freezes after the hotend moves to the second point. 

I also tried to manually level using G29 S2 but that didn't work either.

I am using the nightly build.

SCiunczyk, 2017-05-26:
HI,

I have exactly this same symptoms as @paulr4. Version 1.1.0 works fine, but 1.1.1 and yesterday Marlin-bugfix-1.1.x and gets stuck.... displaying on LCD Moving... 

What is interesting to me, my LCD interface is in Polish, but this message "Moving... " is in English. Previous version 1.1.0 was in Polish. Maybe this information can help.

geeksville, 2017-06-04:
Anyone know of a work around without disabling PROBE_MANUALLY.  Because I'd like to keep using that ;-)

Roxy-3D, 2017-06-04:
I think we should get the Manual Probe working again for Mesh Bed Leveling!   Does anybody know where on the time line it quit working?   Because that information would help a lot in getting it fixed.

But are you aware the Mesh Bed Leveling is going way soon (5 or 6 months???).    It is a very good technology (in fact...  It *IS* a break through technology!), but its functionality have been absorbed into both the Bi-Linear and UBL bed leveling.

geeksville, 2017-06-04:
Actually: I poked around a bit (though I know virtually nothing about marlin).  Here's some extra info:

Marlin 1.1.1 shows this problem (where I noticed it)

However, the current Marlin-1.1.x branch does not show this problem, unfortunately that branch does not compile if you enable PROBE_MANUALLY.  it fails with the following compiler error for lcd_wait_for_move being not defined in the scope it is used.  

So I fixed this as follows: 
https://github.com/geeksville/Marlin/commit/ed9cb78fde6a197cfc420730ebd5ac7d865b3477

I'd happily send in a PR if you want it, but since the problem isn't fully fixed I'm waiting.  Should I send this in so at least the code can compile again?

Once this fix is in on master it will happily go through the 9 pts and let the user adjust the probe position at each point.  However it then hangs after it prints "Leveling Done!" 

Sad panda.  I'm looking into this hang now, but it is a long time since I futzed with avrs ;-).

Roxy-3D, 2017-06-04:
> I'd happily send in a PR if you want it, but since the problem isn't fully fixed I'm waiting. Should I send this in so at least the code can compile again?

Yes!  Please!   Even if it is a 1/2 fix....   I would appreciate seeing it.

geeksville, 2017-06-04:
Sounds good.  Though when I tried turning on UBL it didn't seem to work with PROBE_MANUALLY yet.

I've learned a bit more: MESH_LEVELING crashes as I described above, however, AUTO_BED_LEVELING_BILINEAR seems to work okay with PROBE_MANUALLY (yay!).  So I'm going to change over to that (good idea?) to get my printer working with Marlin 1.1.1.

I'll also send in a PR to fix at least compilation and gathering the 9 datapoints of MESH_LEVELING+PROBE_MANUALLY.

Roxy-3D, 2017-06-04:
> AUTO_BED_LEVELING_BILINEAR seems to work okay with PROBE_MANUALLY (yay!). So I'm going to change over to that (good idea?) to get my printer working with Marlin 1.1.1.

Yes!   Bi-Linear does everything the original Mesh Leveling did.

> Though when I tried turning on UBL it didn't seem to work with PROBE_MANUALLY yet.

UBL does support Manual Probing.  It is built into UBL as a G29 P2 flavor command.   There are reports from reliable sources that you don't have to enable a real Z-Probe.  But I have never done or tested that.    Like I say...  It is reported to work.    

And in fact....  You can do a G29 P0 to create a mesh of all 0.0000's and jump straight to the G26 Mesh validation pattern.   So...   If you can get that far...   You probably don't even need to configure a Manual Probe.     

The choice is yours how you want to proceed.   I'll try to help you regardless of which path you decide.   But.... You know where I'm focused.

closed 2018-02-07 by thinkyhead