MarlinFirmware/Marlin #6596: UBL mesh is not being saved to EEPROM.

knobik, 2017-05-05:
Im having a very strange problem with with UBL mesh not being saved or loaded from eeprom.

commit im using: 68c9693 (RCBugFix).

commands to generate mesh:

first preheat, then:
```
G28 ; home
G29A P1 M ; auto mesh gen
G29 P3 O ; fill in the missing
G29 S0 ; save mesh to eeprom
```

then hard restart the printer, preheat and:
```
G28 ; home
G29 A ; activate UBL
G26 C P O3.0 ; print validation
```

am i doing something wrong?

The lack of documentation about this topic is killing me. i need to look into ubl.cpp to figure out whats happening and thats no fun :(

bgort, 2017-05-05:
You need to do a M500 after G29 Sx to save the active slot and activate state so that a reload occurs after restart.  Or, if you don't want to do that, you can manually load and activate with G29 Lx A after restarting.

There's a lot of documentation in the code itself, but I agree that we need to work on a more comprehensive 'guide'.  I think @Bob-the-Kuhn started one in an issue?

Roxy-3D, 2017-05-05:
Maybe saving the mesh and state information at the same time is more intuitive for the user?

bgort, 2017-05-05:
Yeah, maybe so.

I assume he has UBL_SAVE_ACTIVE_ON_M500 defined, so he could just issue M500 and it would save automatically?  Not sure if there's a default slot, though.

Roxy-3D, 2017-05-05:
There is not a default slot.  But the G29 S0 would have set the active slot to 0.  If he did an M500, that would be the place it goes at reset to get the mesh.

bgort, 2017-05-05:
Re the default slot:  I was thinking of the case where a new user could just issue an M500, and if there is a modified mesh and no active slot (-1 by default, iirc), 0 would be chosen, and then the mesh would be written there?  I don't know, just thinking out loud...

Roxy-3D, 2017-05-05:
Yeah...   For a new user that would be good.    But in my case...   I have different mesh saved in various slots.    I'm not sure I want it making decisions about what to step on.   

> and if there is a modified mesh and no active slot (-1 by default, iirc), 0 would be chosen, 

But at that point...   maybe -1 is never used?  Maybe we just populate the slot number to 0.   And if they don't specify where to save the mesh...  It goes to 0.

bgort, 2017-05-05:
That's what I'm thinking -- if it's -1 that means it's never been used, so it's probably safe to save into 0.  

Maybe this would work:

On G29 P0 and P1:
  If active slot == -1, set the active slot to 0; otherwise leave alone

Everything else would stay the same.  

Need to think that through, but I think it would make things easier for new UBLers, but wouldn't affect those with saved meshes.

Roxy-3D, 2017-05-05:
I think what you said works...   The whole reason the mesh's are stored at the end of memory is to insulate them from changes in the EEPROM format.    If the EEPROM format changes, we will end up with the active slot being -1 even though there is valid mesh data stored in the EEPROM.  But even in the case...  Doing a G29 L 1 will get the mesh and populate the current slot number with something.  At that point it just becomes a question of whether you want to save the updated slot info with a M500.

bgort, 2017-05-05:
Yes.  I feel like this is pretty safe as long as those with stored meshes don't M500, after G29 P0 or G29 P1, after M502 (or a loss of saved settings due to EEPROM layout change).  I think most people will simply load their mesh after a M502/settings reset, rather than start over, and with an M500 after that, all will be well.

Roxy-3D, 2017-05-05:
> Yes. I feel like this is pretty safe as long as those with stored meshes don't M500...

Even that should be safe...   The thing that will kill them is if they change mesh dimensions.  At that point...  They can load a mesh, but they will be getting scrambled data.

knobik, 2017-05-05:
Thank you will all the discussion about this. E500 saved the mesh to eeprom.

bgort, 2017-05-05:
> Even that should be safe... The thing that will kill them is if they change mesh dimensions. At that point... They can load a mesh, but they will be getting scrambled data.

I'll make these changes later today or tomorrow.

@knobik:  No problem!

Roxy-3D, 2017-05-05:
> I'll make these changes later today or tomorrow.

Do you mean the CRC on each mesh?    If so...  Can you please use this code?
```cpp

uint16_t eeprom_16_bit_CRC;
const char version[4] = EEPROM_VERSION;

// This is a CCITT approved 16-Bit CRC.  It will catch most errors
// that a Checksum will miss.

uint16_t crc16mp( void *data_p, uint16_t count) {
    uint16_t  xx;
    uint8_t   *ptr = (uint8_t *) data_p;

    while (count-- > 0) {
        eeprom_16_bit_CRC=(uint16_t)(eeprom_16_bit_CRC^(uint16_t)(((uint16_t)*ptr++)<<8));
        for (xx=0; xx<8; xx++) {
            if(eeprom_16_bit_CRC & 0x8000) { 
		    eeprom_16_bit_CRC=(uint16_t)((uint16_t)(eeprom_16_bit_CRC<<1)^0x1021); 
	    }
            else  { 
		    eeprom_16_bit_CRC=(uint16_t)(eeprom_16_bit_CRC<<1);                }
            }
        }
    return(eeprom_16_bit_CRC);
}

void _EEPROM_writeData(int &pos, uint8_t* value, uint8_t size) {
  uint8_t c;
  while (size--) {
    eeprom_write_byte((unsigned char*)pos, *value);
    c = eeprom_read_byte((unsigned char*)pos);
    if (c != *value) {
      SERIAL_ECHO_START;
      SERIAL_ECHOLNPGM(MSG_ERR_EEPROM_WRITE);
    }
//  eeprom_16_bit_CRC += c;
    crc16mp( &c, 1);
    pos++;
    value++;
  };
}
void _EEPROM_readData(int &pos, uint8_t* value, uint8_t size) {
  do {
    uint8_t c = eeprom_read_byte((unsigned char*)pos);
    *value = c;
//  eeprom_16_bit_CRC += c;
    crc16mp( &c, 1);
    pos++;
    value++;
  } while (--size);
}
```

bgort, 2017-05-05:
> Do you mean the CRC on each mesh? If so... Can you please use this code?

Yes, I was talking about making changes related to the 'default slot' we just discussed, and was also going to look into using something simple to at least ensure the loaded mesh matches the Configuration.h'd mesh dimensions.  CRC is fine, I'm sure.

Roxy-3D, 2017-05-05:
In the devel_ubl branch...   I had CRC-16 in use instead of a checksum.   That is where I Cut & Pasted that code from.   It works.   The big issue with putting CRC-16 in place for the standard EEPROM contents is it is 2 bytes instead of 1 byte in size and that would shift all the offsets down one.   UNLESS...  we stored the first byte of the CRC-16 at 0xff.   (Right now... the checksum is at 0x100)

bgort, 2017-05-06:
Just implemented the 'default slot' solution and did some cleanups, etc.

> Personally... I think we should move away from the checksum routine to the CRC-16 (or maybe even CRC-32). The checksum is very weak at detecting multiple bit errors.

Yeah, I'll take a look at implementing CRC in place of checksum for the config store, generally, too.  Hopefully will have time for this this morning, or later today.

bgort, 2017-05-06:
I'm ~90% done with implementing the CRC for meshes, the config store, etc.  Turned out that we already had a uint16_t for the config store checksum, so nothing needed to move...

Should be able to finish later today.

Roxy-3D, 2017-05-06:
The only bad thing about doing this is it will cause any modified user settings to be lost.   But we are at a big transition point with the new release code so this is a good time to do it.

bgort, 2017-05-06:
Yeah, I hear you, but I don't see it as that big of a deal, really.  Progress marches on and all of that... And we have G29 S-1, so meshes can be saved and reimported.

I think with a few more iterations we can have a relatively stable 'mesh store' with mesh allocation table, even ... I was initially concerned with the amount of EEPROM space that would take vs. the amount we have available, but once we've gone to 16-bit ints, we'll have plenty for that kind of thing.

Roxy-3D, 2017-05-06:
> I was initially concerned with the amount of EEPROM space that would take vs. the amount we have available, but once we've gone to 16-bit ints, we'll have plenty for that kind of thing.

Yes...  But I would argue we have plenty right now to do that.   Even with a 15x15 mesh...   I get to store 3 of them.    And moving to 16-bit int's would double that.

bgort, 2017-05-06:
Yeah, you're probably right.

Let me think on this, but right now I'm considering adding the MEAT (mesh eeprom allocation table; don't you love acronyms?) at the very end of the eeprom and work backwards, have a version number, etc., and then store all of the relevant details in the table -- things like the grid x/y sizes, insets, etc. etc., along with the mesh type so other leveling systems can eventually use the same storage/retrieval functions and same eeprom space, which would get points out of the main config store and support @thinkyhead's goal to have all meshes be dynamic/variable at runtime.  That way, everything could coexist at runtime...

Roxy-3D, 2017-05-06:
Yeah...   You are thinking the same way I am.    If the Mesh Table was at the end of the EEPROM, it will survive changes in the EEPROM data.   It might be possibly to allow it to store the mesh data for other (non-UBL) systems too.     

Let's think on it....

bgort, 2017-05-06:
> Yeah... You are thinking the same way I am. If the Mesh Table was at the end of the EEPROM, it will survive changes in the EEPROM data. 

Yes...

> It might be possibly to allow it to store the mesh data for other (non-UBL) systems too.

Yeah; I think it's pretty easily doable.  I've already added 'write_mesh' and 'read_mesh' to MarlinSettings and framed it out with the idea in mind that others could use the same functions.  Each leveling system would then just call those functions and their meshes would be stored/retrieved alongside every other type of mesh.

I'll finish up with what I have shortly, then push my changes to my github repo and post a link so you can see what I have so far before we decide (and before we do a PR here).

Agreed that we need to keep thinking, but it's within reach....

bgort, 2017-05-07:
Decided to do a WIP PR so others can more easily see what's being worked on, provide feedback, etc.

closed 2017-11-01 by thinkyhead