MarlinFirmware/Marlin #6597: Problem using MANUAL_BED_LEVELING

batata004, 2017-05-05:
Hi,

I almost destroyed the glass over my printer right now. I actually think I damaged the nozzle but I will only know in the next print.

This is the problem: I am not using the inductive probe cause it stoped working and while the new one is arriving I am still trying to print using only MANUAL_BED_LEVELING.

I commented the BILINEAR auto leveling (that I was using previously) so I could use MANUAL_BED_LEVELING.

The problem is this: when I go to the LCD and I go to PREPARE -> LEVEL BED -> LEVEL BED firt the printer starts the homing procedure for the X and Y. After that it starts homing of Z! The nozzle went down, down, down... and hitted the glass and never stopped! I have no idea why it started wenting down... I SHOULD BE the one that will tell it to go down, right?

I dont have Z endstop (I only have X and Y). The pullups are enabled in my config files (below) which are pretty standard. Why is this happening? Why is the printer lowering and lowering and lowering... shouldnt it wait me to tell it to go down so I can manually "probe" the 9 points using paper method?
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/979825/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/979824/Configuration.txt)

kaunderi, 2017-05-05:
I think you should have Z endstop installed when you use MANUAL_BED_LEVELING. First it homes all axis and then you can manually probe the bed with those 9 points.

batata004, 2017-05-05:
That's what I dont get, if I will have to use a Z endstop something is not right cause when I try to level the bed and move the nozzle BELOW the endstop, how will that work? The printer will probably not allow me to go below the endstop! And if I place the endstop below my bed the nozzle will surely hit the bed before it hits the endstop!

So I think the Z endstop should be not used, otherwise how am I gonna answer these 2 questions above?

Bob-the-Kuhn, 2017-05-05:
The endstops are normally only enabled during G28 (unless you've enabled the ENDSTOPS_ALWAYS_ON_DEFAULT option).  You'll also need to disable MIN_SOFTWARE_ENDSTOPS as that will also keep the nozzle from going below zero.

You'll need to switch back when you get your probe.  When probing the Z_MIN endstop is always checked.

batata004, 2017-05-05:
@Bob-the-Kuhn  my friend, I indeed have ENDSTOPS_ALWAYS_ON_DEFAULT enabled. But that does not make any sense, if I am using MANUAL bed leveling, why would Z END STOP be necessary? And why should I disable this really important safety ENDSTOPS_ALWAYS_ON_DEFAULT ?

I really dont get it, why does the homing tries to home the Z axis if I am explicitly saying in the firmware that I want to use MANUAL BED LEVELING, it means that Z endstop shouldnt be used.

Ayway, are you sure that if I disable ENDSTOPS_ALWAYS_ON_DEFAULT  and MIN_SOFTWARE_ENDSTOPS  then I will be able to use MANUAL BED LEVELING without having the risk to the printer destroy itself?

batata004, 2017-05-05:
What is amazing is that I cant find any instrcution online how to make manual bed leveling... very frustrating...

The z end stop is a huge incognite cause if I have to use a Z end stop where should it be? Higher than the bed or lower? If its higher then when the nozzle tries to lower it will hit the endstop and will disallow me to keep getting lower. **AND EVEN IF I CHANGE the software in order to allow me to get lower THE END STOP WILL BE DRAGED BY BRUTE FORCE when the 2 Z motors spin.**

If I place the end stop lower than the bed it's complete pointless cause the nozzle will already have completely destroyed my glass bed when it touhes the end stop.

I would appreciate some help into this, really, I cant understand why this z endstop has to be used with manual bed leveling.

Bob-the-Kuhn, 2017-05-05:
The Z endstop is for G28. 

Is homing in the + Z direction feasible?  That way you don't need a Z_MIN endstop.  You will need to set Z_MAX_POS so that the Z0 position after G28 is above the bed.  I suggest starting out with a very conservative number, measuring via M114 how much more it needs to go down and using that to arrive at a final figure that leaves Z0 in the 5-15mm range above the bed.

You mentioned two Z motors.  At you running with Z_DUAL_STEPPER_DRIVERS enabled?  If yes then you should seriously consider enabling  Z_DUAL_ENDSTOPS.  That'll automatically correct for any missed steps.  I love it.  Here's my config:
```
  #define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _ZMIN_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  -11  // use M666 command to determine this value
  #endif
```

You could leave ENDSTOPS_ALWAYS_ON_DEFAULT enabled and then use M121 to disable them during the manual probing

batata004, 2017-05-06:
I mentioned 2 z motors cause my printer (as prusa i3) has 2 motors in the Z axis, I have a very cheap 3d printer, it has not dual endstop.

I read everything you said and I am not as smart as you think, I couldnt understand everything, sorry. I ask you this: is possible to make the printer only home the X and Y and not home Z while manual bed leveling?

You see, there is no point in Marlin firmware making the printer hooming Z axis when using manual bed leveling **cause there should never be a z endstop!** That's what I am trying to say, it's not possible to use MANUAL BED LEVELING with z end stop switch: if the endstop is above the bed when the nozzle tries to move closer to the bed it will hit the endstop and drag it down! If the endstop is below the bed it will never be triggered cause first the nozzle will have to break the glass and the bed in order to reach the endstop!

So that's why I dont understand what is going on! It manual bed leveling is set, MARLIN firmware should leave Z homing out of everything, EVERYTHING! It should not use Z endstop nor try to home Z cause it's impossible to use that simultaneously with manual bed leveling. Or I am clearly very dumb and dont understand some very basic concept here - and it could be this case!

Bob-the-Kuhn, 2017-05-06:
You're not dumb.  You're just starting the learning curve is all.  I remember not so fondly my first couple of month or so.  10 months later & I'm still finding stuff I don't understand.

You are correct, the Z_MIN endstop **MUST** be above the bed.  

On a 3D printer the only time you can use the Z_MIN endstop is when homing.  You are correct when you say that the Z_MIN endstop will keep the nozzle from reaching the bed if ENDSTOPS_ALWAYS_ON_DEFAULT is enabled.

The usual practice is to enable endstops only when homing.  G28 turns on endstops, homes the axis and then turns them off (assuming ENDSTOPS_ALWAYS_ON_DEFAULT isn't enabled).

Here's  what G28 does for an axis.  I'll use X as my example.
- If **X_HOME_DIR** is -1  then  move in the - direction until the **X_MIN** endstop is triggered.  Assign that X position the value of **X_MIN_POS**.
- If **X_HOME_DIR** is +1 then move in the + direction until the **X_MAX** endstop is triggered.  Assign that X position the value of **X_MAX_POS**

The options I know of are (until you get your new probe):
- Don't enable ENDSTOPS_ALWAYS_ON_DEFAULT.
- Disable the Z_MIN endstop and manually home the Z axis.  You'll need to manually move the head to a known position ABOVE the bed and then issue the G92 Z0 command.  That takes a long time and it suffered from repeatability problems.  You can alleviate some of the time by always moving to 0,0,0 before turning the printer off or disconnection from it.
- Disable the Z_MIN endstop and use the Z_MAX endstop to home the Z axis.  That's what I'm doing.

----

Your inductive probe stopped working.  By chance was this right after changing to a newer version of Marlin?

If the internal pullup is enabled then your inductive probe won't work.  Marlin defaults to the endstop/probe pullups enabled so we've seen many instances where the inductive/capacitive probe stopped working after switching to a newer Marlin.

batata004, 2017-05-06:
@Bob-the-Kuhn  thanks a lot man, you were always very clear but this time I understood everything! I dont speak well english and this also gives me some trouble to understand many things and I am also not a good learner but I am really glad this 3d printing community is so kind! Said that, I read carefully your last post and I think I understand it all! I think your idea of using the Z endstop at the maximum position is amazing! Is the best alternative so I can keep my marlin firmware most the same.

I was using an inductive probe and then it stopped working (I ordered a new one). You or another good person helped me in anothe thread how to setup the firmware to work with this probe and the person said it very clear that indeed this inductive probe should work with pullup disabled at Z, so I disabled it at Z and enabled only for X and Y and it worked for a while till the probe stopped working. It will probably take a few weeks to arrive so I will have to deal with manual bed leveling for some time.

I confess I liked your 2nd suggestion but I would like you just to tell me what do you mean when you say "disable the Z_MIN". How do I do that? You are saying that I have to remove the Z switch end stop of my printer OR there is some configuration that I can change in marlin firmware in order to disable the Z end stop? I dont care using G92 Z0, I can deal with that :)

Roxy-3D, 2017-05-06:
This probe stuff is difficult...  When I'm forced to bring up a new printer or configuration...  raise the nozzle a bunch and put an empty Kleenex box under it.   And I have my finger on the Reset switch.   I don't trust anything...   These printers will find any excuse to tear themselves up if you let them...    Its almost like they have a death wish.

batata004, 2017-05-06:
@Roxy-3D kkkkk totally agree, I take a lot of care with my printer, treat it well and keep it clean and lubricated... but when I take my eyes for a second to do other stuff it gets mad and almost destroys itself. I really dont know how I still didnt break my glass bed, so many times my nozzle hitted it so hard that I could swear it would be damaged - but not.

Anyway, I am still struggling with manual bed leveling. I cant wait weeks to my probe arrive, I am too anxious! I really cant find anyone in youtube that is using the manual bed leveling so I can see where they putted their Z endstop (if any)...

Bob-the-Kuhn, 2017-05-06:
To disable the Z_MIN sensor comment out the following line:
`#define USE_ZMIN_PLUG`

Roxy-3D, 2017-05-06:
> Anyway, I am still struggling with manual bed leveling. I cant wait weeks to my probe arrive, 

Well...  There is another path available.   UBL needs a probe.   But you don't have to use it.   If you turn on UBL,   do a 
- M502
- M500
- M501
- G29 P0
- G29 A
- G29 S1
- G26 C P O2.5
- G29 P4 R
You can bring up UBL.    You won't have the benefit of the probe automatically most mesh points close to correct.   But you will be able to see from the G26 how far out (up or down) they are.   And you just edit them to get them closer and closer.

You can bring up bed leveling without a probe.

batata004, 2017-05-06:
@Bob-the-Kuhn  I disabled the Z plug but when I tried to compile I got this error:

#error "You must enable USE_ZMIN_PLUG or USE_ZMAX_PLUG."

So I must use a switch and stop even if it will have no use in order to use manual bed leveling? 

@Roxy-3D your suggestion of using UBL without a probe looks another good alternative but I really would like to be able to use mesh points cause I am using a very bad manufactured glass which has a small depression in the middle left :(

Sincerally if any of you already got manual bed leveling without using Z end stop I would appreciate instrucstion. Also, I think that for use manual bed leveling Z end stop should be prohibited cause think about this: if you place your endstop ABOVE the bed and the switch gets hitted, if you try to go lower (if the firmware allows you to keep going lower) you will DRAG/CARRY the endstop downwards...I am pretty sure this is not an intended scenario. And if you place your Z endstop below bed it will never get hitted.

Bob-the-Kuhn, 2017-05-06:
I'm starting to understand.  The Z axis can't physically get by the Z_MIN endstop.  That means you can't set up your system to use the Z_MIN endstop for anything.

I've also thought some more about the manual homing of Z.  I'm pretty sure you won't be able to achieve the repeatability you need to consistently print the first layer

I think your only real option is to change your Z_MIN endstop into a Z_MAX endstop and then home in the + direction.

----

I think that switching to UBL is a great idea.  Once you switch to Z_MAX then you can manually probe the bed.

I like using the **G29 P2 B O** command to do the manual probing.  I'm surprised that @Roxy-3D didn't mention it.

Attached is a rough guide on getting started with UBL.  Just ignore the automated probe section.

This is a Markdown document.   You can also view it with a standard editor but you'll see all the HTML style formatting commands.

[UBL Startup Guide.atom.zip](https://github.com/MarlinFirmware/Marlin/files/981091/UBL.Startup.Guide.atom.zip)

.

batata004, 2017-05-06:
@Bob-the-Kuhn thanks for the guide, I agree with you, UBL with max z endstop is the way to go.

Despite this, I dont need great repeatability. I am not even using eeprom so the values of the mesh will not be loaded later. I just want to make the manual bed level right now and print. I can afford maualling bed leveling everytime I turn on my printer, so repeatability will be no problem cause I will make the manual bed leveling everytime I turn on my printer

THE PROBLEM is that I cant use this feature, there is no way I use this feature because of the Z endstop issue. Sincerally I dont know why does Marlin keep this MANUAL BED LEVELING in the firmware if nobody can use it. I see no point using so many lines of code and memory to load a feature that cant be used because of some sort of "bad programming logic" (respectfullly) that demands Z endstop to be used even when manual bed leveling! 

I think the person/people that programmed this manual bed leveling already tested and used (I am pretty sure of that), I just wonder how that person tested it since you cant put Z end stop at minimum. AND IF YOU ARE mandatory to use Z MAX ENDSTOP so the documentation should make it clear that the manual bed leveling can only be used with Z MAX ENDSTOP! Dont you agree?

Roxy-3D, 2017-05-06:
> I like using the G29 P2 B O command to do the manual probing. I'm surprised that @Roxy-3D didn't mention it.

Well...   That is possible!   But if G26 can print a pattern at all...   It is a lot easier to just see what needs to be done, and edit the points.    The manual probing is sooooooo   time consuming.   Can you imagine doing a manual probe of a 10 x 10 grid?

batata004, 2017-05-06:
I can afford time to spend 10x10 manual probe :) But unfortunatelly the firmware does not allow me spend my time

Roxy-3D, 2017-05-06:
> I can afford time to spend 10x10 manual probe :) But unfortunately the firmware does not allow me spend my time

OK.   Two comments...   UBL has a manual probe function built into it:   G29 P2 R      So with that you could do your 10x10 mesh.   I never use it.   So I just checked and it still works...    If you want to manually probe your mesh, you would do a G29 I999 and then follow it with a G29 P2 O.    It will be painful but you can do it.

But my point was...  If you brought up UBL...  I think the right answer is just zero the mesh and see if you can print a mesh validation pattern.   If you can...   editing the mesh with the  G29 P4 R   isn't going to take very long.

batata004, 2017-05-06:
The UBL looks good, but for a totally beginner and not smart person, those are too much GCODEs for one to take in. That's why I really would like to stick with MANUAL BED LEVELING which is already implemented (in a faulty way I think, respectfully).

If no one finds a way to make MANUAL BED LEVELING work I will surelly follow all your help @Roxy-3D  and from @Bob-the-Kuhn to use UBL till my probe arrives. Your help is amazing and I will surelly use it if no one find a solution to use MANUAL BED LEVELING.

Roxy-3D, 2017-05-06:
Yes..  There are a lot of options.   But doing:
- G29 P0
- G29 A
- G29 S1
- M500
- G26 C P O2.5
- G29 P4 R         
- G29 S1  ; go back to the G26 to see if you are done.

Should get you a working system.   You can ignore all of the extra parameters.

batata004, 2017-05-06:
@Roxy-3D thanks man, I will give it a try when I get home ;) I sent a copy to my mail so I will just flash it in my card and hope it works. In the firmware can I disale mesh bed leveling and manual bed leveling so?

Roxy-3D, 2017-05-06:
I'm not sure I understand the question.   But if you turn on UBL...  and build it into your firmware....   You can control whether it is active or disabled with G29 A and G29 D   (respectively).

batata004, 2017-05-06:
Right, I didnt mean that I would keep bilinear active together cause I cant even compile bilienar + manual bed leveling, it displays errors.

I will turn on UBL and disable manual bed leveling to see what happens. Anyway @Roxy-3D do you know who was the person that implemented manual bed leveling so I could send a PM requesting help and documentation about how to make that work cause it seens useless for Marlin repository to keep such a piece of code that does not work or has nobody in depth with it.

Roxy-3D, 2017-05-06:
It will be in the commit history.  I'm not very skilled at Git.   But for sure the answer is there.    It is possible the manual probing got implemented as a follow on to the original mesh bed leveling.   But I don't know if that is true.

@epatel did the original mesh bed leveling.    (and he did a very good job pioneering that new technology.   If I remember right, as a follow on, he made it so his mesh leveling system could do manual probing.   But I may not have that entirely correct.)

batata004, 2017-05-06:
@Roxy-3D thanks again man! I dont understand git but I will try to track it down :) Thanks again good weekend!

thinkyhead, 2017-05-07:
@batata004 

The current Marlin 1.1.0 is very good and has `LCD_BED_LEVELING` in place of the `MANUAL_BED_LEVELING` setting. Give this current version of Marlin a try and then we'll see where we are at.

batata004, 2017-05-07:
@thinkyhead  THANKS MAN! It looks like it was made a great jog improving this manual bed leveling! EVEN better knowing that BILINEAR is now available with PROBE MANUALLY! AMAZING!

JUST ONE PROBLEM! I spend about 8 days trying to migrate all the configs from my old marlin config (previeous to RC8) and now I will have to do that again!!! SAD REALLY SAD!!

I just want to point out that maany people in this git helped me migrating the file but some people forgot some configurations, others helped me to put those configuration in place but it was never perfect, I took almost 8 days to migrate all the configurations and make my printer work. Also some guys that helped me just tried some "hacks" in the original configuration files to make it compile so I usually lost all the important new comments in the new configuration files. The migration should be using the new version as a base not the old one cause doing this way I lose many important comments.

So now am I have to go to this whole process again? I mean, nobody still developed a tool to migrate configuration files accross versions? I just tried to compile the new marlin with the most recent old version and I got many errors, I tried solving them one by one but they were too many so I gave up. :(

batata004, 2017-05-07:
@thinkyhead I opened another thread to report a bug in the bed leveling manual feature -> https://github.com/MarlinFirmware/Marlin/issues/6624

I tried my best to migrate my old config files to the new format, I used some file comparator tools and did my best. It's not easy to tottaly beginners like me to migrate these files cause I dont understand most of the configurations but anyway, I tried my best and in the new thread I opened I proved the files so you could try. I am using a pretty common REPRAP printer (which uses smart disccount lcd which I think is pretty common too).

Roxy-3D, 2017-05-07:
@batata004  A lot of changes were required in the Configuration.h files to clean up the code.  Most of that is done now.  If you can cross your settings over to the new names and organization, things won't change very fast now.

Have you tried a visual diff program to help you?   NotePad++ with the visual compare add in works very well.  You can pull up your old file and right next to it have the new file.   You can go through and make sure you have everything crossed over.

batata004, 2017-05-07:
@Roxy-3D thanks for the advice! I did it manually and I could generate the new conf file correctly. At least I compiled and it didnt say any error :) Thanks for leting me know NOTPAD++ has this, I didnt now, I had to use a file comparator online.

Good to know the bulk settings will not change for a while cause this settings change make a lot difficult to newbies enter this hobby without too much frustration just to geerate the first print :)

Anyway, it looks like the new firmware has a bug exactly related to MANUAL BED LEVEL, I reported it in another thread and hope someone take a look at it.

closed 2017-11-01 by thinkyhead