MarlinFirmware/Marlin #6599: Homing resets the UBL state

MaxRink, 2017-05-05:
After i home UBL is always set to inactive, no matter whether the state comes form a just executed G29A or from the EEPROM.
`>>>G29 A
SENDING:G29 A
Unified Bed Leveling System activated.
ok P15 B3
>>>M503
SENDING:M503
echo:  G21 ; Units in mm
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X200.00 Y200.00 Z3200.00 E290.00
echo:Maximum feedrates (units/s):
echo:  M203 X250.00 Y250.00 Z4.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E2.50
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S1 Z0.00
UBL is Active
Active Mesh Slot: 3
z_offset: 0.000000
EEPROM can hold 5 meshes.
GRID_MAX_POINTS_X  13
GRID_MAX_POINTS_Y  13
UBL_MESH_MIN_X  (9 + 1)=10
UBL_MESH_MIN_Y  (0 + 1)=1
UBL_MESH_MAX_X  192=192
UBL_MESH_MAX_Y  192=192
MESH_X_DIST  15.17
MESH_Y_DIST  15.92
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P12.36 I1.31 D29.11
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
ok P15 B3
>>>G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
ok P15 B2
>>>M503
SENDING:M503
echo:  G21 ; Units in mm
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X200.00 Y200.00 Z3200.00 E290.00
echo:Maximum feedrates (units/s):
echo:  M203 X250.00 Y250.00 Z4.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E2.50
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S0 Z0.00
UBL is Inactive
Active Mesh Slot: 3
z_offset: 0.000000
EEPROM can hold 5 meshes.
GRID_MAX_POINTS_X  13
GRID_MAX_POINTS_Y  13
UBL_MESH_MIN_X  (9 + 1)=10
UBL_MESH_MIN_Y  (0 + 1)=1
UBL_MESH_MAX_X  192=192
UBL_MESH_MAX_Y  192=192
MESH_X_DIST  15.17
MESH_Y_DIST  15.92
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P12.36 I1.31 D29.11
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
ok P15 B3
`
My build is based on the latest 1.1 bugfix

thinkyhead, 2017-05-05:
The `G28` command is specified to turn off leveling.
Are you able to turn on UBL afterward with `M420 S1`?

MaxRink, 2017-05-05:
yes, i am able to turn UBL back on with m420 S1.

I still consider it somewhat of an issue that g28 -disables- UBL instead of ignoring, as placing a G28 W in the startupcode for a print is kinda good practice

Roxy-3D, 2017-05-05:
> I still consider it somewhat of an issue that g28 -disables- UBL instead of ignoring, as placing a G28 W in the startupcode for a print is kinda good practice

Agreed!    G28's purpose is to home the machine's axis.   It **SHOULD NOT BE ALTERING** things the user has turned on.

The user's want the UBL state preserved across G28's.

Redferne, 2017-05-06:
Thank you! I completely wasted two nights of sleep due to this. G28 is
default printer settings in RepetierHost and more? I'll catch up on some
sleep hoping there's a fix when I wake up next week 😂

Roxy-3D, 2017-05-06:
> Thank you! I completely wasted two nights of sleep due to this. 

Don't lose sleep over this...  Send me a message...

Redferne, 2017-05-06:
Sound good 👋

So please help me understand which GCodes I should send in the
initialization of the printer? I mean after a successful UBL session and
storing to eeprom. I need to do homing on all axis and RepetierHost used
G28 for this. I've been down that avenue and it all ended in tears,
headache and severe sleep deprivation. How do you UBL people perform the
magic?

Roxy-3D, 2017-05-06:
Well...  Until we get this resolved...  The simple answer is any time you do a G28, follow it with a G29 A    That will turn the subsystem back on.

Another answer would be never do a G28.   Just do a G29 A   and if the machine needs to be homed, G29 will do it.   And then it will activate the system.

The 'Right Answer' is we put to code back into G28 that remembers the UBL activation state and restore it to what ever it was at the end of G28.

Roxy-3D, 2017-05-08:
BugFix-v1.1.0 should have your preferred G28 behavior again.

MaxRink, 2017-05-08:
thx, will test lateron

MaxRink, 2017-05-10:
seems2be fixed so

closed 2017-05-10 by MaxRink