MarlinFirmware/Marlin #6601: Support for the MK42 heated bed

MaxRink, 2017-05-05:
As "clones" of prusas MK42 heated bed become more common (you can oder one on indiegogo right now) it might be worth  implemention prusas bedleveling in marlin, as its somewhat difficult for normal users to run prusas fw on anything but a rambo with TR8 leadscrews and the same pulleys as the original mk2, as prusa uses a heavily modified marlin version.

golyalpha, 2017-05-06:
The chip on RAMPs+Mega should be the same as the one on RAMBo, am I correct?

If that is the case, running it on the much cheaper, and in my opinion, much more stable, RAMPs should be as simple as going to Configuration.h and inputting the correct motheboard number, which can be found in Marlin's Boards.h.

In the case of RAMPs EFB (Extruder, Fan, Bed), the number is 43.

MaxRink, 2017-05-06:
The chip is the same, correct.
The issue ist, that prusas fw is stopped down for their exact config and actually modifying it is a lot more complicated than with marlin.
( https://github.com/prusa3d/Prusa-Firmware )

Marlin has no support for the special bed leveling needed for the mk42 tho ( https://ibb.co/jaboWQ ,)

Implementing it shouldn't be that difficult tho, as the needed source code can be found in the prusa repo

golyalpha, 2017-05-06:
Oh, yeah, I see, they have changed it a lot since last time I checked. It was just marlin config file, with some stuff of theirs added on top...

jon012198, 2017-05-07:
i found someone that did modify it to work with ramps https://github.com/slippyr4/Prusa-Firmware/tree/v30.10-RAMPS if you guys what to take a look

MaxRink, 2017-05-10:
still quite a lot behind marlin :/

slippyr4, 2017-06-21:
jon012198 linked to a branch I've killed off; for anyone that lands here and wants prusa firmware for RAMPS & derivatives, go to https://github.com/slippyr4/Prusa-Firmware

MaxRink, 2017-10-19:
so, my MK42 arrived. As much as i like slippyr4s portid like to run the printer on marlin bc using TMC2130s isnt an option with the prusa FW.
What would need to be done inside marlin to support the Bed to a certain degree?

slippyr4, 2017-10-20:
I’m not familiar at all with marlins levelling code since UBL was implemented. 

What needs to happen, conceptually, for MK42 support is to have marlin probe the bed at 9 specific points rather than calculating its own grid locations. Whether the marlin devs want to do that in a generic way and support it or not is a question. I doubt it would be hard to hack it in to your own fork as a one off though. 

The reason for this is that the mk42 has 9 x ~10mm copper pads in its top layer, amongst the heater tracks. Inductive probes will trigger over the whole bed surface but at the proper probe points it triggers much higher because of the increased copper density. If you probe the bed at other locations the trigger point will vary greatly because the amount of metal underneath the probe tip varies significantly. 

Prusa fw also uses these probe points to assess the orthogonality of the build and compute a correction matrix to compensate for out of square builds. I don’t think it’s necessary to implement this; it’s not that hard to build the printer square in the first place.

fiveangle, 2017-10-20:
Easiest is to enable BILINEAR, set the probe limits to match the outside edges of the 9 grid points on the bed, set number of grid points x and y to 3, and set custom homing position to ensure it hits the center point.  Done and done.