MarlinFirmware/Marlin #6602: Not compiling with UBL defined

Truetotosse, 2017-05-05:
With bilinear bed leveling marlin compiles just fine, but with UBL defined i have this error
Marlin.ino.elf section `.text' will not fit in region `text'
ld.exe: region `text' overflowed by 28126 bytes
I have Melzi board 
Does it mean melzi board is too weak to support it?
[configuration.h](https://pastebin.com/PqrLFesJ)

Roxy-3D, 2017-05-05:
Yes...  Right now that is true...   But soon we will be adding options to turn of big parts of the system if you don't need them.   So...  You will be able to configure a minimal UBL system and squeeze it into a Melzi.

Also...  If you switch to a 20x4 LCD instead of a Graphical LCD, you save a pile of room.

michalholes, 2017-07-11:
I have 20x4 LCD and have same error with UBL. Compiling with enabled Bilinear or 3 point leveling is OK, only UBL crash with text overflow error :(

Roxy-3D, 2017-07-11:
Incidentally....   I just cut 7714 bytes of program memory out of the Linear Bed Leveling yesterday using the UBL Least Squares Fit algorithms to replace the QR_Solve library that Linear uses.

fiveangle, 2017-07-11:
UBL requires a lot of PROGMEM currently (approx 50k w/o LCD) so the 128k boards will require a lot of slimming to get UBL up and running (disable SD card, disable any graphics LCD, and maybe text LCD or a few other things).

I have a set of configuration files annotated with some observed sizes for the various options:
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration.h#L829-L833
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration_adv.h

```
//#define AUTO_BED_LEVELING_3POINT // 6366 bytes (w/o LCD)
//#define AUTO_BED_LEVELING_LINEAR // 18386 bytes (w/o LCD)
#define AUTO_BED_LEVELING_BILINEAR // 9804 bytes (w/o LCD)
//#define AUTO_BED_LEVELING_UBL //48592 bytes (w/o LCD)
//#define MESH_BED_LEVELING // 7220 bytes (w/o LCD)
```

By picking and choosing options carefully, you should be able to enable UBL on a 128k if you are willing to give up some features (possibly LCD).

@Roxy-3D has indicated that there are a lot of options that can be made optional within the UBL framework, but they are still ironing out the big features of it now, so it won't be until after the dust settles from the 2.0 re-org before that work can be effectively started.

I've been trying to get BILINEAR working as solid as MBL was for me, but I've had nothing but troubles.  If I pursue it, I'll consider a bring-up guide for the 128k masses out there, since if we can get it as solid as MBL, it should be quite capable, if not as shiny and new as UBL.

Roxy-3D, 2017-07-11:
@fiveangle   Do you have any interest of moving to a 32-bit platform?   Do you have an extra printer to put a 32-bit board on?

michalholes, 2017-07-12:
@Roxy-3D I'll be moving to 32 bits. But now I'm searching mainboard with 3.3v logics, which can be bought without breaking the bank. Raspberry, Banana pi and Due is already at my home.

fiveangle, 2017-07-12:
The Printrbot Simple Metal is my testbed printer at this point.  I have a Prusa MK2 for my general personal printing needs. My whole goal with this recent Marlin project to optimize it for the absolute fastest and as-perfect-as-possible printing of PETG and BRIDGE nylon for a product that I intend to market once the product design is complete.

Once I am confident that all the software stability is there, the final step was to take what I'd gained with the Printrbot experience and build a 20x20x20 CoreXY or Ultimaker gantry system printer (haven't decided yet) that will be able to crank production quality 250 micron layer height prints with 0.5mm or 0.6mm nozzle at 120+mm/s day in and day out.  I expected once I've struck a balance with the software combo to get there, moving to 32-bits to enable scaling speed to the larger bed would be the next step.

At least this was the plan   😛

Bob-the-Kuhn, 2017-07-13:
@michalholes - I'm currently bringing up an Arduino Due with a RAMPS-FD shield and a RAMPS-FD LCD adapter.  So far I've got all the following working:
- RepRap Discount full graphics LCD
- All steppers
- All endstops
- A PT100 amplifier
- A thermocouple input
- The extruder heater
- The extruder fan
- The servo for the BLTouch

I need to iron out some firmware issues with the extruder fan, the Z_MIN_PROBE and the beeper before I'd call it a functional system.

When I power up the system it doesn't automatically start the code.  Pressing the reset button or connecting via the host will start the code.  Don't know if I need to set a jumper, if it's a bootloader issue or a firmware issue.

The reason I went with that combo is it's the only 6 stepper 32 bit package I could find.  I need to replace an AzteegX3pro that went up in smoke.  I'm running dual Y & dual Z steppers so a 5 stepper system wouldn't do.

The strangest thing about the RAMPS-FD is they've inverted the logic for the four power FETs.  

The LCD adapter has a quirk.  The kill switch pin is also shared with one of the steppers.  Cutting a trace and adding a jumper got around that.

I'm using the [Marlin Kimbra](https://github.com/MKFirmware/MK4duo) fork.  It's missing UBL so this is definitely not what I'm going to run long term.

Roxy-3D, 2017-07-13:
> The strangest thing about the RAMPS-FD is they've inverted the logic for the four power FETs.

There was a lot of discussion about this point.   The big concern some people have is if the firmware gets sick and doesn't come up when power is applied, the MOS FET's are on with nothing to control them.

Bob-the-Kuhn, 2017-07-13:
I'm "lucky enough" that I have an older gen Due which has a power up issue.  On my system when it powers up, but the software doesn't start running, the four big FETS are off and the two small ones are on.

That's because the I/O pins are configured as inputs with nominal 100k pullups after a reset.  The four big FETs are turned on by a logic low while the two small ones are turned on by a logic high.

Deneteus, 2017-09-19:
@Roxy-3D - I have been going through old posts about ABL and memory limits and found this post about UBL eating 50k of memory and I want to know if the real solution to this problem with the 3D that use the 'REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER'  is to reduce the size of the code needed to run it. The things were built back in 2013 and alot of us don't need the LCD to have extra languages or Fonts that are too big, or code to run the SD card reader.

I have a Wanhao Duplicator i3 Plus and users are disabling SD support just to get Bilinear bed leveling.
I also have a Creality CR-10 basic with the 1284p/128k and based on all the research I have done I have found that there isn't a lot of documentation on slimming Marlin down which leads to a lot of unnecessary trial and error. In all the posts that I saw your responses to they all either had to do with either disabling features or changing the code in 'qr_solve.h' What is the best way to tackle this problem? Find another graphic LCD that uses the same pinout and less code? Reduce the size of some of the other features.? 

There really should be some kind of warning about how much memory it takes to run some of the LCD's that get posted in the FAQ so that designers can take this into consideration when doing fresh builds.

These are are the references I found to this issue across the net. 

LCD Eats Memory 
https://github.com/MarlinFirmware/Marlin/issues/1645

@DC42 -A FAIK the extra code size from adding the mono graphics display support comes from two places: the display library (which I think is U8glib), and the menu code.
mr. sneezy - Interesting, and disturbing, I just hit the "Sketch too big" error too after adding Repraps Discount Graphic Smart LCD support and I DO already have a Atmega1284P with 128k code space !'

Marlin Sketch Too Big
http://forums.reprap.org/read.php?146,462722

#5216 Help Reduce Marlin Size - https://github.com/MarlinFirmware/Marlin/issues/5216#issuecomment-260207470

V1.1.4 with Anet and Bed Leveling Sensor -> "Sketch too big;" #7240
https://github.com/MarlinFirmware/Marlin/issues/7240

Marlin with Anet A8 #7190
https://github.com/MarlinFirmware/Marlin/issues/7190

fiveangle, 2017-09-20:
@Deneteus - The fact is the 128k boards are a minority makes bending over backwards for them a double-edge sword.  Those with them typically stick with the mfr provided FW for just this reason (Printrbot, Anet, etc).  Your concerns are valid but not easy to address since every user's needs are slightly different... if people in the path of a hurricane are evacuating, one can fit only so much in a car: each person will choose different things that are critical for them.

I myself went so far as to compile for *EVERY* option available in Config*.h so I could see which features consumed the most PROGMEM, and then made my own judgement (I chose to abandon SD all together as I use Octoprint almost exclusively).  I had originally disabled LCD, but with knowledge of which less-essential features consumed the most resources, I was able to add up a bunch of non-critical-to-me items to omit in order to regain LCD functionality.  This is the nature of running comprehensively full-featured software on severely resource-constrained 8-bit micro controllers.

I myself am truly amazed at the level of functionality that the Marlin developers have been able to squeeze into these devices with such modest computing power.  Consider these MCUs were intended to run thermostats, coin-op vending machines, and Christmas tree light flashers, and here we are running entire 3D printers on them: it's a testament to human ingenuity.

But as we near the end of the usefulness for these 8-bit MCUs for this purpose, it will become harder and harder for those on the fringe, not easier.  I hope that my annotated Config*.h files linked above can help, but taken at a high level, no other open source project I can think of has catered to the lowest common denominator as Marlin has.  In fact, it was specifically the advent of Linear Advance that improved my 3 year old Printrbot so greatly that I was compelled to donate a token kickback to the effort.

Hats off the all the devs !

-=dave

closed 2017-10-31 by thinkyhead