MarlinFirmware/Marlin #6603: Z Proximity Probe Problem

leftgangster, 2017-05-06:
I have an issue with my Z proximity probe, when I run the M119 command the probe sometimes shows open and sometimes shows TRIGGERED. When I had the Marlin 1.0.2 it worked fine but now with Marlin 1.1.0 it seems like the z-min pin is "floating" and doesn't trigger correctly. Tried using the original push button end stop, still the same. I am using the GT2560 A+ board.

Redferne, 2017-05-06:
Same problem here. I have also seen this while doing UBL, the proximity
probe gets stuck in TRIGGERED causing UBL to raise Z forever... It's almost
like some spurious line of code happens to reconfigure the Z probe pin to
an output...

Roxy-3D, 2017-05-06:
> the proximity probe gets stuck in TRIGGERED causing UBL to raise Z forever...

Are you using a BL-Touch probe?  If that goes into an 'Error State' it reports triggered until cleared.

More information about your hardware setup would be helpful.

Bob-the-Kuhn, 2017-05-06:
If you're using an inductive/capacitive probe make sure that the endstop pullup resistor is DISABLED.

leftgangster, 2017-05-06:
I have disabled the pullups before posting this issue
`// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif`

leftgangster, 2017-05-06:
I found a working solution for this since I knew the firmware wouldn't have any issue because it's based on older Marlin versions (including 1.0.2 that I had).
My inductive sensor connects the signal pin to 12V when it is OPEN and connects the signal pin to GND when TRIGGERED, so I soldered a diode from the z_min_pin -> signal so that I do not burn my board.
The correct settings for my setup are below.

```
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true

#define ENDSTOPPULLUP_ZMIN
#define ENDSTOPPULLUP_ZMIN_PROBE

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define FIX_MOUNTED_PROBE
```

Bob-the-Kuhn, 2017-05-06:
Glad to hear you're up & running.

This is the first I've heard of an interface besides a resistor divider network.  I see no problems with your approach.

jbronks, 2017-08-10:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1213495/Configuration.txt)
I am having this same problem.  I have an NPN inductive sensor that is about 1 mm above the nozzle.  I loaded Marlin 1.1.4 and now the Z sensor triggers intermittently.  I shut down everything, disconnected Repetier and turned it back on and put a metal object up to it, light comes on but M119 shows open.  Hit the home button and it seems to home correctly but metal object up to sensor and it still shows open.
I didn't have issues with earlier Marlin version, any ideas?  Attached is my config.h.

closed 2018-02-09 by thinkyhead

spprod35, 2020-10-28:
Hi 

i'am reopen this issue.
I have this interface on my ramps 1.4 : https://www.bbfil.fr/carte-d-interfacage-de-capteur-inductif-c2x22858635
With Marlin 1.0 my inductive 6-36v probe works perfectly.
With Marlin 1.1.9 my z_min or z_probe stay TRIGGERED always

Do you have an idea ?

thx for help and sorry for my english

leftgangster, 2020-10-30:
The main config of marlin there is a line where you can have the inverted
value of a sensor (I don't remember exactly the property right now). It
should be Z inverted or something.