MarlinFirmware/Marlin #6605: PID Autotuning causes LCD irresponsiveness and board reset

golyalpha, 2017-05-06:
Whenever I try to run PID autotune on my extruder with the default cycles and temperature 200 C, or 150 C, I get PID Autotuning and then T: in serial, display stops responding, I see the extruder heating LED turn on and then turn off as the board reboots. I have tried to run PID autotune from the LCD Temperature menu, and it causes the same behaviour.

Any ideas onto why is this happening?

knobik, 2017-05-06:
did you try the RCBugFix branch? main release 1.1.0 has a critical error.

golyalpha, 2017-05-06:
I just flashed in RC and it's working so far.

knobik, 2017-05-06:
Just keep in mind, that RC branch is far behind the main release and RCBugFix

golyalpha, 2017-05-06:
Alright. Gonna try RCBugFix then. Can I reuse my config from RC there or not?

bgort, 2017-05-10:
bugfix-1.1.x is the latest and is what you should try, generally.  The issue I believe you're describing has been fixed.

golyalpha, 2017-05-15:
Alright, will flash that as soon as I get back home to see how it works.

closed 2017-05-15 by golyalpha