MarlinFirmware/Marlin #6606: Printer killed when homing with BLTouch from high Z

tcm0116, 2017-05-06:
I'm having an issue where the printer will be killed as soon as the BLTouch probe triggers if I started homing with the Z axis at a high position (>100mm perhaps). I plan on looking into this further, but I just wanted to see if anyone else has experienced this issue or has any ideas as to why it may be occuring.

Currently running the 1.1.0 branch with a pretty vanilla config. I did enable interrupt end stops, but they seem to be working for the X and Y axes, and the Z axis if the starting point is pretty close to the bed.

bgort, 2017-05-06:
I just tried homing from Z=100 and got a kill(), too.  Interrupt endstops are enabled.

jeno007, 2017-05-06:
I just did an autohoming on the LCD from Z=128, and it worked OK. 1.1.0. BLTouch, interrupt endstops RAMPS 1.4.

tcm0116, 2017-05-06:
@bgort are you using a clone?

bgort, 2017-05-06:
No, the latest version of the genuine BLTouch.

tcm0116, 2017-05-06:
@bgort I'm running a clone, so it looks like that's not a factor. Have you enabled BLTOUCH_HEATERS_OFF?

bgort, 2017-05-06:
It is enabled, but not sure that could be be a factor.  Hmm.

Roxy-3D, 2017-05-06:
Wow!   I thought a machine could home an axis within 20 seconds...

mperdue, 2017-05-06:
I've encountered this as well. I didn't make the connection about it only occurring when homing from a high Z because I almost always home from a high Z. Disabling BLTOUCH_HEATERS_OFF has corrected it so I've moved on without that feature for now.

Roxy-3D, 2017-05-06:
> Disabling BLTOUCH_HEATERS_OFF has corrected it so I've moved on without that feature for now. 

You can also go in and edit the time just for yourself....

> I don't know how hard it would be to leave the heaters on for the first tap of homing and disable it for the second but that would help. At least it would help if you're double tapping...

This is a good idea!    That addresses a lot of problems!!!

tcm0116, 2017-05-06:
> Wow! I thought a machine could home an axis within 20 seconds...

The default HOMING_FEEDRATE_Z in Configuration.h is 4 mm/s, which would only allow for 80mm of movement in 20 seconds, which explains the timing issue. Perhaps instead of using a fixed time, the timeout could be computed as a function of HOMING_FEEDRATE_Z and Z_MAX_POS.

Roxy-3D, 2017-05-06:
Yes...  That is also a good idea!   Mostly what I was trying to accomplish with the 20 second time limit was this:   I was worried about the case where the probe might be deployed...  The user turns off the heaters...   And then stows the probe.    It was mostly paranoid type thinking.   But I wanted to make sure the deploy/stow code did not accidentally turn the heaters on.

Maybe that is too paranoid...   But that was the intent of the 20 second time limit.   I was trying to bound the amount of time that a problem could occur.

tcm0116, 2017-05-06:
Does homing always do a double touch? If so, then maybe it'd be better just to turn them off during the second touch since that's a much shorter amount of time. 

As a side note, it sure would be nice if there was some sort of reason message that was displayed when the printer is killed.

Roxy-3D, 2017-05-06:
> Does homing always do a double touch? If so, then maybe it'd be better just to turn them off during the second touch since that's a much shorter amount of time.

I believe that is the case.   The only problem with that idea is we would need to tell the deploy & stow code whether or not it was the first or second touch.    That isn't hard to do...  And maybe instead of saying whether it is the first or second touch...  We have a parameter that says whether or not we want it to be messing with the heaters...     Or stated a little differently...  If we knew it was a homing operation instead of a probe...  we don't mess with the heaters on the first bump.

bgort, 2017-05-06:
> As a side note, it sure would be nice if there was some sort of reason message that was displayed when the printer is killed.

I was thinking the same thing:  I'll split apart the conditional and add some detail to the errors so it's clearer what's happening.

bgort, 2017-05-06:
> Maybe that is too paranoid... But that was the intent of the 20 second time limit. I was trying to bound the amount of time that a problem could occur.

FWIW, I actually don't see a need for the time check, at this point...

mperdue, 2017-05-06:
If the condition can be eliminated but either removing the 20 second timer or not timing the first probe of a home, you won't need to clarify the error.

tcm0116, 2017-05-06:
> you won't need to clarify the error

I was speaking more generally in that a message for any error that killed the printer would be useful for troubleshooting.

bgort, 2017-05-06:
> If the condition can be eliminated but either removing the 20 second timer or not timing the first probe of a home, you won't need to clarify the error.

Yeah.. won't be necessary, but may still be a good idea to indicate why we're kill()-ing.  I think right now it's a rather generic message, and we have plenty of progmem available, generally.

Votes on just removing the time check?

mperdue, 2017-05-06:
Certainly if there are other conditions that kill the printer and return that message it should be changed so there is a more meaningful message delivered. But this one should just go away with any of the proposed fixes.

bgort, 2017-05-06:
Yeah - this same message is used 4-5x, I believe.

Roxy-3D, 2017-05-06:
> Votes on just removing the time check?

I'm in favor of it...   But I'm little nervous about doing it.   

I'm definitely in favor of better Kill() messages!

mperdue, 2017-05-06:
OK, if the heating is stopped while the printer is homing and the printer is also in the process of heating the bed, what happens? Will you have a higher chance of getting an error for bed heating? If yes, then turning it off for the first touch of homing may be the better option. If no, then it probably doesn't matter.

bgort, 2017-05-06:
Hmm.

We could move the 'pausing' into the thermalmanager and have a public function that pauses/unpauses, with the pause temps stored in the class, along with a 'paused' flag.. and we'd call that public function from set_heaters_for_bltouch to pause/unpause.  

Then we could simply disallow changing the temperature when 'paused' so people couldn't mess with things until we unpause after probing.  Easy to give an error message like 'temperature can't be set because heating is paused due to probing; stow your probe first!' if they try.  I think that would address your concern?  And I think this would also make it easier to implement pausing for other probe-types -- we could just change the function to silence_for_probing() or something like that and call it from set_probe_deployed rather than set_bltouch_deployed (if it's configured) ..

Thoughts?

mperdue, 2017-05-06:
Whatever solution you guys come up with works for me. I'm just thinking out loud. :-)

Roxy-3D, 2017-05-06:
If we moved the Heater-Pause functionality into the thermal managers...  I don't even know that we disallow the user from changing the temperature.   Nothing happens until the Pause is cancelled or turns off.   But then the temperature would go to what it is supposed to be at.

I like the idea of moving it because it eliminates the concerns I have.   If the probing is pausing the heaters...    It is really the thermal code that is controlling things...    I really like this idea....

bgort, 2017-05-06:
I think it's a little cleaner, and it makes it easier to layer in shutting off the fans, too, which was one of the other possible sources of EMI.

> I don't even know that we disallow the user from changing the temperature. Nothing happens until the Pause is cancelled or turns off. But then the temperature would go to what it is supposed to be at.

I agree that it isn't strictly necessary, but blocking changing temp while paused would at least minimize confusion... or, or.. maybe this is better:  if someone changes any temp while paused, we assume they no longer care about the pausing, cancel it, set the temp they requested be set, and then leave everything else at zero (where it was set when we paused)?  That's safest, I think...?  I mean, if you manually change temps after deploying the probe, that's on you....

Let me think on implementation, but I believe this will work, address all concerns, and be flexible...

Roxy-3D, 2017-05-06:
Let's think on it.   But right now...   I agree it would be OK to cancel the pause if a new temperature value gets set.

bgort, 2017-05-06:
Just to think out loud, to give a little structure to this, and to give us something clearer to think about, here's where I'm going:

Move pausing to the thermal manager by adding a silence(bool) function:  
- 'silence(true)' would 'pause' the heaters (and fans?)
  - check the in-class 'paused' flag to ensure we aren't already paused
  - store all temps in the class
  - set all active temps to 0
  - set the in-class 'paused' flag
- calling silence(false) would unpause
  - check the 'paused' flag to ensure we aren't already unpaused
  - restore all saved temps
  - clear all of the saved temp variables
  - unset the 'paused' flag

- if any temperature-modifying gcode is issued while paused, we can either:
  1.  cancel the pause, leave all temps at 0, set the new temp:
    a. display a message letting the user know the paused state has been abandoned
    b. unset the paused flag
    c. clear the saved temp variables, and 
    d. set whatever temp was requested to be set; **OR**,
  2. unpause and restore all previous temps, set the new temp:
    a. display a message letting the user know that by setting a temp, they've unpaused and that all temps have been restored
    b. call silence(false), then 
    c. set whatever temp was requested to be set

Will give this some thought.....

tcm0116, 2017-05-06:
> We could move the 'pausing' into the thermalmanager and have a public function that pauses/unpauses, with the pause temps stored in the class, along with a 'paused' flag.. and we'd call that public function from set_heaters_for_bltouch to pause/unpause.

I did something similar in PR #6407. Should be able to do something like to immediately idle the heaters:

```
  HOTEND_LOOP()
    thermalManager.start_heater_idle_timer(e, 0);
```

And then this to re-enable them:

 ```
 HOTEND_LOOP()
    thermalManager.reset_heater_idle_timer(e);
```

It currently only does the heaters and not the bed, but it'd be easy to expand.

bgort, 2017-05-06:
I think it makes sense to consolidate what you were doing there and what we're talking about here, if at all possible...

closed 2017-11-01 by thinkyhead