MarlinFirmware/Marlin #6611: Not able to get the correct Z-position

AlpinaV8, 2017-05-06:
Using K8200 with a servo assisted Z-switch on Merlin 1.1.0 Bugfix

When homing Z, it does it work, and then lifts up to 40. The real distance is 28, and whatever settings I change it will show different Z positions, but always 28 in real. So the head will crash.

My settings are:
#define Z_PROBE_OFFSET_FROM_EXTRUDER -10.8   // Z offset: -below +above  [the nozzle]
#define Z_SAFE_HOMING

Thankfull for any suggestions, and sorry if this don't belongs here.
best regards, Knut

FATALHERMIT, 2017-05-10:
seeing your configuration file could help us figure out your problem ,@AlpinaV8

yoch2015, 2017-07-22:
Probably it will be resolved by running G29 Z 10.8.
I was troubled with the same problem.

Roxy-3D, 2017-07-22:
When you change the setting...   Some times people don't remember to store the new numbers in the EEPROM.   That might be the cause of this.

closed 2018-02-09 by thinkyhead