MarlinFirmware/Marlin #6613: config synchronizer

akaJes, 2017-05-06:
config synchronizer:
goal is:
do not waste a time for a configuration examples reworking after each changing of main file
fast rebuild your personal config to another release
explanation:
1.compare all configs with main
2.create unified config file (only changes or full config) for each config file (enabled,value,comment)
3.recover all configs after changing main file

test results after automation https://github.com/MarlinFirmware/Marlin/compare/bugfix-1.1.x...akaJes:mc-result?expand=1
draft sources with node.js 
https://github.com/MarlinFirmware/Marlin/compare/bugfix-1.1.x...akaJes:mc-tool?expand=1

akaJes, 2017-05-09:
hi @thinkyhead  @Roxy-3D
i have coded some stuff to help work with personal user/examples configurations
installation:
`sudo npm -g i marlin-conf`
then run inside any Marlin git repository
brief file for issue publication:
`mct git 1.1.0-RC7 txt` -> generate *.txt files
migrating from 1.1.0-RC7 to 1.1.0-RC8:
`mct git 1.1.0-RC7 json` -> create json
`mct git 1.1.0-RC8 h` extend config files with json
example configurations:
`mct tree json` -> compare root config .h files with each of examples and create json
`mct tree h` -> build examples .h files after change root config
tree function has issue with delta configurations

https://www.npmjs.com/package/marlin-conf

Roxy-3D, 2017-05-09:
Can you give us a high level description of what is trying to be accomplished?

akaJes, 2017-05-10:
Group operations with configuration files
1. Create a diff text file for issue publication
2. Migrate from one release to another
3. Rebuild example configuration folder based on main files - adding new features, changing comments etc.

akaJes, 2017-05-14:
Hi @Roxy-3D @thinkyhead 
I have build new release #6722 with new features
Web config editor and a help for options
Maybe you have some suggestions for adding other features?

thinkyhead, 2017-05-17:
I'll look at it more closely later today. Need to take a break now…

boelle, 2019-02-18:
@akaJes is this issue still relavant?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-06-22:
think we should close

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle