MarlinFirmware/Marlin #6615: HeatBed advise

Radiu, 2017-05-07:
hi all this isn't a issue, but i have a question to make about heatBeds sense im new into the 3d printing world and there so much information that i end up not being sure in what to get and what is good at reasonable prices.

my question is about heat beds what would be best and what to chose 12v -24v heatBeds or 230V heatBeds?
i was looking into something like 300x300 or maybe 400x300, i was looking into this 2 types, could you give me some advise

i been looking into something like this:
http://www.ebay.com/itm/281763020257?_trksid=p2055119.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT
or
http://www.ebay.com/itm/272114400971?_trksid=p2055119.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT

thanks

Roxy-3D, 2017-05-07:
I think you will get better results with the silicon heat pad.    And you will probably need a SSR (Solid State Relay) to switch the current.   My 400 x 400mm heat runs off of 110 VAC and is switched by a http://www.ebay.com/itm/122443335085  (That would also handle your environment with 220 VAC)   The input connects up to your electronics and you just put a piece of glass on top of it and print.

Even for ABS with the bed at 105 C.   It heats up very quickly.

moth4017, 2017-05-07:
I have just installed a silicon  heat pad 240V AC  on a 6mm Aluminum bed 300mm x300mm heats too 110 deg C  in about 3 minutes , plus with the 6mm  bed the temperature is very constane ..

mperdue, 2017-05-07:
I was recently having trouble getting to 100c with my printer. It has 2 200x300 12v beds combined for 300x400 build area. I was looking at doing a major change going to a 110v silicon heat pad with an SSR to control it. Before doing that I decided to try putting a piece of fiberglass insulation with reflective metal surfaces under the existing bed. Surprisingly, that fixed my problem. I haven't tried to go beyond 110c but it gets to that temperature just fine. Of course, if you don't already have a heated bed I'd go with the mains powered silicon heater and an SSR to control it.

moth4017, 2017-05-07:
The silicon bed is so much faster , and it also takes all the load of your 12V supply ..

mperdue, 2017-05-07:
Which is why I said if he doesn't already have a bed heater to go with the one that runs off mains power (110/220)

Radiu, 2017-05-07:
thats what i tought that silicon bed would be a better choise, i was looking into make a machine with 400x300 work dimensions, with a case of 8mm acrylic all around , for the 400x300 work dimension heatbed i was thinking the case maybe be with 600 x 400, i plan have 2 steper motors nema 17 on right and left side with trapezoidal 8mm and 4 rods 8mm in eatch side, about the heat bed support was thinking maybe  6 mm aluminum for make the base to support the heat bed on top and then a glass on top of the heat bed,i would say that the 2 nema 17 would be able to lift the weight of table bed and glass
i see that some heat beds have springs under need to adjust,i would guess that is used for adjustments and for the nozzles don't hit hard on it has a for when doing calibrations?

any advise how to make a support for the heat bed would be welcome

i would like to thank all for the amazing support that this community gives to people, is just a amazing!!!

closed 2017-06-02 by Radiu