MarlinFirmware/Marlin #6617: Ramps 1.4 reset button

Radiu, 2017-05-07:
hi
i have a ramps 1.4 shield, i have tried use the reset button on the side of ramps1.4 and nothing happens, is this button disable by something or is it a problem with my shield?

thanks

Tannoo, 2017-05-07:
Yes, like an issue with it not being connected. Look for bad solder connections and/or broken trace.

Roxy-3D, 2017-05-07:
Something is defective.  Because that reset switch is hard wired to the processor board's reset.

Radiu, 2017-05-07:
@Roxy-3D @Tannoo   sounds like you correct i have some bad joint or something i was messing with button and if i press very hard sometimes works and sometimes doesn't so must be a bad solder join or a defective button..

thanks alot for the support!!

knobik, 2017-05-08:
@Radiu you should close this issue.

vMeph, 2017-05-08:
@knobik I dont have close issue button under comments, where to do it?

knobik, 2017-05-08:
@Radiu only the issue owner and repository owner can close. The button is next to the green "comment" button.

closed 2017-05-08 by bgort