MarlinFirmware/Marlin #6618: Z axes goes down on small and fast moves.

knobik, 2017-05-07:
Everytime the z axies tries to do a z-hop (generated by slic3r), it pushes the hotend downwards just like the z-axies would invert just for second, other layers print fine, printer homes good, motor is not skipping. This happens on 1.1.0-bugfix branch. This happens on all prints, slic3r config did not change, and the gcode is good.

Video of issue:
https://youtu.be/qh-158TpTEc

Configuration.h
https://gist.github.com/knobik/0a405e13f7eff1b66f49df057c05cfce

Configuration_adv.h
https://gist.github.com/knobik/67b8b213835cea44cea341caed30a6ee

example gcode
[compact_direct_drive_extruder.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/981314/compact_direct_drive_extruder.gcode.txt)

Koshu, 2017-05-07:
Hmm, noticed this, too. Thought it was a new feature(?) of cura, because there was an update recently. Guess not if it happens with slic3r as well.

knobik, 2017-05-07:
@Koshu nice feature, moving the hotend into the bed :)

Koshu, 2017-05-07:
@knobik  :D Well in my case it always happened at corners on the first layer and only pushed it down a tiny bit. So, thought it might be to improve bed adhesion.

knobik, 2017-05-08:
@Roxy-3D any info on this issue?

Roxy-3D, 2017-05-08:
It would be nice to know what  you all have in common.  Do you have bed leveling turned on?   Maybe you are using offsets?   When exactly did you see the change in behavior?   Are you all running Bi-Linear bed correction?

knobik, 2017-05-08:
@Roxy-3D i was using bugfix-v1.1.0 and UBL at the time of making the video, commit bef9791 to be precise.

Roxy-3D, 2017-05-08:
OK...  What is in the video is clearly wrong.   In the morning I'll look through your configuration.h files.   (I'm not saying you have them setup wrong.  I'm looking for clues...)

Roxy-3D, 2017-05-08:
@knobik   You do have these changes in your code, right?  https://github.com/MarlinFirmware/Marlin/pull/6540

- Can you please post your Mesh Map  (G29 O)   ?
- If you pause the print and do a G29 D  (disable),  does this behavior stop?

knobik, 2017-05-09:
Cant give you the mesh right now cuz i invalidated it and switched to bilinear. But i can tell you the bed is tilted by 1.5mm and the mesh spans from -1.1mm to 0.4mm, rest you can interpolate, there were no strange values.

Koshu, 2017-05-12:
Ok this tends to get annoying. Here is a part here it happens quite often:
[image]
You can see all the places where the nozzle moves into the part and leave marks. I think it happens on retracts but don't quote me on that.

knobik, 2017-05-12:
@Roxy-3D any info on this? does #6656 fix this?

Koshu, 2017-05-12:
Wait, shouldn't #6656 only fix things for FWRETRACT? I never enabled it...

knobik, 2017-05-12:
@Koshu if you enable zhop with retract and disable it in slicer then it works with #6656

Koshu, 2017-05-12:
Maybe, but this means this is probably just a workaround for the bug and not the fix?

knobik, 2017-05-12:
Yes, exacly i should mention it.

Koshu, 2017-05-12:
Ok, took me a bit but i was able to isolate the bug. The problem are not retractions alone but every G0/G1 command that does NOT move X and Y. 

To reproduce everything
- Heat up hotend 
- Home
- Activate UBL (Doesn't happen without it)
- G01 Z10 (or any other height)
- G01 X100 Y100 (or any other place)
- G01 Z5 <- You probably won't notice it but it actually moves lower than it should (or later moves are too high?)
- G01 X101 Y101 <- at this point you can notice a rather large movement in Z even so it is only a small X/Y movement. I guess at this point it is at the correct UBL-corrected-height or something.
- G01 E-5 <- It moves down, back to the Z-position after G01 Z10

@Roxy-3D Maybe ubl_line_to_destination_cartesian needs another case at the beginning for movements without X or Y?

Koshu, 2017-05-12:
Ok, added some debug outputs to UBL again. The UBL functions aren't even called. Looks, to me like movements without X or Y are handled somewhere else and as if that move is done without(!) mesh-correction.

Koshu, 2017-05-12:
Looks like I have a fix (Doing some testing right now). The problem is in Marlin_main.cpp in function prepare_move_to_destination_cartesian. Thats the one that filters out movements without X or Y and passes it to the non-UBL move function. By commenting out that check it seems to be fixed, however I have no idea why that was there and what other implications  disabling it has.

```
  inline bool prepare_move_to_destination_cartesian() {
    // Do not use feedrate_percentage for E or Z only moves
    // if (current_position[X_AXIS] == destination[X_AXIS] && current_position[Y_AXIS] == destination[Y_AXIS]) {
    //   line_to_destination();
    // }
    // else {
      #if ENABLED(MESH_BED_LEVELING)
        if (mbl.active()) {
          mesh_line_to_destination(MMS_SCALED(feedrate_mm_s));
          return true;
        }
        else
      #elif ENABLED(AUTO_BED_LEVELING_UBL)
        if (ubl.state.active) {
          ubl_line_to_destination_cartesian(MMS_SCALED(feedrate_mm_s), active_extruder);
          return true;
        }
        else
      #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)
        if (planner.abl_enabled) {
          bilinear_line_to_destination(MMS_SCALED(feedrate_mm_s));
          return true;
        }
        else
      #endif
          line_to_destination(MMS_SCALED(feedrate_mm_s));
    // }
    return false;
  }
```

Koshu, 2017-05-12:
P.S: This might also explain why I seemed to have trouble finding the correct z_offset because when I measured it I did it without UBL (only Z movement) while later on UBL got activated.

Koshu, 2017-05-12:
Jepp, fixed the problem. The part from before:

[image]

Still needs fine tuning but the marks from before a gone. However, it seems to have a side effect. That is homing with UBL on produces different results than homing with UBL off. It does some funky movements. I guess for homeing UBl should be automatically disabled and re enabled again afterwards (If you don't know which z-position you are on, there is nothing to correct anyway). But it is also possible that my probe has some bug. So someone else want to confirm?

@Roxy-3D If you have time, can you have a look at this?

Roxy-3D, 2017-05-12:
Yes...  I'll look at it.    But right now G28 does turn off bed leveling to do its work.  And if UBL was active when the G28 was started, it will turn it back on.   However, that later behavior was not in the code for a short period of time.   If you have firmware from that short period of time, you probably want to grab the current bugfix_v1.1.0

Koshu, 2017-05-12:
then my probe might have some problems. Should I make pull request for the bug above or is there a reason that non x/y movements are handled differently? (Or some better fix)

Roxy-3D, 2017-05-12:
You only make a Pull Request if you have an updated version of the code that you want to put back into the main repository.   Z & E moves are handled differently in a number of places in the code base.

Koshu, 2017-05-12:
I know just meant if there is something i'm overlooking or if it is just the few comments to fix that bug.

bgort, 2017-05-13:
I wonder if this explains the weirdness I was seeing while trying to sort out what was going on with my BLTouch.  Seems like it could, potentially.

Koshu, 2017-05-14:
Should be fixed by #6716.

closed 2018-02-06 by thinkyhead