MarlinFirmware/Marlin #6619: AUTO_BED_LEVELING_LINEAR diff AUTO_BED_LEVELING_BILINEAR

trivalik, 2017-05-07:
What is the different between both, because from the description standpoint seems bilinear more complex than linear. But in compile size is linear more than 9000 bytes bigger. Any idea why?

Roxy-3D, 2017-05-07:
Linear assumes the print bed is perfectly flat, but slightly tilted.   It does either a 3-point probe to define that tilted plane or it does a grid of probes and uses a Least Squares Fit to determine the plane.

Bi-Linear assumes the bed is not flat.  It is basically a Mesh Bed Leveling system.  Each mesh point can have a different height that describes its Z position.  In side any given mesh cell, it does a bi-linear interpolation of the 4 corner mesh points to determine the correction applied to the nozzle.

trivalik, 2017-05-07:
Yeah, I get that. I was guessing that just calc proportion of the measured Z values is same. And as you said, the bed is flat and only tilted, I don't understand why to use an algorithm for curves.
Least Squares Fit is maybe more complex than calculate proportion of 4 mesh points. I am not sure how this will be more exact.

Roxy-3D, 2017-05-07:
The mesh leveling works better because it has reference points very close to where ever the nozzle is.  That lets the nozzle track the bed better than making the assumption the bed is flat but a little bit tilted.

Least Squares is used in the Grid based or Linear auto leveling because when you have 9 or 16 or 25 data points...  There is no solution to the set of linear equations.   So instead...  You are looking for the 'best solution' or the one with the least error.  The math is doing different things.

akaJes, 2017-05-10:
I think that linear is less optimized, rarely used, not precision well as bilinear because in real world not exist perfect flat surfaces

closed 2018-02-06 by thinkyhead