MarlinFirmware/Marlin #6620: G29 P4 not moving the z axes

knobik, 2017-05-07:
When mesh is build and i want to fine tune it, i do a G29 P4 command and "Mesh editor: -0.240" is shown on LCD, z-axis moves up couple of milimeters but after i move the knob, the z-axis does not move.

Tannoo, 2017-05-07:
This was taken from this by @Bob-the-Kuhn: [UBL.startup.guide.zip](https://github.com/MarlinFirmware/Marlin/files/981475/UBL.startup.guide.zip)

>To edit a single point move the nozzle close to the point that needs adjustment. Issue a G29 P4 O . The head will move to nearest point. Use the encoder wheel to change the value.  The nozzle will NOT change height.

knobik, 2017-05-07:
It would be nice to use move it cuz i could use a shim to perfectly level it.

Roxy-3D, 2017-05-07:
I'll look at putting the move up or down back in.

Bob-the-Kuhn, 2017-05-07:
I think we're better off as is.

The shim method isn't as precise/repeatable as needed for this portion of UBL mesh creation. 

When I did my mesh I was doing 0.1, 0.05 & 0.02 mm adjustments based on the G26 results.  You're just not going to get the 0.05 & 0.02 resolution with a shim.

closed 2017-11-01 by thinkyhead