MarlinFirmware/Marlin #6624: Bug with the new manual probe

batata004, 2017-05-07:
Hi,

@thinkyhead explained in another thread very well how to enable PROBE MANUAL LEVELING. I just did it, here are my conf files.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/981612/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/981613/Configuration.txt)

The first problem that I am facing, is that with this configs files I cannot even PREPARE -> MOVE AXIS -> MOVE Z AXIS -> I can move Z axis below the current state.

I am not using Z endstop switch cause  intend to use MANUAL probe leveling so Z switch will have no use.

Also, when I try the BED LEVELING feature it starts great, it homes X and Y, and then LCD shows MOVING... but nothing happens... maybe it's trying to move down but firmware does not alow it for some reason.

Could you please investigate?

batata004, 2017-05-07:
I discovered one thing: if I set Z_MIN_PROBE_ENDSTOP_INVERTING and Z_MIN_ENDSTOP_INVERTING to FALSE I can finally move down the nozzle and when I click BED LEVEL (lcd) it homes X, Y and then for some freaking bug reason it tries to HOME Z which is nonsense cause I dont have a Z endstop and **I ALSO SHOULDNT HAVE ONE if I am gonna use MANUAL BED LEVEL! Right? There is complete no point in using Z switch endstop when using manual bed leveling!**

Bob-the-Kuhn, 2017-05-07:
You'll need a reliable & accurate method of setting the Z axis position to a known value unless you're going to do the manual leveling every time the controller is reset.  The endstop is the only way I know of doing that without a probe.

I think you're only practical option is to use a Z_MAX endstop and home in the + direction.

batata004, 2017-05-07:
@Bob-the-Kuhn  you are damm right :) But I will not install a Z MAX endstop switch, I prefer doing the manual bed leveling everytime, I usually only restart my printer once in every 3 days and my arduino mega has some problem with eeprom so I cant rely in it.

Anyway, if Z MAX endstop switch is mandatory, I think there should be some comments in the firmware telling user to not use Z MIN endstop if he wants to use manual bed leveling. Anyway, I think it should be not mandatory using endstop switch, if the user wants so he can use Z MAX endstop, if he does not want so he will have to deal doing manual bed leveling everytime starts the print.

batata004, 2017-05-07:
@Bob-the-Kuhn  I tried you suggestion of using Z_MAX (commenting Z_MIN) and I got this error prior comipling:

#error "Enable USE_ZMIN_PLUG when homing Z to MIN."

So, besides the nonsnse when using Z MIN ENDSTOP to manual bed leveling, if I try to use Z MAX ENDSTOP the compiler does not allow me to compile! So, no alternative to use MANUAL BED LEVELING so far, cause neither end stop (min/max) will work. Sad.

Bob-the-Kuhn, 2017-05-07:
You also need to change **Z_HOME_DIR** to positive 1.  Now G28 will move the Z axis upwards toward the Z_MAX endstop.

The value for **Z_MAX_POS** is now important.    G28 sets the Z location to this value.  You'll need to play with that to get Z 0 close to the bed.

batata004, 2017-05-07:
Thanks man!! I installed and removed so many times the only endstop that I had laying around (the z) and it does not work anymore!!! I am with a really bad luck! Now that you teached me how to compile succesfully the firmware I cant use it because I have no endstop left! Baad luck... so I will have to wait someone to teach me how to use this manual bed level without the endstop... i cant believe this, I think my solder iron was too hot and deformed a lot the endstop ;(

kyberkiller, 2017-05-10:
Hi,

I started with 1.1.0 RC8 few days ago, struggled with the same thing and found that the mesh bed leveling works nice. I have no LCD yet, so this comment is for G-gode terminal only.

Here are my test parameters: (using 2 x 2 points isn't too clever, I have not tested with only one)
#define MESH_BED_LEVELING    // Enable mesh bed leveling.
#if ENABLED(MESH_BED_LEVELING)
  #define MESH_INSET 20        // Mesh inset margin on print area
  #define MESH_NUM_X_POINTS 2  // Don't use more than 7 points per axis, implementation limited.
  #define MESH_NUM_Y_POINTS 2
  #define MESH_HOME_SEARCH_Z 4  // Z after Home, bed somewhere below but above 0.0.
//#define MESH_G28_REST_ORIGIN // After homing all axes ('G28' or 'G28 XYZ') rest at origin [0,0,0]

  //#define MANUAL_BED_LEVELING  // Add display menu option for bed leveling.

  #if ENABLED(MANUAL_BED_LEVELING)
    #define MBL_Z_STEP 0.025  // Step size while manually probing Z axis.
  #endif  // MANUAL_BED_LEVELING

  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT
#endif  // MESH_BED_LEVELING

You can also use #define MANUAL_Z_HOME_POS to have Z homed somewhere else than top (yes, it goes there in any case).

http://marlinfw.org/docs/gcode/G029-mbl.html

Commment   #define MESH_HOME_SEARCH_Z 4  // Z after Home, bed somewhere below but above 0.0." here is not true, the MESH_HOME_SEARCH_Z is used in G29 procedure.

batata004, 2017-05-10:
nice man!!! I was not aware of MANUAL_Z_HOME_POS , I am surprised notobyd mentioned it! Using it worked perfectly! Great! This should be documented in the firmware or itself should auto eneabled MANUAL_Z_HOME_POS  if manual_probe is enabled! Problem solvrd!

closed 2017-05-10 by batata004

Roxy-3D, 2017-05-10:
The code base is huge...  And was developed by hundreds of people...  nobody knows the whole code base.