MarlinFirmware/Marlin #6627: Using USE_CONTROLLER_FAN

batata004, 2017-05-07:
Hi,

Sometime ago I was looking for something exactly as USE_CONTROLLER_FAN and now it exists :) The problem is that the documentation does not say much about it. In the code I see this:

//#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN FAN1_PIN  // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif

I am usins RAMPS 1.4 with arduino MEGA. Which pin should I use? Even if I know the pin I think there is a problem, which mosfet will this feature use since all mosfets of my RAMPS are already being used?

cbusillo, 2017-05-07:
You will need to add an external MOSFET and use a digital pin to drive the fan.  Assuming of course you are using all the available internal MOSFETs for cooling fans, bed, and hot ends.

Might I suggest using a temperature controlled PWM fan controller instead?  That way you can set it to ramp the fan up higher if the temperatures on the drivers spike.

batata004, 2017-05-07:
@cbusillo  temperature controoler would be nice indeed! Good suggestion :)

closed 2017-05-07 by batata004

cbusillo, 2017-05-07:
Might I also suggest an adjustable buck converter before the PWM controller.  That way you can tune the top speed of the fan and let the fan controller ramp the fan speed.  

If you are looking for a silent printer when not printing you can also add a MOSFET and use the RAMPS to turn the extruder fans on when the temp is above 50c.

batata004, 2017-05-07:
@cbusillo great idea to limit the top speed and still allowing arduino to ramp up the speed till a maximum, I realy liked it! I will get a buck converter tomorrow in the electronics store! 

About your second suggestion I would like to use the spare mosfet of the board to turn off the drivers fan and hotend fan... I just still dont know how to do that since I am not using a dual extruder I have an available mosfet already in place in my ramps.

cbusillo, 2017-05-07:
You would need to change the pin in your config file and set E0_AUTO_FAN_PIN.

batata004, 2017-05-07:
thanks man, appreicate! I will test it when I get hoem ;)

cbusillo, 2017-05-07:
You will need to figure out what pin your second hot end is powered from.  Its in your pins file.  Use that PIN number in the configuration_adv to set the pin of the controller or extruder fan.  You might be able to use the same bin on both.