MarlinFirmware/Marlin #6628: Compile error on latest 1.1.0 bugfix

DaveAhrendt, 2017-05-07:
Fails to compile with 1.1.0bugfix from today.  Yesterday's snapshot builds OK.

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Users\Dave\Documents\Arduino\libraries\U8glib\src" "C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp" -o "C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp.o"
In file included from C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\Marlin.h:42:0,

                 from C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:166:

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':

configuration_store.cpp:1546: error: invalid initialization of reference of type 'float&' from expression of type 'int'

         SERIAL_ECHOPAIR(" H", TEMP_UNIT(lcd_preheat_hotend_temp[i]));

                                                                  ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\serial.h:50:76: note: in definition of macro 'SERIAL_PROTOCOLPAIR'

 #define SERIAL_PROTOCOLPAIR(name, value)    (serial_echopair_P(PSTR(name),(value)))

                                                                            ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:1546:9: note: in expansion of macro 'SERIAL_ECHOPAIR'

         SERIAL_ECHOPAIR(" H", TEMP_UNIT(lcd_preheat_hotend_temp[i]));

         ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:1546:31: note: in expansion of macro 'TEMP_UNIT'

         SERIAL_ECHOPAIR(" H", TEMP_UNIT(lcd_preheat_hotend_temp[i]));

                               ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:1260:22: note: in passing argument 1 of 'float temp_abs(float&)'

         extern float temp_abs(float &f);

                      ^

In file included from C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\Marlin.h:42:0,

                 from C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:166:

configuration_store.cpp:1547: error: invalid initialization of reference of type 'float&' from expression of type 'int'

         SERIAL_ECHOPAIR(" B", TEMP_UNIT(lcd_preheat_bed_temp[i]));

                                                               ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\serial.h:50:76: note: in definition of macro 'SERIAL_PROTOCOLPAIR'

 #define SERIAL_PROTOCOLPAIR(name, value)    (serial_echopair_P(PSTR(name),(value)))

                                                                            ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:1547:9: note: in expansion of macro 'SERIAL_ECHOPAIR'

         SERIAL_ECHOPAIR(" B", TEMP_UNIT(lcd_preheat_bed_temp[i]));

         ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:1547:31: note: in expansion of macro 'TEMP_UNIT'

         SERIAL_ECHOPAIR(" B", TEMP_UNIT(lcd_preheat_bed_temp[i]));

                               ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_557039\sketch\configuration_store.cpp:1260:22: note: in passing argument 1 of 'float temp_abs(float&)'

         extern float temp_abs(float &f);

                      ^

Using library U8glib at version 1.19.1 in folder: C:\Users\Dave\Documents\Arduino\libraries\U8glib 
exit status 1
invalid initialization of reference of type 'float&' from expression of type 'int'

DaveAhrendt, 2017-05-07:
Forgot to add that you need #define TEMPERATURE_UNITS_SUPPORT to trigger the error.  With this define commented out everything compiles.

bobc, 2017-05-07:
Temperature variables are ints, but function temp_abs() is expecting a float. Not sure when that happened.

A "quick fix" might be to change 

```float temp_abs(const float &c)```
to 
```float temp_abs(const float c)```

in Marlin_main.cpp and configuration_store.cpp

thinkyhead, 2017-05-08:
It compiles ok for me in PlatformIO… let me try Arduino IDE…

thinkyhead, 2017-05-08:
The const was missing from the extern declaration. Will patch shortly.

closed 2017-05-08 by thinkyhead