MarlinFirmware/Marlin #6630: programming question - what function prepares the X and Y values for printing?

mperdue, 2017-05-07:
I'm wanting to do a little playing around with the code. I'd like to look at what might be required to do some quick and dirty parallelogram correction. I'm not real familiar with the layout of the code so if someone can point me in the right direction I'd appreciate it. Somewhere in the code there must be a function that decides what the X and Y values are prior to sending them to the printer in G-code. What function or functions are involved in doing that and what files will i find them in?

Roxy-3D, 2017-05-07:
Start at the top level...  The printer is told to make simple moves with G1 [X#] [Y#] [Z#] [E#]     But then it goes down different paths depending on whether it is Cartesian, Delta, Scara, etc.   And those sub-paths change depending on what auto bed leveling is turned on....    In Marlin_main.cpp  you can find the G1 command by looking for:
```cpp
inline void gcode_G0_G1(
  #if IS_SCARA
    bool fast_move=false
  #endif
) {
```
And as you can see...   Even that has conditional code to handle Scara's....

mperdue, 2017-05-07:
Thanks, I guess I should have figured the G1 would be a good starting point. I haven't done any coding since I retired 3 years ago so I have an excuse. :-)

Roxy-3D, 2017-05-07:
Yeah well....   As you start tracing down into the code...   Your brain is going to explode...

closed 2017-05-07 by mperdue

thinkyhead, 2017-05-08:
For parallelogram correction, you'll be getting into the leveling system and the interface between the high-level movement code and the `Planner` class. The `Stepper` and `Planner` objects maintain the real physical coordinates of the steppers. The "idealized" cartesian coordinates are maintained in the "`current_position`" variable by code in `Marlin_main.cpp`.

When a move is sent to the planner in the form of an XYZ target, the target XYZ is first adjusted by bed leveling, then kinematically converted to stepper (ABC) step positions. Sometimes the adjustment can be done as part of kinematic conversion. Sometimes kinematic conversion means splitting lines into smaller segments. For mesh-based leveling, the more segments there are, the better the nozzle aligns to the curvature of the bed. Currently we break up moves into small segments for arcs, delta, and scara moves, and with grid-based mesh leveling enabled moves are split on grid-line boundaries.

The question is how will skew compensation interact with Z height adjustment? For bilinear/ubl/mesh leveling, Z height needs to adjust based on the measured bed point as it aligns to the nozzle. So any skew compensation that changes the position of X or Y during the print should be applied one step ahead of the Z bed leveling adjustment. And I believe it will apply to planar leveling the same way as bilinear/mesh leveling.

mperdue, 2017-05-08:
I started looking at all this last night and determined that it's probably not the best place for me to be tinkering with. Plan be is a redesign of my XY carriage to make it more adjustable. :-)