MarlinFirmware/Marlin #6636: not moveing after g29

jon012198, 2017-05-08:
it just sits there after in the back right corner and just dumps filament out after it is heated up  dose not move were it is suppose to print  video slow down the video and it is at the very end
[petgMSV_20mm_hollow_cube_20170508013610.zip](https://github.com/MarlinFirmware/Marlin/files/982160/petgMSV_20mm_hollow_cube_20170508013610.zip)

jon012198, 2017-05-08:
using rc bug fix breach 1.1.x

bgort, 2017-05-08:
Is this with a G26?  If so, it's likely waiting for you to click the encoder button to stop the priming; alternately you can specify a prime filament length by adding Px (x in mm) to your G26.

jon012198, 2017-05-08:
nope  it is not with here is my gcode file  if you want to take a look at it 
[petgMSV_20mm_hollow_cube.zip](https://github.com/MarlinFirmware/Marlin/files/982177/petgMSV_20mm_hollow_cube.zip)

jon012198, 2017-05-08:
AUTO_BED_LEVELING_BILINEAR

jon012198, 2017-05-08:
here is my config of marlin if you need this 
[Marlin-bugfix-1.1.x dup i3.zip](https://github.com/MarlinFirmware/Marlin/files/982197/Marlin-bugfix-1.1.x.dup.i3.zip)

jon012198, 2017-05-09:
so i re flash marlin i did change the esteps in eerom  and i think it messed something else up  but a reflash fixed it

jon012198, 2017-05-09:
worked once then it failed just sat in the corner  and spit out filament

jon012198, 2017-05-12:
any help or am i out of luck

bgort, 2017-05-12:
Do you have a working version of Marlin, or a working version of any other firmware?

Want to rule out a hardware problem....

jon012198, 2017-05-12:
yes i do have one 
 it is rc4 version of marlin

bgort, 2017-05-12:
Did you redo your configuration files entirely?  

A lot has changed and it's best to set up the new files from scratch while comparing to the old.

jon012198, 2017-05-12:
i pulled them up side by side and  look at the marlin website for confirmation
[WanhaoMakerSelectI3Marlin-1.1.0-RC4.zip](https://github.com/MarlinFirmware/Marlin/files/994799/WanhaoMakerSelectI3Marlin-1.1.0-RC4.zip)
 on what i was doing  here is the old zip

bgort, 2017-05-12:
Did you compare all of configuration.h, configuration_adv.h, and pins_XXXX.h ?

jon012198, 2017-05-12:
i did not look at the pins

bgort, 2017-05-12:
RC4 was quite a while ago.. some of the defaults could have changed.  It's worth checking.

jon012198, 2017-05-12:
it looks the same to me if you want to take a look for me that would be great

bgort, 2017-05-12:
Unfortunately I don't have time tonight - going out of town tomorrow and trying to get a few things finished up first.  Just trying to help out.

Can you more specifically describe the problem?  Is it not working after G29 -only- ?  But it works fine after just a G28?  Any other weird things happening?

jon012198, 2017-05-12:
so it dose it g28 then the g29 after that it heats up to the temps i set then it just sits in the top right of the machine  and just spits out filament out of the hotend

bgort, 2017-05-12:
Is your gcode known to be good?

jon012198, 2017-05-12:
same gcode that the rc4 uses and works

jon012198, 2017-05-12:
so same file

jon012198, 2017-05-12:
i use cura 2.5 to slice the stl's

bgort, 2017-05-12:
Try commenting out the G29 in the gcode and then print it.  Grasping at straws here...

jon012198, 2017-05-12:
so load the new  marlin you want me to try right and take g29 out of the start gcode but g29 is my leveling for z

bgort, 2017-05-12:
I was just wanting to know if G29 is the culprit or if it'll do the same thing without G29.  I understand that G29 levels the bed, but you can cancel it quickly if it starts to print...

jon012198, 2017-05-12:
ok i i will try it loading the new marlin now

jon012198, 2017-05-12:
testing now

Bob-the-Kuhn, 2017-05-12:
I need a break from learning how to digitize embroidery fonts.

Is the working config file posted further up this thread?  I can compare that against the latest and create a useable 1.1.0 config.

jon012198, 2017-05-12:
old marlin https://github.com/MarlinFirmware/Marlin/files/994799/WanhaoMakerSelectI3Marlin-1.1.0-RC4.zip that works new marlin that dose once https://github.com/MarlinFirmware/Marlin/files/982197/Marlin-bugfix-1.1.x.dup.i3.zip

jon012198, 2017-05-12:
if i remove g29 it works
in the new marlin

bgort, 2017-05-12:
Hmm...

Bob-the-Kuhn, 2017-05-12:
MKS_BASE, dual Z, all homing in the - directions, bilinear leveling, fix mounted probe, no EEPROM, no SD card, no LCD, single extruder ...  nothing exotic.

Please do a G29 followed by a M420 V.  I want to see if there's anything really strange with the compensation.

jon012198, 2017-05-12:
i need the eeprom in the marlin to be so i can look at it  do you want me to put that in the start gcode or what @Bob-the-Kuhn

Bob-the-Kuhn, 2017-05-12:
Actually, I'd prefer to keep the EEPROM off.  It could confuse us if we start making changes.

Once we have it working correctly then you can enable the EEPROM and save the compensation.

jon012198, 2017-05-12:
ok how should i send the g29 with  m420 v

Bob-the-Kuhn, 2017-05-12:
What host are you using?

jon012198, 2017-05-12:
i use octoprint on a pi 3

Bob-the-Kuhn, 2017-05-12:
There should be a manual control window where you can send commands to the controller.

jon012198, 2017-05-12:
there is

jon012198, 2017-05-12:
is the command one line or two

Bob-the-Kuhn, 2017-05-12:
Then just type G29 press enter, then M420 V and enter

jon012198, 2017-05-12:
Send: N19 M420 V*107
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.568 +0.165 +0.902
Recv:  1 -0.770 -0.132 +0.482
Recv:  2 -0.932 -0.325 +0.250
Recv: 
Recv: echo:Bed Leveling On
Recv: ok

Bob-the-Kuhn, 2017-05-12:
Do M114 followed by M119.

Can you manually move the head in all directions now?

jon012198, 2017-05-12:
Send: N72 M114*18
Recv: X:194.00 Y:185.00 Z:10.75 E:0.00 Count X:15520 Y:14800 Z:4400
Recv: ok

Send: N79 M119*20
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open

jon012198, 2017-05-12:
yes it all move nicely

Bob-the-Kuhn, 2017-05-12:
Is the X:194.00 Y:185.00 Z:10.75 where G29 left it?

If you send G0 X10, does the head move to location X = 10 or does it just move 10 mm?

jon012198, 2017-05-12:
yes the that is were g29 stoped  
 Send: N156 M114*37
Recv: X:10.00 Y:185.00 Z:10.75 E:0.00 Count X:800 Y:14800 Z:3927

Bob-the-Kuhn, 2017-05-12:
Let's see if we can identify where in the gcode file things go wrong.

Does the problem start immediately or does it print normally for a bit and then screw up?

Try sending the following command sequence.  This is the start of the gcode file without the temperature part and without doing any extruding.

g28
g29
G92 E0

G92 E0

M107
G0 F3000 X85.855 Y57.594 Z0.3

G1 F1500 X86.269 Y56.984
G1 X86.747 Y56.423
G1 X87.283 Y55.916
G1 X87.87 Y55.471
G1 X88.503 Y55.091
G1 X89.172 Y54.783

jon012198, 2017-05-12:
Send: N8 G28*27
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:145.00 Y:135.00 Z:11.00 E:0.00 Count X:11600 Y:10800 Z:4400
Recv: ok
Recv: echo:endstops hit:  Z:1.00
Send: N9 G29*27
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.488 +0.242 +0.985
Recv:  1 -0.740 -0.053 +0.555
Recv:  2 -0.880 -0.250 +0.317
Recv: 
Recv: X:194.00 Y:185.00 Z:10.68 E:0.00 Count X:15520 Y:14800 Z:4400
Recv: ok
Send: N10 M105*22
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N11 M105*23
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N12 G92 e0*84
Recv: X:194.00 Y:185.00 Z:10.68 E:0.00 Count X:15520 Y:14800 Z:4400
Recv: ok
Send: N13 M105*21
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N14 G92 e0*82
Recv: X:194.00 Y:185.00 Z:10.68 E:0.00 Count X:15520 Y:14800 Z:4400
Recv: ok
Send: N15 M105*19
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N16 M105*16
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N17 M105*17
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N18 M107*28
Recv: ok
Send: N19 M105*31
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N20 M105*21
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N21 G0 F3000 X85.855 Y57.594 Z0.3*38
Recv: ok
Send: N22 M105*23
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N23 M105*22
Recv: ok T:28 /0 B:30 /0 @:0 B@:0
Send: N24 M105*17
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N25 G1 F1500 X86.269 Y56.984*121
Recv: ok
Send: N26 M105*19
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N27 M105*18
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N28 G1 X86.747 Y56.423*31
Recv: ok
Send: N29 M105*28
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N30 G1 X87.283 Y55.916*18
Recv: ok
Send: N31 M105*21
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N32 G1 X87.87 Y55.471*42
Recv: ok
Send: N33 M105*23
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N34 G1 X88.503 Y55.091*16
Recv: ok
Send: N35 M105*17
Recv: ok T:28 /0 B:29 /0 @:0 B@:0
Send: N36 G1 X89.172 Y54.783*20
Recv: ok

Bob-the-Kuhn, 2017-05-12:
So far I don't see anything strange.

Did the head move to the X85, Y55 area?

----

Back to the failure mode.

How soon into the print does it start misbehaving?

What are the first symptoms of problems?

jon012198, 2017-05-12:
yes it did  it happens before it get to were it is set to print  so it dose g28,g29 then head bed then hotend and just sits in the back corner and just spits filament out  ther is a video of the problem in this post up  near the top

Bob-the-Kuhn, 2017-05-12:
I noticed that the feed rates go up after the first layer.  Is the first layer printing OK and then the trouble starts?

jon012198, 2017-05-12:
it dose not move to the area to print if it dose print it print fine

Bob-the-Kuhn, 2017-05-12:
It took me a while to find a way to play the file.  I had to open up a video editor and advance it a few frames at a time.  Looks like it starts when G28 is finishing the Z safe homing , probes the bed and then doesn't even try to move to the center of the bed.

I guess the next step is to manually enter every command from the gcode file until it screws up.

I'm going to see if I can reproduce this on my machine.

Bob-the-Kuhn, 2017-05-12:
It ran on my printer.

The next step is to do the commands one by one manually exactly as in the gcode file.  

If it moves out to the center then you can stop entering commands.  

A couple of long shots to try.
- Long shot - if the feedrate is too high Marlin is known to have problems.  Usually it'll go through reset but sometimes it'll also do strange moves. Try cutting your feed rates by a fourth & see if that helps.
- Really long shot - Change the lower case g to uppercase G for the G28 & G29 commands in Cura.  The host should automatically do the conversion & marlin should ignore a lower case g.

jon012198, 2017-05-12:
what printer are you useing

Roxy-3D, 2017-05-12:
So...   You Configuration.h file looks reasonable.  But obviously, something is making it not work on your printer.   For starters...  Just to get more information I would make a few changes.   (And once you figure out what is wrong, you can go back to what you have.)

I would turn these off:
```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOP
```
I would set these to the middle of the bed:
```cpp
#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS 0
#define MANUAL_Z_HOME_POS 0
```
and you may have already tried bumping these way down:
```cpp
#define DEFAULT_MAX_ACCELERATION      { 200, 200, 50, 500 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          200    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  200    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   500    // X, Y, Z acceleration for travel (non printing) moves
```

jon012198, 2017-05-12:
can you tell me what to put there to test  so i dont kill my printer or board

Roxy-3D, 2017-05-12:
For the debug version...  Use the values I just edited into the post.   They may not help.   But those represent the values I would try.

jon012198, 2017-05-12:
any other thing besides ponterface
 i am on windows

[87 more replies]

closed 2017-11-03 by thinkyhead