MarlinFirmware/Marlin #6639: Developer build environment/toolchain documentation

bgort, 2017-05-08:
From #6610:

@bgort:
I think whether or not this is merged, we should consider adding a 'Getting started with PlatformIO' guide somewhere, and then maybe even consider including a brief excerpt in the main readme with a step-by-step for new users.

The excerpt could look something like:

Get started with Marlin using PlatformIO:

Step 1:  Install PIO
Step 2:  git clone https://github.com/MarlinFirmware/Marlin.git MyMarlin
Step 3:  cd MyMarlin
Step 4:  Check to make sure your board is listed in platformio.ini, then:  pio run

To upload:  pio run --target upload
... etc.
(with more detail/instruction, of course)

Or maybe just a link to the guide; might be too messy.

We could give both cli and GUI/Atom examples/instruction, and could also go into detail about how to integrate PIO with various IDEs, etc. Think this would really help new users (and maybe even some experienced users looking to make an environment change).

I can start on some of this soon, but would be good to have help. Anyone?

----

@thinkyhead:
I will include a section about setting up PlatformIO (DevIoT) with Sublime Text 3 in my first video on getting started with Marlin development. It's actually very easy. Explaining how to set up Jekyll (for the website) might be harder.

----

@bgort:
Great. I can do a brief writeup about my setup also; I now have a proper cmake config., and integrated compiling, uploading, etc. work fine. Sorting out jtag debugging is next -- I think I may be able to get at least limited functionality.

Do you have any problem with moving the platformio.ini out to the main directory? In addition to that being more 'normal' and slightly better for IDE integration, I think it'll help potential users realize more quickly that platformio is supported, which I think is a good thing. Don't want to clutter the main directory, but think the benefits outweigh the cost in this case.

markshancock, 2017-05-10:
I am ready to start playing!! Is there a developer "getting started" guide?
On the Wiki front page I saw "head over to the Marlin Development Wiki." and was very hopeful; but when I clicked the link I found a decommissioned repository and my hopefulness went "poof".

1) How do I get started
2) If there isn't something in writing, if someone can direct me a bit, I will create the "getting started" guide myself
3) Where do we want to put the "getting started" guide?  What was the concept behind using the separate repository and why was it abandoned?

markshancock, 2017-05-10:
Looks like this might be a good starting point
http://solidutopia.com/marlin-firmware-user-guide-basic/

Roxy-3D, 2017-05-10:
> How do I get started
> If there isn't something in writing, if someone can direct me a bit, I will create the "getting started" guide myself

Step 1 is to get the build environment setup.   You have to do that just to use Marlin.  So, grab the latest stable release of Arduino  (and probably PronterFace and Slic3r) and try to get them going.

Then...  Pull down the bugfix-v1.1.0 code and start editing the Configuration.h file so it is setup right for your printer.   That will take a little bit of time, but you will learn a huge amount doing that.   With Arduino installed, you double click Marlin.ino and that brings up Arduino.  You can compile with a Ctrl-R or Compile & Upload with a Ctrl-U.   If you can upload your code...   You can keep making changes to your Configuration.h until everything is correct.

And at that point... You can finally print.   And you are ready to start making changes to the firmware.

At 80,000'  That is what you need to do.

markshancock, 2017-05-11:
Ok, hardware arrived today.  I already had Arduino Studio and also created a local clone of the repository using SourceTree.  I have been looking over the code and have answered a couple or StackExchange Marlin questions by reverse engineering the code.
I found some Marlin builds that supposedly work for the X3 at https://github.com/TronxyX3/Wiki/wiki
Should I start with one of them first or try to pull one of their Configuration.h files into the bugfix-v1.1.0 code base?

FHeilmann, 2017-05-11:
That config is for Marlin 1.0, not Marlin 1.1 so DON'T just copy paste it. Have a look at what is defined there and look for the corresponding values in the 1.1 config for a starting point.

markshancock, 2017-05-11:
Thanks, will do.  Since their repository is forked off of yours, I just added their remote and I can see see where they branched.  Looks like they have several branches; but, they are all old.

closed 2017-11-01 by thinkyhead