MarlinFirmware/Marlin #6641: Problems with Z offset using Bilinear auto bed leveling option Marlin V1.1.0

Part0, 2017-05-08:
Hi,
I’m having problems with the Bilinear auto bed levelling. Having set up auto levelling and attempt to run a print, I can see that the autoleveling is working in that the Z axis is constantly correcting within each layer but the first layer height results with auto levelling is worse than I can achieve by manually adjusting my four bed screws. Below is the output from tests ran. All the steps below were done in direct sequence as listed. 

My printer is a hybrid cnc type Cartesian,  running marlin firmware 1.1.0. Electronics is Generation 7 1.5 board. I’m using an infrared proximity sensor EE-SPY402 for the Z probe. Slicer is Simplify 3D. I had the same problems with RC8. I have added questions at the appropriate point in the output below. I have attached my config files at the end.

SENT: M501
READ: echo:V37 stored settings retrieved (517 bytes)
V37 stored settings retrieved (517 bytes)
READ: echo:  G21 ; Units in mm
G21 ; Units in mm
READ: 
READ: 
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D3.00
M200 D3.00
READ: echo:  M200 D0
M200 D0
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X11.94 Y13.93 Z500.00 E129.70
M92 X11.94 Y13.93 Z500.00 E129.70
READ: echo:Maximum feedrates (units/s):
Maximum feedrates (units/s):
READ: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
M203 X300.00 Y300.00 Z5.00 E25.00
READ: echo:Maximum Acceleration (units/s2):
Maximum Acceleration (units/s2):
READ: echo:  M201 X3000 Y3000 Z100 E10000
M201 X3000 Y3000 Z100 E10000
READ: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P500.00 R3000.00 T500.00
M204 P500.00 R3000.00 T500.00
READ: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
READ: echo:  M205 S0.00 T0.00 B20000 X5.00 Y5.00 Z0.40 E5.00
M205 S0.00 T0.00 B20000 X5.00 Y5.00 Z0.40 E5.00
READ: echo:Home offset:
Home offset:
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:Auto Bed Leveling:
Auto Bed Leveling:
READ: echo:  M420 S0 Z0.00
M420 S0 Z0.00
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P15.00 I0.68 D52.99
M301 P15.00 I0.68 D52.99
READ: echo:  M304 P76.44 I3.17 D461.06
M304 P76.44 I3.17 D461.06
READ: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
READ: echo:  M851 Z-5.80
M851 Z-5.80
READ: ok
SENT: G28
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
[... 349 lines cut ...]
READ: ok
READ: ok
READ: ok
SENT: G91
SENT: G1 Z-0.100 F1200
SENT: G90
READ: ok
READ: ok
READ: ok
SENT: G91
SENT: G1 Z-0.100 F1200
SENT: G90
READ: ok
READ: ok
READ: ok

Sending and M114 at this point gave the following.

SENT: M114
READ: X:179.00 Y:142.00 Z:0.00 E:0.00 Count X:2137 Y:1978 Z:-148
READ: ok
-148/500 = -0.296mm????????
Hope someone can make sense of what’s going on here because I’m certainly confused.

[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/983273/Config.files.zip)

knobik, 2017-05-08:
i have the same problem as you, doing G1 Z0 after G29 (everything preheated) then its moving to a good distance between bed and hotend, but as soon as i start printing, it looks like the mesh is inverted OR not active at all...

tcm0116, 2017-05-08:
> Does the output matrix above not represent the Z endstop position to which the Z offset is then applied??

It's not a matrix in the math-sense, but each number does represent the distance between the Z endstop position (Z=0) and the Z position at each point that was probed when the probe was triggered.

> What is the significance of values I got close to +2mm?

I suspect that when your probe is being triggered, the nozzle is roughly 2mm above the bed, which you appear to have confirmed:

> I issued a G1 Z0 and the nozzle went to within approx. 2mm of the bed.

There's a new feature in Configuration_adv.h called `BABYSTEP_ZPROBE_OFFSET`. If you enable it along with `BABYSTEPPING`, then you'll be able to adjust the Z-Offset in the Control/Motion menu on the LCD (or via the `M851 `command) and it will cause the nozzle to raise or lower accordingly. Here's the process I usually follow when first setting up the Z-Offset:

1. Heat the bed if you have a heated bed.
2. Set the Z-Offset to 0.0mm.
3. Run `G28 `and then `G29` to home and collect the leveling grid.
4. Move the nozzle to the center of the bed (`G1 X100 Y100 F6000`).
5. Incrementally lower the nozzle until you get to Z=0 or the nozzle touches the bed.
6. If the nozzle touches the bed, set the Z-Offset to the current Z position, which should raise the nozzle by that amount, and then repeat step 5.
7. If the nozzle is above the bed when Z=0, then decrease the Z-Offset until the nozzle grabs a piece of paper between it and the bed.
8. Print a test cube and adjust the Z-Offset during the first layer to perfection (don't forget to add G28 to your slicer's start G-Code after the G28).
9. Save settings.

Part0, 2017-05-09:
Thanks tcm0116 for the feedback. Can you try the sequence below on your printer with auto bed disabled and the Z probe as the Z end stop, which I take it is your setup. I'm running firmware V1.1.0

1)Turn off Auto be levelling with M420 S0. Not sure if this is necessary but just to be sure.
2) Issue G28 to home all axis.
3) Send the Z axis nozzle to Z0 with G1 Z0 command. The nozzle moves to the correct height to print the first layer.
4) Move to X10mm Y10mm and Z 10mm.
5) Send the Z axis nozzle to Z0. The nozzle moves to the correct height to print the first layer.
6) Issue home Z only command G28 Z0. 
7) Send the Z axis nozzle to Z0 with G1 Z0 command. This time the nozzle moves to about 2mm above the glass ????????
8) Issue M114. This gives the Z axis as 0. But it’s not.!!!!

I have only moved 10mm across the glass so there is no possibility that there is any change in bed height. 
Here's my printout in response to the sequence.

READ: ok
SENT: M420 S0
READ: echo:Bed Leveling Off
Bed Leveling Off
READ: ok
SENT: G28
READ: X:0.00 Y:0.00 Z:15.80 E:0.00 Count X:0 Y:0 Z:7900
READ: ok
READ: echo:endstops hit:  Z:5.80
endstops hit:  Z:5.80
SENT: G1 Z0
READ: ok
SENT: G1 X10 Y10 Z10 F1000
READ: ok
SENT: M114
READ: X:10.00 Y:10.00 Z:10.00 E:0.00 Count X:119 Y:139 Z:5000
READ: ok
SENT: G1 Z0
READ: ok
SENT: G28 Z0
READ: X:10.00 Y:10.00 Z:15.80 E:0.00 Count X:119 Y:139 Z:7900
READ: ok
READ: echo:endstops hit:  Z:5.80
endstops hit:  Z:5.80
SENT: G1 Z0
READ: ok
SENT: M114
READ: X:10.00 Y:10.00 Z:0.00 E:0.00 Count X:119 Y:139 Z:0
READ: ok

thinkyhead, 2017-05-14:
> I would have expected the matrix of values resulting from G29 to be close to Zero.

When the Z probe offset is well-tuned, that is the case. If **every point** in the matrix is off by some minimum amount, adjust your Z probe Z offset by that amount.

Part0, 2017-05-15:
Thanks thinkyhead for your comments.  Based on what you said I tried the following three tests ;
1) Run G29 with an A4 white printing paper sheet clipped on top of 6mm glass plate on heated bed.
READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4
READ:  0 +0.338 +0.236 +0.324 +0.088 +0.098
READ:  1 +0.356 +0.130 +0.170 +0.084 +0.014
READ:  2 +0.168 +0.060 +0.202 +0.122 +0.122
READ:  3 -0.100 +0.016 +0.208 -0.016 +0.058
READ:  4 -0.062 +0.146 +0.212 +0.202 +0.082

2) Run G29 with A4 sheet under glass.
READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4
READ:  0 +1.030 +1.116 +0.894 +0.918 +0.494
READ:  1 +0.676 +0.570 +0.982 +1.064 +0.982
READ:  2 +0.948 +0.574 +0.262 +0.552 +0.894
READ:  3 +0.544 +1.078 +0.822 +0.524 +0.238
READ:  4 +0.096 +0.606 +0.424 +0.854 -0.464

3) Rub G29 on bare glass plate.
READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4
READ:  0 -2.268 -1.942 -2.278 -2.040 -2.350
READ:  1 -2.756 -2.408 -2.190 -1.988 -1.982
READ:  2 -2.356 -2.746 -2.320 -2.438 -1.998
READ:  3 -2.398 -2.206 -2.306 -2.574 -2.966
READ:  4 -3.142 -2.686 -2.826 -2.590 -3.548

Thanks for your help. Hopefully by covering my bed as you suggest this will solve my issues.

closed 2018-02-06 by thinkyhead