MarlinFirmware/Marlin #6642: M81 More than just the ATX power supply?

davidgrm, 2017-05-08:
Hi

I have version 1.1.0 of Marlin running on a Ramps board. I am using the single output to control an ATX power supply. The ATX power supply is only being used for the Heatbed. I have a different power supply connected to the other 12 volt input for the motors and hotend.
I had a strange problem where just before finishing a print both heaters showed that they had been turned off. For some reason Mattercontrol sent the M81 command before it finished the print. This should have just turned off the heatbed but when examining the log file I saw this...

->N53688 M105*39 [4234.121]
<-ok **T:210 /210 B:58 /65 @:63** B@:127 [4234.124]
->N53689 M81*27 [4234.124]
<-echo:busy: processing [4236.156]
<-ok [4238.925]
->N53690 M105*46 [4238.925]
<-ok **T:209 /0 B:58 /0** @:0 B@:0 [4238.942]

The bold text shows that both hotend and bed have temps set. 
After the M81 both have temps set to 0.

I did not see this funstionality mentioned in the doc.
Any thoughts on this?

I am still trying to figure out why when the M81 was sent the bed already seemed to be off - see the temp is only 58 and should be 65 ... but thats a different matter I assume?

Tannoo, 2017-05-08:
Yup, by design. I'm sure the assumption is only one power supply. So, rightfully, shut everything down.
```
/**
 * M81: Turn off Power, including Power Supply, if there is one.
 *
 *      This code should ALWAYS be available for EMERGENCY SHUTDOWN!
 */
inline void gcode_M81() {
  thermalManager.disable_all_heaters();
  stepper.finish_and_disable();
  #if FAN_COUNT > 0
    for (uint8_t i = 0; i < FAN_COUNT; i++) fanSpeeds[i] = 0;

    #if ENABLED(PROBING_FANS_OFF)
      fans_paused = false;
      ZERO(paused_fanSpeeds);
    #endif
  #endif
  safe_delay(1000); // Wait 1 second before switching off
  #if HAS_SUICIDE
    stepper.synchronize();
    suicide();
  #elif HAS_POWER_SWITCH
    OUT_WRITE(PS_ON_PIN, PS_ON_ASLEEP);
  #endif
  #if ENABLED(ULTIPANEL)
    #if HAS_POWER_SWITCH
      powersupply = false;
    #endif
    LCD_MESSAGEPGM(MACHINE_NAME " " MSG_OFF ".");
  #endif
}
```

davidgrm, 2017-05-09:
So does that mean there you can turn it on with an M80 but then never turn it off again without turning all heaters off?

Where can I find info regarding the replies from Marlin? Such as what does the B@:127 mean?

M8127 seems to have been just an error introduced from the cut and paste. I originally asked the question on reprap forum but got no reply so I cut and pasted from there to here.
I have discovered that the bed temp was too low because the ATX power supply appears to be faulty. Guess its time to purchase a decent one.

davidgrm, 2017-05-13:
I dont think this is  correct. Wiki = "Turns off the ATX power supply. Counterpart to M80. "
The ATX power supply is controlled by a single out and that is all this should do.
If you want everything to shut down then there are other codes which can be used such as M112.

If you just want the heaters off there are codes for them as well.  I think this is just creating a duplicate code and removing something usefull

closed 2018-02-06 by thinkyhead