MarlinFirmware/Marlin #6643: Weird issue with M303?

bgort, 2017-05-08:
Was just autotuning (M303) my extruders and bed and somehow my LCD ended up so dim I could hardly see it.

Then, after restarting the printer, I see:
Recv: echo:LCD Contrast:
Recv: echo:  M250 C-32768

Bizarre.  It was fine previously, and I didn't change anything (certainly didn't save anything).  Easy enough to fix, of course - M502, M500 did the job, and I'm sure M250 would have, too - but thought I'd post here.

I haven't yet looked at the code, but may do that sometime soon.

Tannoo, 2017-05-08:
Check for shorts on the 5v bus. When the regulator gets overloaded, that is the first thing that is noticed because the LCD usually is the power hog on the 5v bus.

Tannoo, 2017-05-08:
Is your contrast software adjustable?

Tannoo, 2017-05-08:
Yeah... I didn't look up M250. Sorry.

bgort, 2017-05-08:
No worries.  What yo suggested was my first guess, too -- an overloaded regulator.  But a quick reset of the firmware fixed it, and M250 adjusts it fine, so I don't think that was it.  

Maybe M303 overwrote the contrast value in memory... ?  Not sure yet - may check soon.

closed 2017-11-01 by thinkyhead