MarlinFirmware/Marlin #6646: G26 tried to move X axis off of end of carriage

bgort, 2017-05-08:
Was just printing a validation mesh and canceled by holding the control wheel because I wasn't getting good bed adhesion.  

After cancel, G26 normally returns the nozzle to the coordinates it started at, but this time it tried to send the X axis off into never-never land (< X0) and I had to power down the printer to prevent damage.

I had started the G26 in the X~0, Y~310 corner.

Roxy-3D, 2017-05-08:
The X & Y position get initialized at the start of the G26 to the current position if not specified:
```cpp
    x_pos = current_position[X_AXIS];
    y_pos = current_position[Y_AXIS];
```
and then at the very end...
```cpp
    move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], 0); // Raise the nozzle
    destination[X_AXIS] = x_pos;                                               // Move back to the starting position
    destination[Y_AXIS] = y_pos;
     move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], 0); // Move back to the starting position
```
And here is the other thing...   x_pos & y_pos are used constantly as the algorithm tries to find the next closest mesh point that has not been processed yet.   So if the pattern was drawing normally...  that implies x_pos & y_pos were not corrupted (at least at that point in time).

So...  That is a long way of saying "I don't know what could have caused that!"

bgort, 2017-05-08:
I wonder if it's that I started the G26 outside of the UBL area (the bed minus the insets) ...?

I'm not sure exactly where the Y was, but it's likely I was at the max limit for the axis, which is just outside the UBL area (I left a few mm off on that side) ...

Roxy-3D, 2017-05-08:
> I wonder if it's that I started the G26 outside of the UBL area (the bed minus the insets) ...?

Oh!!!   That may be part of the equation to re-duplicate this.   Because G26 does try to go back to where it started!

bgort, 2017-05-08:
That's probably it.  I'll try it in a bit.

In other news, I'm currently extremely frustrated, as I can't get a decent mesh created -- every time I G29 P1 I get a different looking mesh (different z values), and G29 P2 B (which I like to use to fill in the rows the probe can't hit) is giving unexpected values (almost entirely inconsistent with the probed values) .. as in a P1-probed z value will be -0.050, which looks and is generally right, but the adjacent P2 B-probed z value will be +0.200 .. and my bltouch is good within 10um at the moment and my bed is pretty flat, so that's not really possible,  I know it used to work, so not sure what happened...  Maybe there was some random squaring-the-same-variable-instead-of-multiplying-different-variables or some other new maths applied.

Roxy-3D, 2017-05-08:
Does M48 give you good results?   The G29 P1 code is fairly simple because it just invalidates the mesh and uses the standard lower level probe functions.    It might be useful to do a G28 and a G29 P1 V4 and let it get a few points...    Stop it and do another G28 and G29 P1 V4 and see if those points compare at all?   And probably, for that little test, it makes sense to have the machine's heaters turned off.

If you can't get any consistency with that little test...   We may have a bigger problem with the probing sub-system.   I sure hope that is not true.   But that would explain some of the craziness we have been seeing.

bgort, 2017-05-08:
Yes.  I'm averaging about 10um stddev, which is more than I used to have, but acceptable -- and certainly good enough for this purpose.

Right now I'm going back to a broken branch I still have lying around (one of the ones before the optimization issue was sorted out) that I -know- worked very well (other than the memory issue) ...

Will report on that shortly, then try your test if I'm no further along.

> But that would explain some of the craziness we have been seeing.

Yeah, there seems to be a lot of unusual stuff happening...

bgort, 2017-05-08:
edcffdc0e3e461afb27466aa75f3bf59660a6e3d (3 weeks ago), bed cold, endstop interrupts
Recv: Standard Deviation: 0.003623
Recv: Standard Deviation: 0.004301
Recv: Standard Deviation: 0.009017
Recv: Standard Deviation: 0.008497
Recv: Standard Deviation: 0.007009
Recv: Standard Deviation: 0.006270
Recv: Standard Deviation: 0.002267
Recv: Standard Deviation: 0.007579
Recv: Standard Deviation: 0.005305
(avg of 6, which is only ~60% of what I was getting earlier in the day (average of 10-11um) .. and better than anything I've seen recently .. also more consistent with what I used to see regularly)

edcffdc0e3e461afb27466aa75f3bf59660a6e3d, bed hot and stable (soaked for 20 min), endstop interrupts
Recv: Standard Deviation: 0.013143
Recv: Standard Deviation: 0.010023
Recv: Standard Deviation: 0.026279
Recv: Standard Deviation: 0.004274
Recv: Standard Deviation: 0.005317
Recv: Standard Deviation: 0.017623
Recv: Standard Deviation: 0.029364
Recv: Standard Deviation: 0.015365

edcffdc0e3e461afb27466aa75f3bf59660a6e3d, bed hot and stable, endstop interrupts off
Recv: Standard Deviation: 0.011665
Recv: Standard Deviation: 0.023522
Recv: Standard Deviation: 0.008746
Recv: Standard Deviation: 0.006380
Recv: Standard Deviation: 0.012174
Recv: Standard Deviation: 0.002875
Recv: Standard Deviation: 0.016737
Recv: Standard Deviation: 0.006750

So this is no help.

What I really don't understand is how a probe that can do this:
Recv: Standard Deviation: 0.002267

Can also do this, without being touched:
Recv: Standard Deviation: 0.029364 (that one had a range of 70um)

No way is the bed changing shape to that degree due to thermal or other effects.  I saw only 35um deviation earlier with bang-bang (between heater on and heater off).  I'm using PWM now and there's no more than 5um variation once the bed is to temp.

I guess I'll go back to an even older version and try that.

Roxy-3D, 2017-05-09:
> I guess I'll go back to an even older version and try that.

I have another idea...  I have a 'brand new' probe from back when they were still hand assembling things...    It has the 5 ms. pulse.   If you are willing to bolt it on your machine with the dial indicator...   We will know a lot more about what is going on....

bgort, 2017-05-09:
I'm going to try my old one (which I don't necessarily think was damaged, now that I know the new one is doing similar things), but may take you up on that if I continue to struggle.  

Going to revert to a Marlin version another week or so older than the one I just tried - the exact one I got the beautiful G26 prints out of, if possible.  Am also going to spend some more time 'profiling' the bed to make extra-double-sure it isn't changing shape any more than I saw last time I checked.

Obviously something is going on here...

Roxy-3D, 2017-05-09:
The thing is...   The code that does the probing is a little bit convoluted...  But the actual code that locks in the sampled points is pretty simple.   I don't have the Git skills to look for changes in that code, but it is all down in the interrupt handlers.    There is very little code involved in this.

The one thing I'm wondering is:   Did something happen such that the point is not being sampled as quickly?   Did nested interrupts get turned on and thus we have a big delay because something (like the LCD Update) is happening after the sampled point triggers but before it got locked in?   Or maybe the opposite...   Maybe the processor is now in a timer update...  and we have locked out interrupts from the probe coming in?

bgort, 2017-05-09:
> The thing is... The code that does the probing is a little bit convoluted... But the actual code that locks in the sampled points is pretty simple. I don't have the Git skills to look for changes in that code, but it is all down in the interrupt handlers. There is very little code involved in this.

Can probably use 'blame', which is something available on github.com and is pretty easy to use.  If you go to the source file (I assume the code in question is in Marlin_main.cpp) and then click the 'Blame' button in the upper right hand corner, it'll show you who the last person to change a particular line was off to the left.  Not sure if you're familiar with that, but thought I'd mention.

> The one thing I'm wondering is: Did something happen such that the point is not being sampled as quickly? Did nested interrupts get turned on and thus we have a big delay because something (like the [...]

That was one of my early concerns - a longer routine causing a delay that's preventing the detection of the endstop .. or maybe the triggering of an overlapping interrupt in the middle of that process .. I think either of the those would explain why I see the axis moving further before stopping and reversing on 'bad' probes.  But that doesn't explain why it happens when the bed is warm.

I just tried the Marlin version that gave me the beautiful G26 right out of the bo, without any modification of the grid (which implies pretty strongly that the probing was really good/reliable) and I'm seeing the same 3-6um M48 sigma when cold.  Haven't tried M48 hot or generating a mesh yet, but will shortly.  

Tweaking my bed leveling screws after rebuilding the bed just now, then going to 'profile' with the dial indicator to make sure the bed is stable when hot.

Roxy-3D, 2017-05-09:
> just tried the Marlin version that gave me the beautiful G26 right out of the bo, without any modification of the grid (which implies pretty strongly that the probing was really good/reliable) and [...]

This would be horribly brute force...   But pulling that directory into a Visual Diff program and comparing against the current would tell us a huge amount.   If that old version checks out...  How about you .ZIP it up and post it here...   I'll be able to help with the scavenger hunt.

bgort, 2017-05-09:
Thank you.

Yeah.. we'll see here shortly.  So far, the bed is really stable.  It's going convex as it's warming up, but is relatively stable with just a few um delta / min, so far.  Nothing that would affect M48 even if it were in the middle of heating -- it doesn't run long enough for these changes to matter.

bgort, 2017-05-09:
```
Recv: 1 of 10: z: 0.009 mean: 0.0087 sigma: 0.000000 min: 0.009 max: 0.009 range: 0.000
Recv: 2 of 10: z: 0.001 mean: 0.0050 sigma: 0.003750 min: 0.001 max: 0.009 range: 0.008
Recv: 3 of 10: z: -0.000 mean: 0.0033 sigma: 0.003864 min: -0.000 max: 0.009 range: 0.009
Recv: 4 of 10: z: 0.007 mean: 0.0044 sigma: 0.003802 min: -0.000 max: 0.009 range: 0.009
Recv: 5 of 10: z: -0.008 mean: 0.0020 sigma: 0.005842 min: -0.008 max: 0.009 range: 0.016
Recv: 6 of 10: z: -0.021 mean: -0.0019 sigma: 0.010174 min: -0.021 max: 0.009 range: 0.030
Recv: 7 of 10: z: -0.030 mean: -0.0059 sigma: 0.013623 min: -0.030 max: 0.009 range: 0.039
Recv: 8 of 10: z: -0.045 mean: -0.0108 sigma: 0.018157 min: -0.045 max: 0.009 range: 0.054
Recv: 9 of 10: z: -0.048 mean: -0.0149 sigma: 0.020645 min: -0.048 max: 0.009 range: 0.056
Recv: 10 of 10: z: -0.060 mean: -0.0194 sigma: 0.023811 min: -0.060 max: 0.009 range: 0.069
Recv: Finished!
Recv: Mean: -0.019375 Min: -0.060 Max: 0.009 Range: 0.069
Recv: Standard Deviation: 0.023811
```

With the bed hot.  No way did the bed change shape that much.  It's stable at temp +/- <2um currently.

bgort, 2017-05-09:
Notice that on the first few points all is well.. then starting with point 6 we have the exact same progressively-worse z.....

bgort, 2017-05-09:
Same thing, again .. first few points fine, last points really 'off' .. watching dial indicator confirmed it, too.  

The axis started at and returned to 0.000 on the dial indicator, too, so no issue there.

Roxy-3D, 2017-05-09:
OK...   Several questions:
- This is with the fairly old version of the firmware, right?
- This is a little bit better than with the current firmware, right?

Now some speculation...   Is it possible we are seeing an electrical issue?  The BL-Touch does suck a lot of current when it moves the pin.   What if we put a 10,000 uf capacitor on its power???   Do we get better readings????

martindb, 2017-05-09:
I'm following this conversation with atention.
If there is some kind of timing issue, could a slower z feedrate during the probe give an accurate measure?

Roxy-3D, 2017-05-09:
That is a good point.   And remember...  With BL-Touch probes we used to have to move super fast because of the short pulse length.   However...  I thing bgort is using interrupts and in theory...  when it triggers, it should get the point very quickly.

bgort, 2017-05-09:
>This is with the fairly old version of the firmware, right?

Yes.  This is with RCBugFix from April 13 -- the one where I got the great G26 (and great real prints).

>This is a little bit better than with the current firmware, right?

Yes, I believe so.  And -- also, the weird issue where Z moves didn't match up on the dial indicator is completely gone.  I don't know where I posted that, but it was like it was off by 32um, then 27, etc. at every 1mm step.  This version doesn't have that issue at all -- it's dead on at every Z position.

> Now some speculation... Is it possible we are seeing an electrical issue? The BL-Touch does suck a lot of current when it moves the pin. What if we put a 10,000 uf capacitor on its power??? Do we get better readings????

I wondered/worried about this, too, so my BLTouch is on its own linear regulator (a Fairchild LM7805, which is really solid) with hefty caps.  I really doubt that's the problem here.

bgort, 2017-05-09:
> That is a good point. And remember... With BL-Touch probes we used to have to move fast because of the short pulse length. However... I thing bgort is using interrupts and in theory... when it triggers, it should get the point very quickly.

Good idea.  Going to slow it way down now -- but why is it 'fine' when the bed is cold and not when it isn't?  So I don't have to go looking, does manage_heaters() or the thermal ISR fire more often when target temps are set?

Roxy-3D, 2017-05-09:
>> Now some speculation... Is it possible we are seeing an electrical issue? The BL-Touch does suck a lot of current when it moves the pin. What if we put a 10,000 uf capacitor on its power??? Do we get better readings????

> I wondered/worried about this, too, so my BLTouch is on its own linear regulator (a Fairchild LM7805, which is really solid) with hefty caps. I really doubt that's the problem here.

OK...  But...  it seems like in a lot of your posts, 3 or 4 or 5 points are good...   And then the numbers start getting bad.  Is it an internal heat issue inside the BL-Touch?   One thing that would be very easy to do is put a 30 second safe_delay() call inside of M48.    If we do that...  Do you then get 5 or 6 or 8 good points before they go to shit?

Roxy-3D, 2017-05-09:
> So I don't have to go looking, does manage_heaters() or the thermal ISR fire more often when target temps are set?

No...  I don't believe so.   But depending on what the counters are doing....  there can be much more or less calculations being done.

bgort, 2017-05-09:
> OK... But... it seems like in a lot of your posts, 3 or 4 or 5 points are good... And then the numbers start getting bad. Is it an internal heat issue inside the BL-Touch?

It doesn't do it when the bed is cold, and I can't imagine heat radiated from the bed is enough to throw things off in the BLTouch itself...  I still need to try the blow dryer idea you had, though.

> One thing that would be very easy to do is put a 30 second safe_delay() call inside of M48. If we do that... Do you then get 5 or 6 or 8 good points before they go to shit?

That's a good idea.  I'll try that too.

-- 

Here's 1/4 probing speed:
```
Recv: 1 of 10: z: -0.023 mean: -0.0225 sigma: 0.000000 min: -0.023 max: -0.023 range: 0.000
Recv: 2 of 10: z: -0.023 mean: -0.0225 sigma: 0.000000 min: -0.023 max: -0.023 range: 0.000
Recv: 3 of 10: z: -0.031 mean: -0.0254 sigma: 0.004125 min: -0.031 max: -0.023 range: 0.009
Recv: 4 of 10: z: -0.034 mean: -0.0275 sigma: 0.005078 min: -0.034 max: -0.023 range: 0.011
Recv: 5 of 10: z: -0.026 mean: -0.0273 sigma: 0.004569 min: -0.034 max: -0.023 range: 0.011
Recv: 6 of 10: z: -0.019 mean: -0.0258 sigma: 0.005237 min: -0.034 max: -0.019 range: 0.015
Recv: 7 of 10: z: -0.023 mean: -0.0254 sigma: 0.004987 min: -0.034 max: -0.019 range: 0.015
Recv: 8 of 10: z: -0.020 mean: -0.0247 sigma: 0.004990 min: -0.034 max: -0.019 range: 0.015
Recv: 9 of 10: z: -0.019 mean: -0.0240 sigma: 0.005061 min: -0.034 max: -0.019 range: 0.015
Recv: 10 of 10: z: -0.019 mean: -0.0235 sigma: 0.005056 min: -0.034 max: -0.019 range: 0.015
Recv: Finished!
Recv: Mean: -0.023500 Min: -0.034 Max: -0.019 Range: 0.015
Recv: Standard Deviation: 0.005056
```
So that's a significant improvement.  Going to try this a few more times to make sure it isn't a fluke.  Took forever....

And, yes I'm using interrupts.  There doesn't seem to be any difference with or without interrupts, though -- does the same thing, more or less.

bgort, 2017-05-09:
```
Recv: 1 of 10: z: -0.041 mean: -0.0413 sigma: 0.000000 min: -0.041 max: -0.041 range: 0.000
Recv: 2 of 10: z: -0.036 mean: -0.0388 sigma: 0.002500 min: -0.041 max: -0.036 range: 0.005
Recv: 3 of 10: z: -0.028 mean: -0.0350 sigma: 0.005683 min: -0.041 max: -0.028 range: 0.014
Recv: 4 of 10: z: -0.050 mean: -0.0388 sigma: 0.008149 min: -0.050 max: -0.028 range: 0.023
Recv: 5 of 10: z: -0.050 mean: -0.0410 sigma: 0.008566 min: -0.050 max: -0.028 range: 0.023
Recv: 6 of 10: z: -0.041 mean: -0.0410 sigma: 0.007820 min: -0.050 max: -0.028 range: 0.023
Recv: 7 of 10: z: -0.033 mean: -0.0398 sigma: 0.007833 min: -0.050 max: -0.028 range: 0.023
Recv: 8 of 10: z: -0.034 mean: -0.0391 sigma: 0.007597 min: -0.050 max: -0.028 range: 0.023
Recv: 9 of 10: z: -0.034 mean: -0.0385 sigma: 0.007355 min: -0.050 max: -0.028 range: 0.023
Recv: 10 of 10: z: -0.043 mean: -0.0389 sigma: 0.007081 min: -0.050 max: -0.028 range: 0.023
Recv: Finished!
Recv: Mean: -0.038875 Min: -0.050 Max: -0.028 Range: 0.023
Recv: Standard Deviation: 0.007081
```
and
```
Recv: 1 of 10: z: -0.061 mean: -0.0613 sigma: 0.000000 min: -0.061 max: -0.061 range: 0.000
Recv: 2 of 10: z: -0.056 mean: -0.0588 sigma: 0.002500 min: -0.061 max: -0.056 range: 0.005
Recv: 3 of 10: z: -0.064 mean: -0.0604 sigma: 0.003118 min: -0.064 max: -0.056 range: 0.007
Recv: 4 of 10: z: -0.078 mean: -0.0647 sigma: 0.007875 min: -0.078 max: -0.056 range: 0.021
Recv: 5 of 10: z: -0.061 mean: -0.0640 sigma: 0.007176 min: -0.078 max: -0.056 range: 0.021
Recv: 6 of 10: z: -0.058 mean: -0.0629 sigma: 0.006985 min: -0.078 max: -0.056 range: 0.021
Recv: 7 of 10: z: -0.048 mean: -0.0607 sigma: 0.008421 min: -0.078 max: -0.048 range: 0.030
Recv: 8 of 10: z: -0.040 mean: -0.0581 sigma: 0.010440 min: -0.078 max: -0.040 range: 0.038
Recv: 9 of 10: z: -0.021 mean: -0.0540 sigma: 0.015204 min: -0.078 max: -0.021 range: 0.056
Recv: 10 of 10: z: -0.001 mean: -0.0488 sigma: 0.021418 min: -0.078 max: -0.001 range: 0.076
Recv: Finished!
Recv: Mean: -0.048750 Min: -0.078 Max: -0.001 Range: 0.076
Recv: Standard Deviation: 0.021418
```

So maybe a fluke??  Look at the last points in that last one - progressively 'worse' starting at #4...

martindb, 2017-05-09:
Very strange.... some kind of acumulative error...? What if you reset your board and prior to each M48 test?

Roxy-3D, 2017-05-09:
Yeah...   The big problem is we don't know what we are chasing...

bgort, 2017-05-09:
With a 30 second safe_delay between probes:

```
Recv: 1 of 10: z: -0.095 mean: -0.0950 sigma: 0.000000 min: -0.095 max: -0.095 range: 0.000
Recv: 2 of 10: z: -0.088 mean: -0.0913 sigma: 0.003750 min: -0.095 max: -0.088 range: 0.007
Recv: 3 of 10: z: -0.090 mean: -0.0908 sigma: 0.003118 min: -0.095 max: -0.088 range: 0.007
Recv: 4 of 10: z: -0.079 mean: -0.0878 sigma: 0.005888 min: -0.095 max: -0.079 range: 0.016
Recv: 5 of 10: z: -0.075 mean: -0.0853 sigma: 0.007348 min: -0.095 max: -0.075 range: 0.020
Recv: 6 of 10: z: -0.073 mean: -0.0831 sigma: 0.008221 min: -0.095 max: -0.073 range: 0.022
Recv: 7 of 10: z: -0.066 mean: -0.0807 sigma: 0.009633 min: -0.095 max: -0.066 range: 0.029
Recv: 8 of 10: z: -0.085 mean: -0.0813 sigma: 0.009122 min: -0.095 max: -0.066 range: 0.029
Recv: 9 of 10: z: -0.080 mean: -0.0811 sigma: 0.008609 min: -0.095 max: -0.066 range: 0.029
Recv: 10 of 10: z: -0.081 mean: -0.0811 sigma: 0.008167 min: -0.095 max: -0.066 range: 0.029
Recv: Finished!
Recv: Mean: -0.081125 Min: -0.095 Max: -0.066 Range: 0.029
Recv: Standard Deviation: 0.008167
```

I'm going to do this a few more times, and I'm also charging my thermal camera; will record the bltouch during a longer M48 to see if there's any noticeable deviation in temperature....

Roxy-3D, 2017-05-09:
Wow!!!    I know it is a total nuisance...  But if 30 seconds between probes did this...   I bet 60 seconds fixes the issue!!!!   If so...   The Antclabs people are going to want to know about this!

bgort, 2017-05-09:
Second time...

```
Recv: 1 of 10: z: -0.081 mean: -0.0813 sigma: 0.000000 min: -0.081 max: -0.081 range: 0.000
Recv: 2 of 10: z: -0.081 mean: -0.0813 sigma: 0.000000 min: -0.081 max: -0.081 range: 0.000
Recv: 3 of 10: z: -0.091 mean: -0.0846 sigma: 0.004714 min: -0.091 max: -0.081 range: 0.010
Recv: 4 of 10: z: -0.079 mean: -0.0831 sigma: 0.004801 min: -0.091 max: -0.079 range: 0.013
Recv: 5 of 10: z: -0.083 mean: -0.0830 sigma: 0.004301 min: -0.091 max: -0.079 range: 0.013
Recv: 6 of 10: z: -0.104 mean: -0.0865 sigma: 0.008673 min: -0.104 max: -0.079 range: 0.025
Recv: 7 of 10: z: -0.100 mean: -0.0884 sigma: 0.009323 min: -0.104 max: -0.079 range: 0.025
Recv: 8 of 10: z: -0.105 mean: -0.0905 sigma: 0.010307 min: -0.105 max: -0.079 range: 0.026
Recv: 9 of 10: z: -0.095 mean: -0.0910 sigma: 0.009821 min: -0.105 max: -0.079 range: 0.026
Recv: 10 of 10: z: -0.123 mean: -0.0941 sigma: 0.013277 min: -0.123 max: -0.079 range: 0.044
Recv: Finished!
Recv: Mean: -0.094125 Min: -0.123 Max: -0.079 Range: 0.044
Recv: Standard Deviation: 0.013277
```

bgort, 2017-05-09:
Third time...
```
Recv: 1 of 10: z: -0.115 mean: -0.1150 sigma: 0.000000 min: -0.115 max: -0.115 range: 0.000
Recv: 2 of 10: z: -0.125 mean: -0.1200 sigma: 0.005000 min: -0.125 max: -0.115 range: 0.010
Recv: 3 of 10: z: -0.119 mean: -0.1196 sigma: 0.004125 min: -0.125 max: -0.115 range: 0.010
Recv: 4 of 10: z: -0.109 mean: -0.1169 sigma: 0.005896 min: -0.125 max: -0.109 range: 0.016
Recv: 5 of 10: z: -0.105 mean: -0.1145 sigma: 0.007098 min: -0.125 max: -0.105 range: 0.020
Recv: 6 of 10: z: -0.086 mean: -0.1098 sigma: 0.012362 min: -0.125 max: -0.086 range: 0.039
Recv: 7 of 10: z: -0.070 mean: -0.1041 sigma: 0.018024 min: -0.125 max: -0.070 range: 0.055
Recv: 8 of 10: z: -0.070 mean: -0.0998 sigma: 0.020285 min: -0.125 max: -0.070 range: 0.055
Recv: 9 of 10: z: -0.061 mean: -0.0956 sigma: 0.022647 min: -0.125 max: -0.061 range: 0.064
Recv: 10 of 10: z: -0.056 mean: -0.0916 sigma: 0.024508 min: -0.125 max: -0.056 range: 0.069
Recv: Finished!
Recv: Mean: -0.091625 Min: -0.125 Max: -0.056 Range: 0.069
Recv: Standard Deviation: 0.024508
```

If you stack all 3 of these up there's an interesting pattern....
```
Recv: 1 of 10: z: -0.095 mean: -0.0950 sigma: 0.000000 min: -0.095 max: -0.095 range: 0.000
Recv: 2 of 10: z: -0.088 mean: -0.0913 sigma: 0.003750 min: -0.095 max: -0.088 range: 0.007
Recv: 3 of 10: z: -0.090 mean: -0.0908 sigma: 0.003118 min: -0.095 max: -0.088 range: 0.007
Recv: 4 of 10: z: -0.079 mean: -0.0878 sigma: 0.005888 min: -0.095 max: -0.079 range: 0.016
Recv: 5 of 10: z: -0.075 mean: -0.0853 sigma: 0.007348 min: -0.095 max: -0.075 range: 0.020
Recv: 6 of 10: z: -0.073 mean: -0.0831 sigma: 0.008221 min: -0.095 max: -0.073 range: 0.022
Recv: 7 of 10: z: -0.066 mean: -0.0807 sigma: 0.009633 min: -0.095 max: -0.066 range: 0.029
Recv: 8 of 10: z: -0.085 mean: -0.0813 sigma: 0.009122 min: -0.095 max: -0.066 range: 0.029
Recv: 9 of 10: z: -0.080 mean: -0.0811 sigma: 0.008609 min: -0.095 max: -0.066 range: 0.029
Recv: 10 of 10: z: -0.081 mean: -0.0811 sigma: 0.008167 min: -0.095 max: -0.066 range: 0.029
-
Recv: 1 of 10: z: -0.081 mean: -0.0813 sigma: 0.000000 min: -0.081 max: -0.081 range: 0.000
Recv: 2 of 10: z: -0.081 mean: -0.0813 sigma: 0.000000 min: -0.081 max: -0.081 range: 0.000
Recv: 3 of 10: z: -0.091 mean: -0.0846 sigma: 0.004714 min: -0.091 max: -0.081 range: 0.010
Recv: 4 of 10: z: -0.079 mean: -0.0831 sigma: 0.004801 min: -0.091 max: -0.079 range: 0.013
Recv: 5 of 10: z: -0.083 mean: -0.0830 sigma: 0.004301 min: -0.091 max: -0.079 range: 0.013
Recv: 6 of 10: z: -0.104 mean: -0.0865 sigma: 0.008673 min: -0.104 max: -0.079 range: 0.025
Recv: 7 of 10: z: -0.100 mean: -0.0884 sigma: 0.009323 min: -0.104 max: -0.079 range: 0.025
Recv: 8 of 10: z: -0.105 mean: -0.0905 sigma: 0.010307 min: -0.105 max: -0.079 range: 0.026
Recv: 9 of 10: z: -0.095 mean: -0.0910 sigma: 0.009821 min: -0.105 max: -0.079 range: 0.026
Recv: 10 of 10: z: -0.123 mean: -0.0941 sigma: 0.013277 min: -0.123 max: -0.079 range: 0.044
-
Recv: 1 of 10: z: -0.115 mean: -0.1150 sigma: 0.000000 min: -0.115 max: -0.115 range: 0.000
Recv: 2 of 10: z: -0.125 mean: -0.1200 sigma: 0.005000 min: -0.125 max: -0.115 range: 0.010
Recv: 3 of 10: z: -0.119 mean: -0.1196 sigma: 0.004125 min: -0.125 max: -0.115 range: 0.010
Recv: 4 of 10: z: -0.109 mean: -0.1169 sigma: 0.005896 min: -0.125 max: -0.109 range: 0.016
Recv: 5 of 10: z: -0.105 mean: -0.1145 sigma: 0.007098 min: -0.125 max: -0.105 range: 0.020
Recv: 6 of 10: z: -0.086 mean: -0.1098 sigma: 0.012362 min: -0.125 max: -0.086 range: 0.039
Recv: 7 of 10: z: -0.070 mean: -0.1041 sigma: 0.018024 min: -0.125 max: -0.070 range: 0.055
Recv: 8 of 10: z: -0.070 mean: -0.0998 sigma: 0.020285 min: -0.125 max: -0.070 range: 0.055
Recv: 9 of 10: z: -0.061 mean: -0.0956 sigma: 0.022647 min: -0.125 max: -0.061 range: 0.064
Recv: 10 of 10: z: -0.056 mean: -0.0916 sigma: 0.024508 min: -0.125 max: -0.056 range: 0.069
```

Look at how the z values, point to point, cycle .... and keep in mind that the printer is returning to nearly the exact same point after each series, so it's certainly keeping track of where it is, I'm not losing steps, there's no major mechanical problem, etc.

bgort, 2017-05-09:
A plot of the z values:

[image]

There are trend lines there; if this was some random issue, this wouldn't be (nearly) continuous...

bgort, 2017-05-09:
> What if you reset your board and prior to each M48 test?

You may be on to something there.  I've noticed that the first set of probes after a restart is more accurate than later on.  

Maybe I should scope the power to the BLTouch and the endstop signal and see what that looks like as I move through a series of M48.  I suppose it's possible my regulator is going bad, though it doesn't seem likely.

Roxy-3D, 2017-05-09:
> Maybe I should scope the power to the BLTouch and the endstop signal and see what that looks like as I move through a series of M48. I suppose it's possible my regulator is going bad, though it doesn't seem likely.

My guess is this doesn't show anything...  But it would be good to rule out power as being an issue.

bgort, 2017-05-09:
During a retract...

[image]

My DMM says 4.95V nominal, but the scope says 4.73 with a dip to 4.55 on pin retract.  Maybe it is the power...  Going to try a different supply.

bgort, 2017-05-09:
```
Recv: 1 of 10: z: 0.032 mean: 0.0325 sigma: 0.000000 min: 0.032 max: 0.032 range: 0.000
Recv: 2 of 10: z: 0.040 mean: 0.0362 sigma: 0.003750 min: 0.032 max: 0.040 range: 0.008
Recv: 3 of 10: z: 0.036 mean: 0.0362 sigma: 0.003062 min: 0.032 max: 0.040 range: 0.008
Recv: 4 of 10: z: 0.040 mean: 0.0372 sigma: 0.003109 min: 0.032 max: 0.040 range: 0.008
Recv: 5 of 10: z: 0.054 mean: 0.0405 sigma: 0.007185 min: 0.032 max: 0.054 range: 0.021
Recv: 6 of 10: z: 0.057 mean: 0.0433 sigma: 0.009119 min: 0.032 max: 0.057 range: 0.025
Recv: 7 of 10: z: 0.057 mean: 0.0454 sigma: 0.009791 min: 0.032 max: 0.057 range: 0.025
Recv: 8 of 10: z: 0.072 mean: 0.0487 sigma: 0.012824 min: 0.032 max: 0.072 range: 0.040
Recv: 9 of 10: z: 0.060 mean: 0.0500 sigma: 0.012597 min: 0.032 max: 0.072 range: 0.040
Recv: 10 of 10: z: 0.061 mean: 0.0511 sigma: 0.012418 min: 0.032 max: 0.072 range: 0.040
Recv: Finished!
Recv: Mean: 0.051125 Min: 0.032 Max: 0.072 Range: 0.040
Recv: Standard Deviation: 0.012418

Recv: 1 of 10: z: 0.075 mean: 0.0750 sigma: 0.000000 min: 0.075 max: 0.075 range: 0.000
Recv: 2 of 10: z: 0.061 mean: 0.0681 sigma: 0.006875 min: 0.061 max: 0.075 range: 0.014
Recv: 3 of 10: z: 0.059 mean: 0.0650 sigma: 0.007144 min: 0.059 max: 0.075 range: 0.016
Recv: 4 of 10: z: 0.050 mean: 0.0612 sigma: 0.008970 min: 0.050 max: 0.075 range: 0.025
Recv: 5 of 10: z: 0.045 mean: 0.0580 sigma: 0.010326 min: 0.045 max: 0.075 range: 0.030
Recv: 6 of 10: z: 0.036 mean: 0.0544 sigma: 0.012432 min: 0.036 max: 0.075 range: 0.039
Recv: 7 of 10: z: 0.017 mean: 0.0491 sigma: 0.017291 min: 0.017 max: 0.075 range: 0.058
Recv: 8 of 10: z: 0.014 mean: 0.0447 sigma: 0.019958 min: 0.014 max: 0.075 range: 0.061
Recv: 9 of 10: z: 0.002 mean: 0.0400 sigma: 0.023019 min: 0.002 max: 0.075 range: 0.072
Recv: 10 of 10: z: -0.004 mean: 0.0356 sigma: 0.025478 min: -0.004 max: 0.075 range: 0.079
Recv: Finished!
Recv: Mean: 0.035625 Min: -0.004 Max: 0.075 Range: 0.079
Recv: Standard Deviation: 0.025478
```

Nope..that wasn't it.  No meaningful difference.

Roxy-3D, 2017-05-09:
But the 30 second pause between points usually makes things better?

bgort, 2017-05-09:
No, I don't think so.  I'll try 60, but I don't think the delay is helping.

bgort, 2017-05-09:
60s safe_delay:
```
Recv: 1 of 10: z: -0.006 mean: -0.0063 sigma: 0.000000 min: -0.006 max: -0.006 range: 0.000
Recv: 2 of 10: z: -0.001 mean: -0.0038 sigma: 0.002500 min: -0.006 max: -0.001 range: 0.005
Recv: 3 of 10: z: 0.001 mean: -0.0021 sigma: 0.003118 min: -0.006 max: 0.001 range: 0.007
Recv: 4 of 10: z: -0.004 mean: -0.0025 sigma: 0.002795 min: -0.006 max: 0.001 range: 0.007
Recv: 5 of 10: z: -0.000 mean: -0.0020 sigma: 0.002693 min: -0.006 max: 0.001 range: 0.007
Recv: 6 of 10: z: -0.019 mean: -0.0048 sigma: 0.006709 min: -0.019 max: 0.001 range: 0.020
Recv: 7 of 10: z: -0.011 mean: -0.0057 sigma: 0.006610 min: -0.019 max: 0.001 range: 0.020
Recv: 8 of 10: z: -0.025 mean: -0.0081 sigma: 0.008883 min: -0.025 max: 0.001 range: 0.026
Recv: 9 of 10: z: -0.020 mean: -0.0094 sigma: 0.009169 min: -0.025 max: 0.001 range: 0.026
Recv: 10 of 10: z: -0.023 mean: -0.0108 sigma: 0.009539 min: -0.025 max: 0.001 range: 0.026
Recv: Finished!
Recv: Mean: -0.010750 Min: -0.025 Max: 0.001 Range: 0.026
Recv: Standard Deviation: 0.009539
```

Looked great up until probe 6....

Going to try an old copy of RC7 I have..

Roxy-3D, 2017-05-09:
If the Z-Stepper motors are not losing position.....    It **_HAS_** to be the probe, right????

bgort, 2017-05-09:
It has to be the probe, or Marlin, or something physical/mechanical on the printer that I can't measure or detect.

Here's what we know:
 * It seems to probe reliably when the bed is cold (~5um sigma)
 * It doesn't probe reliably when the bed is warm - it doesn't matter if the heaters are on or off - once the bed is warm, it starts screwing up
 * The bed is pretty flat and generally maintains a stable shape when at a stable temperature (+/- ~2um) according to the dial indicator
 * The first probes in a series of M48 are nearly always reliable; after probes 4, 5, 6, error begins and typically gets progressively worse
 * We've tried two power supplies and everything seems fine there
 * The axis always returns to the same position +/- 1-2um, so the steppers aren't losing steps and Marlin knows where the axis is at all times (otherwise it couldn't return to 0)
 * No older version back to RC7 is any better than the current version (despite having successes in the past?!)
 * ...?

Roxy-3D, 2017-05-09:
And you know....   

Once we do figure this out....  It is going to be something very simple and we aren't going to be able to figure out why it took us so long to figure it out...

bgort, 2017-05-09:
I was thinking the same thing.. it's going to end up being something simple....

Roxy-3D, 2017-05-09:
Also...   This is part of the reason I've become such a big fan of the Z-BabyStepping.    Once a mesh is fine tuned, you are only dependent on the probe accuracy to set the Z height for the print.  If that is off...  You might have Z-BabyStep extra to get the filament squishing the 'right' amount.  But the print is going to work. 

But really...   The print's first layer should be the same every time.  And it isn't.

bgort, 2017-05-09:
The bed is reasonably uniformly heated:
[image]
[image]

bgort, 2017-05-09:
> But really... The print's first layer should be the same every time. And it isn't.

If I could get a decent initial mesh - like I did the first time I tried it - I wouldn't care so much about the accuracy of the probe.  I don't mind simply adjusting the z position once per day.  That's what I was doing before....

I'm about 2 minutes from buying the OptoNCDT 1320 and integrating it, but that'll take me another how many weeks?  Starting to get burnt out.... 😬

Roxy-3D, 2017-05-09:
Well...   I tell you...   I really like servo's that kick down a probe leg with a micro-switch on it.   That is simple and it does work very well.

bgort, 2017-05-09:
As far as I can tell, the BLTouch doesn't meaningfully change temp during M48:
https://youtu.be/qtMSGkNUAFA

bgort, 2017-05-09:
Yeah -- I bought servos and I have what I need to do it, I imagine, but I'm not sure I have the patience for it.

Roxy-3D, 2017-05-09:
> As far as I can tell, the BLTouch doesn't meaningfully change temp during M48:
https://youtu.be/qtMSGkNUAFA

I think the heat bed needs to be turned off to 'see' the BL-Touch.  And very possibly, the plastic cover would have to be opened up to see which part is getting hot doing the probe.

bgort, 2017-05-09:
Yeah, would probably make it a lot easier.  But I'd think if it were getting hot enough to matter it would be obvious.  As far as I know, the only component in there that could (should) generate heat is the solenoid coil.  The rest of the components don't/won't generally run hot.  I believe they're just using a simple passive hall effect sensor and an attiny.

Roxy-3D, 2017-05-09:
Yeah...  But what I was thinking is...  How are they switching the current for the coil?  Is it a transistor that is very close to the Hall effect sensor?   I agree what ever is switching the current for the coil is the only thing that is likely to get warm.   And almost for sure, the AT-Tiny can't drive the coil directly.   So what is driving the coil?  And where is it mounted relative to the sensor stuff?

bgort, 2017-05-09:
I would hope they used a mosfet; if they did there isn't likely any meaningful heat generated, as it wouldn't run in the linear region, but would only be acting as a switch.  The switching time would be relatively low, I'd think, so not much, if any, heating.

I think it's just a short duration pulse that pulls it up and pushes it down.. otherwise it rests against the head or is held in place by a small permanent magnet.

Those are good questions, though.  How they designed it is important...

bgort, 2017-05-09:
Also, would like to say thank you for your help.  It's greatly appreciated.

Roxy-3D, 2017-05-09:
No!   Thank You!

bgort, 2017-05-09:
Small bug:

I started a G26, then canceled it by holding down the encoder button so I could save first.  This is what I got:

```
Send: G26 H204 B45 O0.5 P2 N0.4
Recv: G26 command started.  Waiting for heater(s).

***held down encoder a few seconds after I did the G26

Recv:  T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:0 B@:12 @0:0 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 T:28 /204 B:45 /45 T0:28 /204 T1:28 /0 @:127 B@:12 @0:127 @1:0 

--- snip 300+ lines of this (seriously) ---

B@:16 @0:127 @1:0 T:55 /204 B:45 /45 T0:55 /204 T1:28 /0 @:127 B@:16 @0:127 @1:0 T:55 /204 B:45 /45 T0:55 /204 T1:28 /0 @:127 B@:16 @0:127 @1:0 T:55 /204 B:45 /45 T0:55 /204 T1:28 /0 @:127 B@:16 @0:127 @1:0echo: cold extrusion prevented
Recv: ok
```

Roxy-3D, 2017-05-09:
OK...  But it looks like the nozzle was at 204 C.   Why would the extruder move think it was going to be a 'cold extrusion' ?

bgort, 2017-05-09:
It wasn't there yet.  It was at about 70 by the time I canceled it.  Just a few seconds after it started heating....

Roxy-3D, 2017-05-09:
Oh!  So if you cancel the heating...  It still might try to do an extrude???   Is that what you think happened?

bgort, 2017-05-09:
I canceled the whole thing, I thought, by holding down the encoder button.  It then moved a little, output all of what I pasted, and then was done.  Was kind of weird.  Not sure what it tried to do.

[48 more replies]

closed 2017-05-10 by bgort