MarlinFirmware/Marlin #6650: Z_SAFE_HOMING Not Going to Center of Bed

HomeBrewedCoffee, 2017-05-08:
Hello:
I searched the forums but could not find a direct solution to my problem. 

I am modifying a MakerGear M2 to have auto leveling. 
I have the sensor working correctly and is all end stops and sensors are triggering. 
However when I run G28 the probe does not move to the center of the bed after X and Y are homed. 
The bed and X carriage remain in place like it would as if it were homing a normal switch. 
Can still trigger endstop with probe and run G29 afterwards.

I have attached my Marlin Config and Adv Config files
[configuration_adv_h.txt](https://github.com/MarlinFirmware/Marlin/files/985022/configuration_adv_h.txt)
[configuration_h.txt](https://github.com/MarlinFirmware/Marlin/files/985023/configuration_h.txt)

. Any suggestions would be greatly appreciated.  

Best:

HBC

Roxy-3D, 2017-05-08:
This is a very very old Configuration.h file.    You should bring up NotePad++ and use the visual diff plug-in to map your settings to the new file.    Some of the option names have changes so you are not really turning on things that you think you are.

HomeBrewedCoffee, 2017-05-09:
Thanks! Thought I was using the 1.1.X but somehow grabbed the 1.0.X by mistake.

closed 2017-05-09 by HomeBrewedCoffee