MarlinFirmware/Marlin #6651: Shifting layers

NovemberCo, 2017-05-09:
Hi... i have this problem where after about 6 hours of continious work the printer shift layers a litle bit... but then the next layer goes back to the original position.. Does anyone have any idea what could be the problem???...

I already checked the tension on the belt.. the current of the drivers... the temp on the drivers... i even put a fan to cool down the board.

my last option here is to think that its related to the firmware..

i would apreciate any help :).

Roxy-3D, 2017-05-09:
If you are shifting layers...  Probably your X or Y stepper motor is losing position....    You can turn the power up to it.   You can make sure the axis isn't hitting something or stiff...  Etc.   Etc.

NovemberCo, 2017-05-09:
Hi roxy... the shifted layer is on the X axis... when you mean turn the power up... you are saying from the firmware?? because i already set the trims on the drivers... and in any range... i have the same problem... the axis is not hitting anything... and the rare thing its that the problem appears after many hours of printing... could be the slicer ??

Roxy-3D, 2017-05-09:
> Hi roxy... the shifted layer is on the X axis... when you mean turn the power up... you are saying from the firmware??

No.  If you have step sticks...  You can turn the power up until the motors are uncomfortable to hold in your hand.   It is OK if they are very warm.

>  and the rare thing its that the problem appears after many hours of printing... could be the slicer ??

One of the bad things about RepRap printers is there is not feed back about the motor position.  The machines are homed and the assumption is the motors have help position from then on.

A million things can affect whether they hold position.  For example...  If you have your acceleration too high...  or you have the max step rate too high....   or you have some extra binding at one end because the rails are not perfectly parallel under bed....   But if you can...  turning the power up might compensate for the error.

NovemberCo, 2017-05-09:
you mean the trim next to the chip right?
[image]

Thanks :D!

Roxy-3D, 2017-05-09:
Yes...    Turn it up...  For sure, there will be people that post "Oh! That is a bad thing to do..."   I have always ran all my stepper motors so hot you can't hold your hand on them for 10 or 15 seconds.   They all work perfectly.

NovemberCo, 2017-05-09:
Okey... thanks alot... im going to pump those numbers up... because 0.8v its a rookie number? haha..

thanks again :)

psavva, 2017-05-09:
You should I have active cooling on the Stepper drivers.  A heatsink is also important to stop skipping.

Most cheap A4988's can't even reach their maximum amperage due to wrong resistors used on them..  suggest you look at some nice TMC drivers.  I'll be upgrading to TMC2130's or 2100's when 32bit support is out.

I'm currently using DRV8825 drivers with no complaints.  Just remember to actively cool them to avoid thermal protection shutdown and thus skipping steps and layer shifts.

NovemberCo, 2017-05-09:
I just checked my drivers and they are A4988 :( ..... so... im guessing i will be buying the 8825 ones... 

thanks a lot again. :)

psavva, 2017-05-09:
Try upping the trimpot as suggest by Roxy first.
If they are genuine A4988's, they should have a 50ohm resistors rather than a 200ohm resistor.

But try what Roxy said first...

Roxy-3D, 2017-05-09:
But...  If you can make your different axis move easier....  That should be done too.

NovemberCo, 2017-05-09:
Thank you both... one last thing... i saw that you can adjust the micro steps in the driver??? like 1/8 1/16 1/32... its that correct?? and if it is... how do you set that?

psavva, 2017-05-10:
This is from the geeetech website assuming that you are using a ramps 1.4

Step size of stepper driver (A4988)
  jumper       Yes/No     stepsize     
   ms1     ms2      ms3 
   no      no       no     fullstep
   yes     no       no     halfstep
   no      yes      no     1/4step
   yes     yes      no     1/8step
   yes     yes      yes    1/16step 

Step size of stepper driver (Drv8825)
  jumper   Yes/No       step size
   ms1     ms2      ms3
   no      no       no    full step
   yes     no       no    half step
   no      yes      no    1/4 step
   yes     yes      no    1/4 step
   no      no      yes    1/16 step
  yes      no      yes    1/32 step
  no      yes      yes    1/32step
  yes     yes      yes    1/32step

psavva, 2017-05-10:
You will also need to set your steps per mm in the Firmware.

I highly suggest got read this from top to bottom.
http://marlinfw.org/docs/development/configuration.html

But for his skip to the section "Default Steps per mm".  Please use the Prusa Calculator as suggested in the documentation.

NovemberCo, 2017-05-10:
Thank you so much.... i have already set the voltage to 1.023 volts in the X driver.

Roxy-3D, 2017-05-10:
> i have already set the voltage to 1.023 volts in the X driver.

Your steppers don't know how good they have it...  Mine step sticks have the reference voltage at 1.25 volts.

NovemberCo, 2017-05-10:
what.... 1.25 =O!!! and i thought that 1 volt was going bananas :P

closed 2017-05-11 by NovemberCo