MarlinFirmware/Marlin #6652: Prevent retract un-hop at wrong height not working?

Squid116, 2017-05-09:
I have been running RC8 for a while with no issues and upgraded to 1.1.x when it was released.

This PR seems to be causing me grief, it never unretracts: https://github.com/MarlinFirmware/Marlin/commit/94ffb58da7d85f82c6a48917c2cf739f53f5e846

Even if you do nothing between retract and unretract, it doesn't ever unhop.  My c++ is a little rusty but I think hop_height should be set after the hop is performed, not before:

```
hop_height = current_position[Z_AXIS];
// Pretend current position is lower
current_position[Z_AXIS] -= retract_zlift;
```

should be:

```
// Pretend current position is lower
current_position[Z_AXIS] -= retract_zlift;
hop_height = current_position[Z_AXIS];
```

Roxy-3D, 2017-05-09:
We need to get all bug fixes (and not necessarily all new features) crossed over from BugFix-v1.1.0 to v1.1.0

Right now... you are better off running BugFix-v1.1.0

Squid116, 2017-05-09:
@Roxy-3D the same problem exists in RCBugFix, BugFix-v1.1.x and v1.1.x but it was never in RC - which is what I have been running.  I can't see how any branch including 94ffb58 would work.

Roxy-3D, 2017-05-09:
OK.  Let's change the title and see if we can get some people to weigh in.

FHeilmann, 2017-05-09:
It may be the commit that "broke" it was a result of the discussion I started in #5486
FWIW, my machine hops and unhops flawlessly with this commit.

Squid116, 2017-05-09:
Take this (extreme) example:

```
G28
M207 S8.00 F4800.00 Z10.00
G0 Z0.1
G10
G0 Z0.2
G11

G10
G0 Z0.3
G11
```

I'd expect the nozzle to be atZ=0.3 (the last of the G0 commands), however mine is ~20mm above the bed.

FHeilmann, 2017-05-09:
yes, I see the problem, slicers with activated "retract between layer changes" will cause problems here.

Squid116, 2017-05-09:
I just delete the  `&& hop_height == current_position[Z_AXIS]` and everything is back working as expected.

The only way I can see this being an issue is if you LOWER z while retracted.
```
G28
M207 S8.00 F4800.00 Z10.00
G0 Z10
G10
G0 Z3 //lower than before the retraction
G11 //crash!

G10
G0 Z0.3
G11
```

FHeilmann, 2017-05-09:
Yes, I agree, but instead of deleting it I'd suggest changing  ```&& hop_height == current_position[Z_AXIS]``` to ```&& hop_height <= current_position[Z_AXIS]```

which should solve your problem and keep the nozzle from crashing into the bed.

Squid116, 2017-05-09:
@FHeilmann Agreed that would be a safe solution.

FWIW Even with retract on layer change disabled, Cura can be in a retracted move when the layer changes.  (it isn't retracting because of the layer change, but happens to change layer during a retraction).

FHeilmann, 2017-05-09:
See #6656

VanessaE, 2017-05-09:
Well that would certainly explain why my Z offset went all wonky after enabling FW retract today (because I enabled Z hop also).  :stuck_out_tongue_closed_eyes:

Squid116, 2017-09-12:
Just referenced on https://github.com/MarlinFirmware/Marlin/commit/fee696db5d293d9a93997e25facc6448f6ee4abf

@thinkyhead This has been reverted and now I see the problems again.

As an aside - has unit/integration testing been considered for v2?  It really helps to be solve problems like this, I have plenty of experience (albeit not arduino/c++) in the field, I'd be happy to help out.

thinkyhead, 2017-11-27:
A lot of reform was done to firmware retraction, so these issues should be sorted out now in the bugfix branches.

> has unit/integration testing been considered for v2? It really helps to be solve problems like this, I have plenty of experience (albeit not arduino/c++) in the field, I'd be happy to help out.

It's a pretty motley crew around here, most of whom hardly even understand git, and for the most part they don't seem too interested in adopting formal procedures. So, what are you proposing?

thinkyhead, 2018-02-06:
@Squid116 We're increasingly interested in being able to make a build of Marlin that includes unit tests to make sure all functions are returning sane values. But, as you can imagine, this is not as straightforward in an embedded universe as it is in a "modern application." I'd be interested to see what unit tests would look like when applied in this milieu.

closed 2018-02-06 by thinkyhead

Squid116, 2018-02-06:
@thinkyhead I'll see if I can put together a PR with some thoughts, covering (part of) a feature.  There are definitely challenges in the way it is written being embedded, but there are ways around that, namely mocking frameworks.  Unit testing purists will be up in arms and suggest that the design is wrong and you should never need to mock -  but in a more pragmatic approach a mocked test is better than no test.

My c++ is rusty, so it might take me a while get something together but I'll see what I can do.  I might create a separate issue about choosing a mocking framework? as that is a decision that everyone would need to be comfortable with.

thinkyhead, 2018-02-07:
For AVR-only, were you aware that Ultimaker made a desktop environment where Marlin can be embedded and run? https://community.ultimaker.com/topic/10523-ultimaker-simulator-getting-started/ I'm not sure how useful it would be. I assume we want to exercise all the HAL code as much as we can too.