MarlinFirmware/Marlin #6653: M105 on LCD, and message stating Paused for user

pmbroth, 2017-05-09:
Hello Uploaded the latest release 11x as of 5/8/17.  When doing a UBL mesh build, I get at the message on my lcd M105.   

On my comm screen on my pc I get "Paused for User".   Until I press the toggle button on the lcd, it keeps repeating  "Pause for User".  After I press the button  I can then save the mesh.  I did not get this with the firmware 3 days ago, and did not know if you added this function.

pmbroth, 2017-05-09:
[Marlin 5817.zip](https://github.com/MarlinFirmware/Marlin/files/985242/Marlin.5817.zip)

pmbroth, 2017-05-09:
I have attached my Marlin files just in case you wanted to see them, if the error is not normal.
Thank you

Brian

Tannoo, 2017-05-09:
What was you 'mesh build' command?

pmbroth, 2017-05-09:
Hello

G29 P1

pmbroth, 2017-05-12:
I noticed the only time I'm getting this issue, is f I use PROINTERFACE as the software for doing the G29 P1, S3D does not have this issue.

closed 2018-02-06 by thinkyhead