MarlinFirmware/Marlin #6655: Checksum of serial-command containing asterisks is incorrectly computed

deepthought-64, 2017-05-09:
I was sending an serial-command to the firmware that contained and asterisk as part of the command. It was expected to execute the command but it failed with an checksum error.

It seems like the firmware uses the first occurrence of an asterisk as the marker for the checksum instead of the last one.
This leads to incorrect checksum calculation when the command itself contains an asterisk. 

Example:
When 
`N1185 M117 *** Test ****15` (show '*** Test ***' on LCD)
is send to the printer, the firmware responds with 
`Error:checksum mismatch, Last Line: 1184`
although the checksum is okay. 

This could be fixed quite easily by looking for the _last_ occurence of an asterisk and then iterating up to this position in the string when calculating the checksum. (See lines 1053 and 1073 in Marlin_main.cpp)

My Version: Marlin V1.0.2 (in form of the current Prusa3D firmware) _Before anybody shouts at me: this issue still exists in the source of the current version_
Printer: original Prusa i3 MK2

foosel, 2017-05-09:
To add on to this: [This line](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1053)

    char* apos = strchr(command, '*');

should probably rather use `strrchr` instead of `strchr`.

deepthought-64, 2017-05-09:
Also [this line](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1073) should iterate up to the _last_ asterisk (not the first one).

thinkyhead, 2017-11-01:
Marlin has always thrown an error if you include an asterisk in a G-code command that is sent without a line number, so asterisk is now essentially a reserved character. This can be altered by first removing that error check. Then Marlin can be made to only look for a checksum at all if there is a line number. I will apply this change shortly. Thanks!

closed 2017-11-01 by thinkyhead