MarlinFirmware/Marlin #6659: Suggestin in BED LEVEL

batata004, 2017-05-09:
Hi,

I borrowed an inductive sensor to use with my reprap printer and with the new firmware it works amazing well! Really good, I just have 2 suggestion:

1) when I go in the LCD to bed level, I have to go to PREPARE -> BED LEVEL -> BED LEVEL. Do I really need to click twice to bed the level? Cant I just click the first time and it starts leveling?

2) I decided to remove the BED LEVELING gcode from my prints, I usually turn on my printer and immediatlly after that I do a BED LEVELING. Right after I click BED LEVELING -> BED LEVELING I click PRINT FROM SD and already choose what I wanna print. Marlin is smart, it first starts the bed leveling and only after that it starts printing. But I have a suggeston: after I click bed leveling I click PRINT FROM SD and the topmost codes of my SD are to preheat the BED and nozzle. I need to wait the bed leveling to only after that Marlin start to heat my bed and nozzle. So my suggestion is this: when bed leveling, if the user requests the printer to pre heat ORif the user tells the printer to start printing, it should execute all the gcodes related to temperature while it's bed leveling. Any other gcode related to movements it should not execute, if it finds any GCODE that is not a TEMPERATURE RELATED GCODE, it stops and do nothing more till the bed leveling is complete

I like to bed level when I turn on my printer cause many times I print dozens of parts and I dont want to make a bed leveling before eeveery single print. I decided it's much better doing only one bed leveling at the start and after that my gcode only executed a HOMING and starts printing.

Do you think it would be possible to have this in the next releases?

One final consideration: when I start printing and the printer indeed starts printing, if I click the STOP PRINT button, the LCD gets all blank, nothing shows in it. I need to click the button in the LCD so the LCD starts showing something. This never happend with the release of marlin before the latest one.

Tannoo, 2017-05-10:
On the temperature thing..

If you use M104 and M140 to set the temps, you can do whatever while they heat up.

If you are using M109 and M190, all stops waiting for temps to reach the setpoint.

Tannoo, 2017-05-10:
Example:

M140 S50
M104 195
G28
G29
M190 S50
M109 S195
// Any other gcode from here on..

Roxy-3D, 2017-05-10:
> I like to bed level when I turn on my printer cause many times I print dozens of parts and I dont want to make a bed leveling before eeveery single print. I decided it's much better doing only one bed leveling at the start and after that my gcode only executed a HOMING and starts printing.

In the RepRap world... People can use what ever hardware, what ever software, and do pretty much what they want...   But taking the glass off the bed, removing the print and putting back on is a pretty fair disruption the machine.   Me????   I'm much more comfortable letting the machine re-home and start the process over again.   Even with a Mesh Bed Leveling system...   I like Z-BabyStepping while the skirt is printing to get everything perfect before I walk away.

akaJes, 2017-05-10:
Better spend one minute before printing, but not throw print result in trash after an hour

closed 2018-02-06 by thinkyhead