MarlinFirmware/Marlin #6660: Git Branches

apelly, 2017-05-09:
I have failed to find any docco on Marlin branches.

I'm confident I've seen `RCBugFix` discussed as what might be called `master` in other repos, but while I was fooling around yesterday rebasing my local repo I noticed it was behind `1.1.x` and well behind `bugfix-1.1.x`

Is the project policy on this documented somewhere?

Tannoo, 2017-05-09:
Yes, the readme.

apelly, 2017-05-09:
> Yes, the readme.

Which is different in every branch.

`bugfix-1.1.x` says
> This branch is 53 commits ahead, 23 commits behind 1.1.x. 

Which doesn't gel with the readme:
> Proposed patches should be submitted as a Pull Request against this branch (bugfix-1.1.x).

>    This branch is for fixing bugs and integrating any new features for the duration of the Marlin 1.1.x life-cycle. We've opted for a simplified branch structure while we work on the maintainability and encapsulation of code modules. Version 1.2 and beyond should improve on separation of bug fixes and cutting-edge development.

Tannoo, 2017-05-10:
Yes. Cryptic, I know. It's like there's a hidden message in there somewhere.

Also, the `bugfix1.1` was a little of a shock at release time. Patches and features were still being merged into RCBugFix when that's wasn't where they were supposed to go anymore.

Things will settle out, people will conform. Resstance is futile.

apelly, 2017-05-10:
Understood. Thanks mate.

closed 2017-05-10 by apelly