MarlinFirmware/Marlin #6663: Bug with Marlin using G1

batata004, 2017-05-09:
Hi,

I created a file and the only line inside it is:

G1 X0 Y0

After that I pluged the SD card in the printer, selected the option PREPARE -> BED LEVEL -> BED LEVEL and waited it finished (I have an inductive sendor). After that I selected the option PRINT FROM SD and selected the file with the single line above.

Right after I do that the printer starts the AUTO BED LEVELING (I have no idea why) and after that nothing happens, I mean, the printer does not move itself to the 0,0 position!

batata004, 2017-05-10:
If I just turn on the printer (and do nothing else) and open PRONTERFACE and paste the code:

G1 X0 Y0

The printer will do nothing (cause I think it believes it is in the right stop). If I send this code:

G1 X10 Y0

The X axis move 10 mm to the right. So everything looks right.

Please use the newest marlin firmware and use the config files below (it makes use of inductive sensor).
[Configuration_adv (5).txt](https://github.com/MarlinFirmware/Marlin/files/989975/Configuration_adv.5.txt)
[Configuration (4).txt](https://github.com/MarlinFirmware/Marlin/files/989976/Configuration.4.txt)

Turn on your printer. If you say to the printer to print the GCODE file right now it looks like it freezes, nothing happens. The printer shows in the display the name of the GCODE file and nothing more happens. Try turning on your printer and right after that tell it to print a GCODE file with this content:

G1 X10 Y0

You will see nothing happens.

Now do this: turn on your printer go to PREPARE -> AUTO HOME. Wait till it finishes. NOW TRY to prin the CODE file with this content:

G1 X10 Y0

For some reason the printer will AGAIN try to AUTO HOME all axis and after that nothing happens. It's nonsense trying to home again since I had already said the printer to AUTO HOME before printing the file. And finally, the bug part is also because the printer is not moving to the right 10 mm when I select PRINT FROM SD (using the gcode file containing G1 X10 Y0). It sits there doing nothing.

bgort, 2017-05-10:
Sounds like maybe two issues.

Either way, I will try this out later today, but a quick question:  is there a newline at the end of the `G1 X10 Y0` in the test gcode file?

batata004, 2017-05-10:
Thanks man. No, I doubled check the file and there is no newline at the end. Should it exist?

bgort, 2017-05-10:
Yes -- my guess is that the input parser isn't seeing the end of the line, so doesn't actually process it the gcode because it's incomplete.  

Add a newline and try again.  Let me know if that solves the issue.

batata004, 2017-05-10:
I just tried, it's even weird! I attached the 2 files, one with the newline at the end and another without.
[weird.txt](https://github.com/MarlinFirmware/Marlin/files/990131/weird.txt)
[nothing.txt](https://github.com/MarlinFirmware/Marlin/files/990130/nothing.txt)

The nothing files, as it says, nothing happens. The weird file is really weird, after homing the printer moves verry slowly to a spot in the left (triggering the X endstop) and to the front of the heated bed BUT without triggering the Y endstop.

I think you will have to see this weirdness by yourself, it's really strange.

bgort, 2017-05-10:
Try adding a feedrate at the end of the command (`F2000` or whatever is appropriate for your printer).  That will make it go faster.  Perhaps your default feedrate is low?

So you're saying it's reversing the X and Y coordinates?

batata004, 2017-05-10:
@bgort why are you telling me to to use F20000? My gcode file has only one single line, nothing else. Why adding this line would make solving this bug easier?

I am not saying X and Y are reversed, but if you try to import the 2 gcodes files in an SD card and try to run it on your printer you will see the problem I am repoting. I hav no idea what is going on, really weird.

bgort, 2017-05-10:
No, it's not a line by itself.  I was suggesting adding `F2000` to the end of the `G1` line because you said it moved slowly; I was mostly pointing out that the speed it moves is a function of the feedrate set in the gcode, or the default feedrate if no feedrate has been used previously.

Maybe try adding a `G90` or `G91`, according to your preference, to the beginning of the file (before the `G1`) so you can be sure that it's in the mode you want it to be in?

batata004, 2017-05-10:
Ahhh! I got it, F is related to speed! I did it I set F1000 (to be faster) but nothing happens in the first file and the second moved faster but the behaviour is completely weird. If you have the chance to save those 2 files in your SD and try to print it I am sure you will see what I am saying!

OBS: g90 and g91 wouldnt affect too much cuse X10 is 10 and it shold move at least a little but in the first gcode file it never moves.

bgort, 2017-05-10:
Is it still autohoming twice?

batata004, 2017-05-10:
No, for some reasons it's not autohoming anymore. It just does nothing OR it moves randomly to the left.

Tannoo, 2017-05-11:
Try a copy of `bugfix-1.1.x` from today. There have been lots of changes already.
Start with minimal settings to get your printer moving.

I say this because most of us do not have this issue for something so basic.

bgort, 2017-05-11:
> I say this because most of us do not have this issue for something so basic.

Yes, this is why I'm confused.  There should be no problem with basic movement.  There are enough people using the latest version (including me) that anything like this would have been uncovered immediately.  This almost has to be a configuration issue.

Tannoo, 2017-05-11:
His configs looked ok to me.

Driver wires crossed somewhere?

bgort, 2017-05-11:
Maybe, but I wonder if he's having trouble with more complex gcode files?  From his other recently closed issues I believe he may have okay movement, so just not sure what could be going on here.

batata004, 2017-05-11:
My friends, I really cant believe you are not having the problems I am having ;( Did any of you tried copying the 2 files I attached to you to a SD card and tried to print it? Please, any of you tried? Cause it's very clear this bug, very clear. You just need to do a simple operation, no code skills requiered, copy the filess to SD card and try to "print it". You will see the bug right away!

bgort, 2017-05-11:
@batata004 My SD card reader is disabled.  I will try it sometime soon.

Also, there's no need to keep trying the first file (the one without the newline), as the file shouldn't work unless there's a newline at the end of each line.  Without it, it's incomplete.  I can't imagine anyone is going to make a change to allow a partial line to be parsed and processed..

In any case, try this:  Restart the printer and try -only- the file with the newline.  If you're trying them one after the other, it's possible the one that is incomplete is somehow affecting the complete one (though I'm not sure how the string buffer works, as I've never worked with that part of the code).

batata004, 2017-05-11:
@bgort I tried one each time I restarted the printer, I thought it could be related to one final after another leveaing some problem behind, but anyway, the "weird.txt" file does not work as it should.

One important thing: in Windows the newline characters is always represented by "\r\n" (I know this cause I am a C++ and PHP programmer and have do deal with thi). Mac and Linux make the new line be represented by "\n".. That's the default, you can have \n in Windows (using dreameaver there is a special setting to this).

I am using Windows and my newline default is \r\n. Do you have any idea if this could be causing problems?

Tannoo, 2017-05-11:
Nothing weird on my end.
The nothing file does just that, nothing. Due to no new line.

The weird file does what it is supposed to do. X moves to 10.
That's it.

No funny business here.

bgort, 2017-05-11:
> I am using Windows and my newline default is \r\n. Do you have any idea if this could be causing problems?

No, it shouldn't matter.

Tannoo, 2017-05-11:
I did not change the files other than the extension to .gcode.

batata004, 2017-05-11:
@Tannoo  in what speed did it move? Mine moved slowly, than a little faster and at the end the Y moved. Sometimes it's faster sometimes not.

bgort, 2017-05-11:
What, exactly, is the issue again?  Sorry to ask you to repeat, but I just want to make sure it's clear.

BTW, the speed can change, depending on your configured acceleration and default speed settings.

Tannoo, 2017-05-11:
I am running bufix-1.1.x from this morning. (with additional code by me, of course).

Tannoo, 2017-05-11:
I can't run @batata004's config files without changing a few things.

Tannoo, 2017-05-11:
X moved the default slow rate due to no speed command.

When it got to 10mm, everything stopped and it said print complete.

batata004, 2017-05-11:
Ok, I know the speed can change but not randomly right?

@Tannoo what errors are you gettins when trying to use my config files? I can use they perfectly fine (with the newest marlin version which changed lots of things inide the config files). I am using a pretty cheap printer with nothing fancy, the only thing that I chnge wa removing the servo and replacing it by an inductive sensor that this community helped to configure

Tannoo, 2017-05-11:
It got to the 10mm in about q sec.

batata004, 2017-05-11:
EDIT: the speed can changen randonly if my start positiong is further/closer to the target position? I think tihs could explain the random speed! I just realized now that if I am too far away the speed is faster and if I am close to the final point the speed is a lot slower! Could this make diffrence? If this is expected behaviour so I think this problem is solved! But I am not sure if this is expected

bgort, 2017-05-11:
Yes.  The acceleration takes time, so if you're closer you won't achieve the same speed as if further away.

Tannoo, 2017-05-11:
I am running nothing special, 12v prox for probe, and full graphic discount lcd. I have other code changes to support all kinds of leds and rgb strip (addressable and non). 

Your stepper feedrates are different than mine. Mine are also different than default.

My endstop setting are different than yours and stock.

Tannoo, 2017-05-11:
My printer is also cheap (~$300) from FolgerTech.

Tannoo, 2017-05-11:
I've replaced the Mega once due to a short with a previous prox probe that blew out some inputs. (12v on the 5v rail was not good)

batata004, 2017-05-11:
My printer, converting to dolars costed me 150 dolars (I paid in real around 600). It's really cheap ;) But as @bgort said, if it's normal to the speed to change if you are further away from the target point and you dont specificy and speed (I am not saying any speed in my code) than this random behaviour that I am facing maybe is expected. If you agree, can I close this?

Tannoo, 2017-05-11:
Acceleration time....if there's not enough time to get to full speed, then it won't reach the given speed.
Yes, I agree.

Tannoo, 2017-05-11:
Given speed can be the default, or a specified speed with F.

Tannoo, 2017-05-11:
If there was no default speed, it wouldn't move at all.

batata004, 2017-05-11:
Yeap but in my small understanding I think that if you explicitly use F parameter then the printer will use that speed and nothing more, no matter if the target is away or close. If you dont use F I think that Marlin does some sort of weird calculation in order to arrive at the specific spot in a certain fixed amount of time. I think the default is the time it will take not the speed, dont you think?

Tannoo, 2017-05-11:
You stated that Y was moving also, but Y0 was given, so, if you are homed, Y shouldn't move. BUT, if your Y is somewhere else, it will move to Y0 at the same time that X was moving to 10.

If you included a Z movement, Z moves slower than X or Y. All axis will move at the same time with a G0 or G1 command. And all will move at the same speed of Z. Even if no F was given.

Tannoo, 2017-05-11:
Well, you'd have to get @thinkyhead to say for sure. But, it boils down to a federate to move the steppers.
If no federate is given, there is a default 'safe' speed to run.

Tannoo, 2017-05-11:
Not all machines can handle an 8000 mm/s feedrate, so something slower, like 1000 mm/s should be supported by most, if not all.

bgort, 2017-05-11:
Also, consider that it remembers the last feedrate given, so you can do F1000, and then the next command without F and it'll still move at 1000.  It stays at 1000 until a new F is specified.

Tannoo, 2017-05-11:
Yes, I see that, but there are some weird traveling rates going on now that weren't there previously.

When I tell my printer `G1 X100 Y100 F8000`, it almost looks confused. But, it gets there with quick pauses along the way.

Tannoo, 2017-05-11:
Ah, I found out why. UBL was active. I disabled it and it's a smooth move at the given feedrate to the destination.

bgort, 2017-05-11:
Yes.. it's UBL doing it.  It's calculating along the way, keeping the nozzle at the right height.

Tannoo, 2017-05-11:
Yeah, but when the nozzle is 10mm above the bed, it's shouldn't care. lol

Tannoo, 2017-05-11:
Maybe it's still in the range of the fade height.

bgort, 2017-05-11:
I agree.. that's what I was just going to suggest.  Try 20mm.   After the fade height it shouldn't matter.

Tannoo, 2017-05-11:
Well, I've gone up to 50mm and it's still moving Z when traveling anywhere.

No matter, it's leveling all the time.. lol

bgort, 2017-05-11:
Haha.. but not sure that's desired behavior...

Tannoo, 2017-05-11:
I've never set the fade height. The G29 W command revealed this:

`planner.z_fade_height : 0.0000`

So, default is 0?

bgort, 2017-05-11:
I thought it was 10.. hmm.

batata004, 2017-05-11:
@Tannoo go to know you find out that weird traveling rates, I thought I was the only one seeing that. Anyway, it does not bother me, it does not cause any problem except that it's not consistent.

Tannoo, 2017-05-11:
Disable bed leveling and see if you have the same issue.

Tannoo, 2017-05-11:
To really see it, I was moving across the bed. Not just 10mm. I was moving 170.

Tannoo, 2017-05-11:
You can close this issue.

closed 2017-05-11 by batata004