MarlinFirmware/Marlin #6664: [another] but with marlin, X Y Z always blinking

batata004, 2017-05-09:
Hi,

I am using the latest marlin firmware and even after homing, auto bed leveling and even after start printing, the X Y and Z characeteres remain blinking in the LCD. I am using inductive sensor. Confi files attached.

This does not bother me and it does not look to cause any prblem so far, but I think if you be checked cause in the future if could bring some problem.
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/988204/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/988205/Configuration.txt)

Tannoo, 2017-05-10:
I just tested this and have no issues.

batata004, 2017-05-10:
So let me help you do the test.

1) please use the conf files I attached and upload them with the newest firmware of marlin to your printer;

2) turn on your printer go to PREPARE -> BED LEVEL -> BED LEVEL. Wait the bed level procedure. After the procedure is over the X Y and Z letters will stop blinking (great, that was supposed to happen);

3) GO PRINT FROM THE SD and select the file I am attaching you (actually I think you can use any other file, but if you wanna give it a try please use the *.gcode file attached).

4) first it will heat the bed to 60 C degres. Then it will HOME all axis;

5) NOW YOU CAN SEE THE BUG. After it home all the AXIS it immediatly starts bliking X Y and Z. 

6) After that my GCODE tells the printer to move to the final temperature of 70C/238C.

I am pretty sure you will be able to reproduce this!

Tannoo, 2017-05-10:
In that file...
```
G90
M83
M106 S0
M104 S0;
M190 S60;
M140 S70;
M104 S238;
G28;  <-- Homing. XYZ stop flashing after this is done.
;G29;
M84;  <-- Disable Steppers.... This also negates any homing and will cause the XYZ to flash.
M190 S70;
M109 S238;
```

Tannoo, 2017-05-10:
Remove the M84 and see if that helps you.

Tannoo, 2017-05-10:
The steppers are re-enabled once any movement is asked of them. But 'home' is not remembered.
So XYZ will continue to flash regardless of any printing movements.

batata004, 2017-05-10:
I am trying it right now, just waiting the heated bed to warm up.

batata004, 2017-05-10:
I CAN CONFIRM THAT! Commenting M84 line works out perfectly! So my questino is: how do I turn of the motor while the print is heating? It's useeless keep the motors powered on while nothing is happening, right?

Tannoo, 2017-05-10:
If keeping the motors on while heating bothers you, then heat the bed and hotend BEFORE homing. Not after.

Tannoo, 2017-05-10:
You were heating, homing, then heating again.

Just heat, then home, level and print. Viola!

batata004, 2017-05-10:
@Tannoo  I am doing the way I am doing cause it's more efficient.

Instead of waiting to everyting warm up AND ONLY after that start homing/auto leveling I decided to do this:

1) warm up the bed close to the final temperature cause it usually takes  long time;
2) when it's close to the final temperature start heating the hot end and while that start the HOMING/AUTO LEVELING procedure which usually also takes some time, at least 1 minute;
3) when heated bed reaches temperature (usually it's the last thing that happens) start printing!

You see, I save some time homing/autoleveling while temperature is getting there!

Tannoo, 2017-05-10:
That's ok. But remember that the steppers will timeout after 60 secs.
I think there is an option to disable the stepper timeout.

batata004, 2017-05-10:
So you are saying that even if I comment the line M84 I can have this kind of issue again if my heated bed takes too long to reach the final temperature? Even if I dont disable the motors?

Tannoo, 2017-05-10:
Correct.

batata004, 2017-05-10:
Too bad for me :( Do you think there is any chance that the dev team can make the DISABLE MOTORS GCODE not forgeting the position? If you think this is not possible I will close this thread and have to live with it!

Tannoo, 2017-05-10:
I looked around and found that it's only for the filament change option and only during an M600:
`FILAMENT_CHANGE_NO_STEPPER_TIMEOUT`

batata004, 2017-05-10:
Thanks for searching a solution but as I see I will have to deal with this. Anyway, i appreciate your attention, at least now I solvd this problem that was driving me crazy! Really thank you man!

closed 2017-05-10 by batata004

Tannoo, 2017-05-10:
I did find this in `configuration_adv.h`:
```
// Default stepper release if idle. Set to 0 to deactivate.
// Steppers will shut down DEFAULT_STEPPER_DEACTIVE_TIME seconds after the last move when DISABLE_INACTIVE_? is true.
// Time can be set by M18 and M84.
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // set to false if the nozzle will fall down on your printed part when print has finished.
#define DISABLE_INACTIVE_E true
```
You can deactivate all steppers, OR just simply change the time from the 120 to however long you think you need.

batata004, 2017-05-10:
@Tannoo  GREAT GREAT! I raised that to 600, it never takes that long to heat but with 600 I am pretty sure I will not have any problem! Thanks man, really thanks!

Tannoo, 2017-05-10:
Wow, yeah. If so, yould have another issue.