MarlinFirmware/Marlin #6667: auto level timeout

trivalik, 2017-05-10:
It comes to my mind when leveling with AUTO_BED_LEVELING_BILINEAR the Z0 height is known when measured with inductive/capacitiv sensor. So it would be possible to add a timeout for the other probes. Because I have sometimes that when I put to much ABS to improve stick, the probe is not working anymore, because my capacitive sensor is not switch anymore.

What do you think?

Tannoo, 2017-05-10:
Sometimes my truck goes past the speed limit if I leave my foot on the gas for too long....

Don't put too much ABS to improve stick.

trivalik, 2017-05-10:
I have another idea, I see when bed leveling start, Z is on display -39,99 but I cannot find this value somehow in code. When this way this configurable, then is it like a timeout.
Any hint for me?

Tannoo, 2017-05-10:
It's like a timeout... Tries to find home by moving commanding a large distance that it should not reach before a limit.

trivalik, 2017-05-10:
I believe it is somehow this code:

https://github.com/MarlinFirmware/Marlin/blob/99d18ecd0cc9be27cfc32f37cc7d1b476b4139a1/Marlin/planner.cpp#L624

```
      #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
        const float c = bilinear_z_offset(logical);
        logical[Z_AXIS] = (z_fade_height * (RAW_Z_POSITION(logical[Z_AXIS]) - c)) / (z_fade_height - c);
      #else
        logical[Z_AXIS] -= bilinear_z_offset(logical);
      #endif
```

drkztan, 2017-05-12:
I agree, there should be a time out. I don't use any kind of surface adhesion additive, I print directly on my aluminum heatbed and sometimes the probe just fails to read a value, crashing the nozzle into the bed. Luckily, nothing bad has happened (yet), but it feels like the probe should stop descending and abort the leveling process if it goes something like 2mm below the G28 zmin value.

boelle, 2019-02-18:
@trivalik is this issue still relavant with latest bugfix 2.0?

boelle, 2019-03-12:
@thinkyhead i think we can close this one

trivalik, 2019-03-12:
This ticket can be close in case a timeout is existing.

boelle, 2019-03-12:
have you tried with latest bugfix 2.0? is the problem gone?

trivalik, 2019-03-12:
No, I haven't. Since my time with 3D printer is getting less. At the moment is no need for, but a timeout is always a good idea, this say my knowledge in machine programming.

boelle, 2019-03-12:
like if it does not find the bed within X number of sec it should abort?

trivalik, 2019-03-14:
Yes, that's the idea behind this timeout.

boelle, 2019-03-14:
as it is now it will abort probing if it does not hit the bed and has figured that the probe will now be under the bed

like if it thinks the probe is 2mm under the bed but has not triggered it will abort

same effect as you want just done in a different way

trivalik, 2019-03-14:
But when the capacity sensor is not switching, the the z axis drives to minus until death, I did never saw a stop when this happen to my printer, it looks very terrible, so I was afraid not to be next to the printer, that I in case could unplug power cable or press reset button. What you mean is the difference between two probe points, this works only if the z pin detect something.

boelle, 2019-03-14:
i have a probe of that type too and it does stop if it does not trigger

trivalik, 2019-03-19:
In case the z axis is 20cm above bed. It will drove 20 cm. So for me, so where is the timeout like behavior?

boelle, 2019-06-22:
think we should close

InsanityAutomation, 2019-06-22:
To confirm and apply correct information, there is a z probe minimum distance now. By default it will abort probing at Z-2mm. I believe this covers the original request here so this should be closed.

See #define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping

trivalik, 2019-06-24:
Z_PROBE_LOW_POINT helps only if one Z could already be measured, but this is not the case with a probing sensor for the find of Z at all.

BigIronGuru, 2019-06-24:
@trivalik 
**Probing** (e.g. G29) will go down to Z_PROBE_LOW_POINT if Z axis is "homed / known". It will go down to -10 if "not homed / unknown ".
**Homing** (e.g. G28) will go down "very far". Of course, because it could be coming from very high up, who knows? Homing will also be quite brutal, first pass is speedy. 

That's why some people put an unreliable PROBE on a separate pin for PROBING (which is more careful and has many checks) and use a reliable switch for HOMING (which is difficult to do carefully because coming from an unknown height).
Of course one could program HOMING to be more "careful", but then it would take ages.

So yes, I think this can be closed.

trivalik, 2019-06-25:
Good point, to have an additional sensor to get an around value for Z. This sensor must then work 100%. Thanks for good explanation, I am not able to.

BigIronGuru, 2019-06-25:
Please close this if you are satisfied

trivalik, 2019-06-26:
A timeout would be anyway good :-)

closed 2019-06-26 by trivalik