MarlinFirmware/Marlin #6669: Another freaking self destructing bug with marlin

batata004, 2017-05-10:
Hi,

This is another bug but by far is the most dangerous and I think because of it you should fix is as soon as possible or remove this latest firmware from the GIT and place the older back.

This bug is crazy, when I heard the strange sound I runned to see my printer and it was screaming and almost destroyed itself.

Lets make you reproduce this make step by step:

1) use the latest marlin firmware;
2) use these configs files, they are pretty default to the latest original marlin configuration, I just changed a little so it would support an inductive sensor
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/990051/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/990052/Configuration_adv.txt)
3) save this GCODE file in your SD (generated in SIMPLIFY3D).  
[gcode.txt](https://github.com/MarlinFirmware/Marlin/files/990059/gcode.txt)
4) Send the firmare to the printer;
5) turn on the printer -> PREPARE -> LEVEL BED -> LEVEL BED
6) after it finishing leveling the bed go to PRINT FILE FROM SD and select the GCODE I attached earlier;

Wait the printer heat up, wait it home again in the middle of the hitting. After it home again I tell the printer "G1 X-1000 Y-1000", which means "move to the left most and front most position" before start printing. THIS WORKS PERFECTLY. The printer moves to this position and of course it hits the endstops as it should and nothing dangerous happens.

So I wait the printer to finishes heating up, and when it starts printing it goes faaar to the right and to back and of course it hits the limit in the right/back and keeps trying to go, almost destryoing my printer.

I think there are 2 flaws:

1) when the printer hits the endstop it should keep the last position that it hitted the entstop as the current position instead of using the -1000 (told in the GCODE G1 command);
2) if the printer is trying to go below the print area (which is very small in my config files, way small than 200mm in X and 200m in Y) it should stop moving!

bgort, 2017-05-10:
This may be related to https://github.com/MarlinFirmware/Marlin/issues/6581 and/or https://github.com/MarlinFirmware/Marlin/issues/6646 (pretty sure they're the same issue) ... which are being worked on.

Was there anything in the serial console when this happened?

bgort, 2017-05-10:
And what kind of bed leveling are you using?

batata004, 2017-05-10:
In the serial console? I didtn use computer, I just used USB. I can also confirm this bug is not related to the -1000 number nor to it hitting the endstop! I just replaced this:

G1 X-1000 Y-1000

with

G1 X-60 Y-60

So the endstop does not get triggered AND if I DO THAT when the printer starts printing it goes waaaay to the right/back and starts printing there (out of the heated bed)!!!

So if you are wondering if this bug is related somehow to a very low number (-1000) or hitting the endstop, you can disccard this assumption, the bug happens even if you tell the printer to move just a little bit to the left/front before it starts printing.

I am using inductive sensor and it works perfectly, I can print with no problems with it. The only problems I am facing is when I try to use some very specfici GCODES inside the printing file.

bgort, 2017-05-10:
Are you in relative or absolute mode when you enter the coordinates (-1000 or -60) ?

batata004, 2017-05-10:
My friend @bgort  I am using just GCODE file. I am a totally beginner, I am not using USB, I am just using this GCODE FILE -> https://github.com/MarlinFirmware/Marlin/files/990059/gcode.txt

If you see the file I believe you can check if it's absolute or relative position, I have no idea how to know that. Could you please help me discovering if it's absolute or relative and if it would change anything?

bgort, 2017-05-10:
You're in absolute positioning mode.  For future reference, `G90` puts the printer in absolute mode, while `G91` puts it in relative.  

Since you're new to this, you may want to review the G-code reference:
http://reprap.org/wiki/G-code#G90:_Set_to_Absolute_Positioning

I just tried this here and I don't see a problem, so I'm wondering if you have a configuration problem somewhere.  Maybe I'll get some time to go through your configuration file later, but I recommend you review it and make sure your endstops are configured correctly, and you may want to enable software endstops too.

All of that aside, there's no good reason to ever issue a move with a negative coordinate when in absolute mode, and Marlin ought to return an error when that occurs, in my opinion.  If you're trying to position the nozzle at 0,0 at the start of your print (which is what -1000,-1000 will do), you should just change your gcode to X0, Y0.  Try that and let us know how it works.

batata004, 2017-05-10:
Thanks my friend but in my printer X0 Y0 moves the nozzle to the middle of the printer. I think X0 Y0 is indeed the middle right?

I think, from what you said, I am not using absolute positioning cause I never set G90. My beginning gcode (before printing is this):

M104 S0; 
M190 S60;
M140 S[bed0_temperature]; 
M104 S[extruder0_temperature]; 
G28;
G1 X-60 Y-60
M190 S[bed0_temperature]; liga a heated bed para a temperatura final e aguarda ate chegar nessa temperatura
M109 S[extruder0_temperature]; liga o hotend para a temperatura final e aguarda ate chegar nessa temperatura
G1 E8 F50; empurra 8 milimetros de filamento (ver aba "Ending Script")

You see? I am trying to move the nozzle a little bit to the left/front so when start printing the nozzle drags a little in the heated bed till reach the initial printing position cause this always helps me to remove any filament that is left in the tip of it.

bgort, 2017-05-10:
The gcode you posted:
```
G90
M83
M106 S0
M104 S0; desliga o hotend e nao aguarda ate chegar nessa temperatura para evitar o pingamento de filamento caso o hotend esteja aquecido por conta de alguma impressao anterior
M190 S60; liga a heated bed para uma temperatura proxima da final e aguarda ate chegar nessa temperatura
M140 S70; liga a heated bed para a temperatura final e nao aguarda ate chegar nessa temperatura
M104 S238; liga o hotend para a temperatura final e nao aguarda ate chegar nessa temperatura
G28; realiza o homing de todos os eixos
;G29; realiza automaticamente o homing de todos os eixos (caso ainda nao tenha sido feito anteriormente com o GCODE G28) e apos isso realiza o procedimento de auto nivelamento
G1 X-1000 Y-1000; move o hotend para a posicao mais a esquerda e a frente para que, quando comece a imprimir, o nozzle arraste na heated bed ate a posicao inicial a fim de soltar qualquer filamento derretido preso a ele
M84; desliga todos os motores caso contrario eles continuarao ligados enquanto a temperatura do hotend e da heated bed atingem os valores finais
M190 S70; liga a heated bed para a temperatura final e aguarda ate chegar nessa temperatura
```

The very first line is G90.

batata004, 2017-05-10:
@bgort I was not aware of it, Simplify3D is adding that automatically. That G90 is not mine, my START SCRIPT is from M104 S0 to the next lines. I think for some reason Simplify3D always adds the G90 :(

Anyway I can "hack" this?

bgort, 2017-05-10:
> Thanks my friend but in my printer X0 Y0 moves the nozzle to the middle of the printer. I think X0 Y0 is indeed the middle right?

Ah, I see.  As far as I know, most users use a corner as 0,0 instead of the center of the bed.  I suppose what you're doing should work okay, but I'd try adding a G91 before the G1 X-whatever Y-whatever, and then a G90 after it.

batata004, 2017-05-10:
WORKED PERFECLY @bgort !!! Adding G91 before everything and restoring G90 at the end solved the problem! :) Thank you all for all the kind help, this community is awesome! I will close this thread now and hope someone in the future that faces this problem can get a solution :)

closed 2017-05-10 by batata004

bgort, 2017-05-10:
No problem.  Happy to help!

JohnOCFII, 2017-05-10:
> @bgort I was not aware of it, Simplify3D is adding that automatically. That G90 is not mine, my START SCRIPT is from M104 S0 to the next lines. I think for some reason Simplify3D always adds the G90 :(
> 
> Anyway I can "hack" this?

Check the G-Code tab in the FFF configuration for Simplify3D.  There is a setting there for relative extrusion distances.  In my printer, this is checked.

batata004, 2017-05-10:
@JohnOCFII  yeap I can confirm that! :) Thank you so much, it's much easier using it!