MarlinFirmware/Marlin #6670: Another bug with incompatility

batata004, 2017-05-10:
Hi,

Using the latest marlin firmware (previous to the new one we have now) I could use this command to autotune PID of my heated bed with no problem:

M303 E-1 S70

Now when I try to send that command in the Serial it says "READ: PID Autotune failed! Bad extruder number". What changed from the previous to the new firmware that is making this command incompatible?

These are my config files.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/990235/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/990234/Configuration.txt)

bgort, 2017-05-10:
You need to #define PIDTEMPBED.

From your Configuration.h:
`//#define PIDTEMPBED`

Should be
`#define PIDTEMPBED`

batata004, 2017-05-10:
Worked! For some reason I never changed this (even in older firmware versions) but now I think I have to uncoment this! :) Thanks man.

batata004, 2017-05-10:
Worked! For some reason I never changed this (even in older firmware versions) but now I think I have to uncoment this! :) Thanks man. Sorry for bothering with this small issue.

closed 2017-05-10 by batata004