MarlinFirmware/Marlin #6673: WRITE_HEATER_BED Compile Error

Guyver1n3, 2017-05-10:
Hi

When Changing #define TEMP_SENSOR_BED 0 to #define TEMP_SENSOR_BED 1

The following compiling error occurred at Line 822 in temperature.cpp - temperature control

Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\GUYVER~1\AppData\Local\Temp\arduino_build_875315\sketch\temperature.cpp" -o "C:\Users\GUYVER~1\AppData\Local\Temp\arduino_build_875315\sketch\temperature.cpp.o"
C:\Users\GUYVER~1\AppData\Local\Temp\arduino_build_875315\sketch\temperature.cpp: In static member function 'static void Temperature::manage_heater()':

temperature.cpp:822: error: 'WRITE_HEATER_BED' was not declared in this scope

         WRITE_HEATER_BED(LOW);

                             ^

exit status 1
'WRITE_HEATER_BED' was not declared in this scope

Please Help on this error or point to Forum that this error is been worked on.
[Arduino Compile Error.txt](https://github.com/MarlinFirmware/Marlin/files/990733/Arduino.Compile.Error.txt)

.

bgort, 2017-05-10:
Check your pins_XXXX.h file and make sure that your HEATER_BED_PIN is correct.

Guyver1n3, 2017-05-11:
Hi bgort,
Thank you for your fast response. I checked out the pins_XXXX.h file and picked up from there that in the configuration.h #define MOTHERBOARD BOARD_RAMPS_14_EFF must change to #define MOTHERBOARD BOARD_RAMPS_14_EFB. You were a great help in finding the correct file to help me forward.

I'm a NOOB with the code thus greatly appreciate.
Hope we can keep this error up for other NOOBS too as there is not a lot said on google about that error.
Even had a guy almost decoding the whole config file as he said it might be something to do with Arduino 1.x.x Capability :-(, Didn't feel like the right answer to me...lol.

closed 2017-05-11 by Guyver1n3

bgort, 2017-05-11:
No problem.  Glad to help!