MarlinFirmware/Marlin #6674: Strange Z homing behavior

GrevanGER, 2017-05-10:
I am working on an old Prusa i3 printer atm. 
M5 rods, wade extruder, all in all pretty outdated stuff. But it worked just fine, gave me some 'decent' prints, considered the hardware. Yet it ran Sprinter, which is pretty limited. So I went ahead and switched to the current Marlin version.
Now here comes the fun stuff: On sprinter, all rods ran smoothly, I calibrated the steppers properly, no strange or loud noises at all whilst homing or moving the axes.
Now with Marlin, when ever I home the Z axis, the Z motor 'stutter' like a pretty old paper printer until, what seems like they have gathered enough momentum. Then both rods run smothly until they slow down right before they come to a stop.
When I move Z in a normal motion, both motors are running smooth as ever, no matter how big or small the distance it travels.

What I have tried so far:
- playing around with z jerk, accileratin, max acc., 
- double touch
- different poty settings

All without getting rid of the problem.
I don't know what to look out for anymore.

Roxy-3D, 2017-05-10:
I would turn the acceleration and max feed rates way down until the problem is figured out.  I would look and see if I can crank the power to the stepper motors up.    I don't know what is wrong...  but it sounds like your stepper motors can't keep up and do what they are being told to do.

KevMags, 2017-05-10:
Sounds like your Z homing feedrate is too high.  Pretty common.

GrevanGER, 2017-05-11:
Actually it was too low.
On a fresh Marlin the homing feedrate starts out as 4mm per Minute, which was way too low. Nothing moved, the steppers just made a sound like the were trying to crush themaelfs.
Considered that the printer is working with M5 rods, I have set the homing feedrate to the same value as the standard XY setting. 50mm per minute. 
That was yesterdays final status. Z is moving perfectly smooth when printing or telling it to travel, but during a homing process, including the lift during safe home, the Z steppers behave like a train that is starting from, or entering a station. Slow and noisy calling, than it travels as usuall without any problems just to get noisy again as it slows down. 
Double tapping is one giant cry of metal on metal.

I will copy all relevant Z settings, once I am at my workstation again.

GrevanGER, 2017-05-11:
As promised, here is everything related to the Z axes and Z homing: 

```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 3200/0.8 , 500 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 150, 25 }
#define DEFAULT_MAX_ACCELERATION      { 5000, 5000, 5000, 50000 }

#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves

#define DEFAULT_ZJERK                  0.4

#define PROBE_MANUALLY

#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
#define PROBE_DOUBLE_TOUCH

#define DISABLE_Z false

#define Z_HOMING_HEIGHT 4

#define Z_SAFE_HOMING

#define HOMING_FEEDRATE_Z  (50*60)
```

In config_adv:
```
//#define Z_DUAL_STEPPER_DRIVERS
#define Z_HOME_BUMP_MM 2 
```

FHeilmann, 2017-05-11:
Your HOMING_FEEDRATE_Z is 50*60 = 3000 mm/m 

You need to turn that WAY down. On my CoreXY which has super smooth moving hardware I use 4*60 but with M5 rods you probably have to go lower still.

GrevanGER, 2017-05-11:
As stated above, I started with 4*60, but got no movement out of Z when homing at all.
50*60 is the value that allowed the z homing to run smooth for longer traveling ways.

GrevanGER, 2017-05-11:
Okay, I found the source of my Problem...
You guys were perfectly right when you said that my HOMING_FEEDRATE_Z was way to high.
I guess I jus landed a 'lucky hit' by copying the XY Feedrate `50*60` for Z, ending up with something that somehow worked and somehow didn't, but misslead me to think it was 'about right'.
I took a look back in to the Sprinter Software and saw that the feedrate for Z homing was set to 150 there. 150 defided by 60 is 2.5; so I set the HOMING_FEEDRATE_Z to `2.5*60`. 
Problem solved, not strange starting and breaking noises anymore when Z is moved during a homing process.

Thanks alot to everyone who participated in helping me to figure this out.

closed 2017-05-11 by GrevanGER