MarlinFirmware/Marlin #6683: Z Moves After Level

HomeBrewedCoffee, 2017-05-10:
I am using the latest 1.1.X build (downloaded a day or two ago) on a modified MakerGear M2 printer.

I have everything working for the most part except for the leveling. I am using an inductive probe attached to the x carriage. The Z axis homes to min and I am using the bilinear autoleveling. 
It seems to home fine but then moves 2 mm away. Offset won't change this. After trying to debug using the M111 S255 command it seems that the printer is sending a "deploy probe" command that moves it 2mm. 

This appears to stem from the Z_CLEARANCE_BETWEEN_PROBES variable since when I set it to zero this issue stopped. So my question is why is the "deploy" using this variable when I have the Z_CLEARANCE_DEPLOY_PROBE set to 0?   

I have attached config, adv config, and the log from my homing debug.

[configuration_adv_h.txt](https://github.com/MarlinFirmware/Marlin/files/991411/configuration_adv_h.txt)
[configuration_h.txt](https://github.com/MarlinFirmware/Marlin/files/991412/configuration_h.txt)
[HOMING_LOG.txt](https://github.com/MarlinFirmware/Marlin/files/991413/HOMING_LOG.txt)

Thanks.

HBC

Bob-the-Kuhn, 2017-05-11:
The value Marlin uses for Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES is the higher of the two.

#define Z_CLEARANCE_DEPLOY_PROBE    0 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  2 // Z Clearance between probe points

HomeBrewedCoffee, 2017-05-11:
Ok, but why is it sending a deploy command for a fixed mount and moving after homing?

Bob-the-Kuhn, 2017-05-11:
The deploy command is used for all probes.  That's where endstops are turned on, the slower speed is started and the probe extended if it's servo/solenoid based.

I don't find a command to move in the X or y direction after safe homing Z.  Where does it go to?

HomeBrewedCoffee, 2017-05-11:
It moves the bed 2mm in the Z direction after homing. H
Here is a debug section just ran. 

RECEIVED: <<< set_axis_is_at_home(Z)
RECEIVED:   current_position=(61.00, 127.00, 0.00) : sync_plan_position
RECEIVED:   current_position=(61.00, 127.00, 0.00) : > AFTER set_axis_is_at_home
RECEIVED:   current_position=(61.00, 127.00, 0.00) : set_probe_deployed
RECEIVED: deploy: 0
**RECEIVED: do_probe_raise(2.00)**
RECEIVED: >>> do_blocking_move_to(61.00, 127.00, 2.00)
RECEIVED: <<< do_blocking_move_to
RECEIVED: >>> do_blocking_move_to(61.00, 127.00, 2.00)
RECEIVED: <<< do_blocking_move_to
RECEIVED: <<< homeaxis(Z)
RECEIVED: <<< Z_SAFE_HOMING
RECEIVED:   current_position=(61.00, 127.00, 2.00) : > (home_all_axis || homeZ) > final
RECEIVED:   current_position=(61.00, 127.00, 2.00) : sync_plan_position
RECEIVED:   current_position=(61.00, 127.00, 2.00) : clean_up_after_endstop_or_probe_move
RECEIVED: X:61.00 Y:127.00 Z:2.00 E:0.00 Count X:5422 Y:11288 Z:800
RECEIVED: <<< gcode_G28

So it is doing a probe raise at the end of the level. Is there a way to disable that?

Bob-the-Kuhn, 2017-05-11:
They're getting ready for the next move by raising Z to Z_CLEARANCE_BETWEEN_PROBES.

Looks like if you set both Z_CLEARANCE_BETWEEN_PROBES and Z_CLEARANCE_DEPLOY_PROBE to zero then it won't move.

Personally I think raising the nozzle after homing is a good idea.  Makes it safer for those of us that are forgetful.

closed 2017-11-01 by thinkyhead