MarlinFirmware/Marlin #6684: Random behaviour of G1 command

batata004, 2017-05-10:
Hi,

In my START GCODE SCRIPT I have a line like this:

G1 X0 Y0

It's executed right at the middle of the preheating bed. I heat it to 60 C, then I run the GCODE above and keep heating it to 100C. My question is this: sometimes the printer moves in a good speed to the coordinate X0 Y0 but sometimes it goes veeery slow, veeery freaking slow! Why does this happen?

Also, if I use this:

G1 X0 Y0 Z0 F2000

I understand it will move in the speed of 2000 right? But Z axis surelly cant keep up with this speed. What will happen? I didnt try this cause I was afraid I could damage my little cheap printer. Does any of you know if the printer will try to move Z at the same speed of 2000, and of course because it wont be able to reach that speed it will miss steps?

Tannoo, 2017-05-11:
You already have #6663 open on this.

batata004, 2017-05-11:
@Tannoo  this is completely different cause this is related to speed only, the other one is related to the movement itself. In this case that I am reporting here the behaviour (position) is as expected, in the other thread the behaviour is not reliable. Only thing that looks wrong in this thread is the speed which looks random. G1 in Marlin defaults to what speed, since G1 is a linear movement?

Tannoo, 2017-05-11:
Are you closing this one also?

batata004, 2017-05-11:
Sorry really sorry!

closed 2017-05-11 by batata004