MarlinFirmware/Marlin #6687: Preheat ABS nozzle temp has issues

TalMcMahon, 2017-05-11:
I am running RC8 so I am not sure if this is the right place to post this. When I use the prepare screen and select "Preheat ABS" The system seems to heat as normal except that the nozzle temp reading rolls from 98 degrees to 99 degrees then rolls over and starts again at 0. This will continue to loop until a runaway temp error occurs. 

When I set the temp from an external source (octoprint) all works fine.  Here is a quick video showing what I am describing.  [https://youtu.be/RJ2zSLhHeuY]

Bob-the-Kuhn, 2017-05-11:
Any chance of you moving to 1.1.0 or Marlin-bugfix-1.1.x?  Chances are excellent that the problem has been fixed.

You'll need to manually transfer your machine specific items to the new config files.

Are  you happy with your current bed leveling?  If not give the UBL system a whirl.  Definitely a better system.

TalMcMahon, 2017-05-12:
Sounds like a weekend project.  :)

closed 2017-11-01 by thinkyhead