MarlinFirmware/Marlin #6688: Realtime commands in Marlin

109JB, 2017-05-11:
Hello. First post here on Marlin, so a little background.  I am just getting into 3D printing with a clone prusa i3 type machine that is running marlin with ramps and LCD/SD. Works great.  However, I also do CNC machining on metal cutting milling machines. On my smaller milling machine I currently run Grbl. On that machine I would like to modify it to be dual purpose to do CNC machining and also 3D printing by adding a removable heated bed and extruder to it. To that end, I am searching for a firmware that can handle both tasks so I don't have to switch firmware or controllers. I have researched and found that Marlin has sufficient G-code capability for my machining purposes, but I have a question.

For machining, I currently use a PC based interface that I wrote myself. I can modify this to send/receive to/from Marlin for machine control via USB com port, but have a few questions in regard to realtime tasks.. I will reference how Grbl does it to see if there are any similar methods in Marlin since I am familiar with Grbl.

In Grbl there is a realtime command set that the firmware intercepts and acts on immediately. For example, sending a "!" in the serial stream results in Grbl going into a "feed hold" state and pausing motion.  There are other commands such as these:

"!" -  Feed hold
"?" - Get status report
ctrl-X - Soft reset the firmware

I have looked for a reference to realtime commands for Marlin, but have not found anything. Does Marlin have a similar set of realtime commands, and if so is there a list of them?

Bob-the-Kuhn, 2017-05-11:
If you enable the emergency parser then you can send the M112 command which will immediately send the controller through reset.

There's also M400 that finishes all queued moves and M410 that aborts all the planned moves.

I also run a 3D printer & CNC setup.  Here's some of the things I've run into:
- You'll find that any machine strong enough for CNC work will be a slow printer unless you've invested huge $$ in steppers & stepper controllers.   
- My Z axis resolution is 0.027mm.  That's considered rather course by most 3D printer people.
- There's enough differences between the CNC setup and the printer setup that you'll end up re-flashing the board whenever you switch modes.
- Most controllers meant for grbl don't have provisions for heaters, fans, hotbeds & thermal inputs.  
- Most controllers meant for 3D printing are not compatible with the GRBL firmware.

For the interface there are several popular choices - Repetier Host, Octoprint & Pronterface.

I'll be glad to help get Marlin running on your controller.  If you point me to your controller I'll see if we can set the Marlin pin definitions to suit your board.

109JB, 2017-05-11:
Thank you for the reply.  

For the machine in question I plan to replace the current Arduino UNO running GRBL and use an Arduino Mega with a Ramps board. I will pick off the step and direction pins from the ramps board to feed the stepper drivers I currently have for X Y Z. The extruder will use a DRV8825 on the ramps board and the hotbed will hook to the ramps as normal.

For 3d printing I would use one of the 3d printer interfaces like you mentioned, but would modify my milling interface to work with Marlin for cutting.  My milling interface has some features I really need because of the way I use the mill. I use canned cycles a lot and have probing routines and some other stuff integrated into the user interface that I don't want to lose.

The machine in question can move at 200 IPM (5000 mm/min, 85mm/s) in X and Y comfortably. I realize this may be a little slow, but the prints I have done so far have been done at 50mm/s (120ish IPM) which is quite doable on this machine.  I could probably push the X and Y up to 250 IPM with a little tweaking.

The theoretical machine resolution is 0.00013" (0.0032mm) so that shouldn't be an issue.

My hope is that I won't have to re-flash, but only time will tell. If it comes to that, I would probably have 2 Arduinos and just wire up a DB25 connector to switch between milling and 3D printing.

Thanks again.  I'll look into the emergency parser to see what I can do with my interface program.

Bob-the-Kuhn, 2017-05-11:
Looks like you have a good handle on what needs to be done.

I see there is a  [version of GRBL for the RAMPs board](https://sourceforge.net/projects/grblforramps14/).  That didn't exist 9 months ago (that I knew of).

Re-flashing isn't that big of a deal unless you're trying to cut the cord to the PC.  You'll spend more time changing the head & bed than it'll take to re-flash.

I ended up moving all my CNC & 3D printing specific items into two sections and then wrapping them in a `#if CNC_MODE .... #else ... #endif` wrapper.

There are more CNC specific commands coming to Marlin.  Don't know what the time frame is.  I can provide M3, M4, M5 code now.  

Your canned cycles - are they implemented in simple g-codes without any looping?  If yes then you could feed them directly to Marlin.  For my very limited CNC work I use Sketchup & SketchUcam.  The SkecthUcam output can be fed directly to Marlin (except for theM3, M4 & M5).

109JB, 2017-05-12:
I have a few of the canned cycles implemented in my user interface on the laptop.  The way it works is that the interface reads the line of g-code before it is sent to the Arduino. If the line contains a canned cycle that is supported it converts the line to code to send to the Arduino.  For example a peck drilling cycle is sent as a bunch of G0 and G1 Z-axis commands. My GUI actually only sends a few different commands to Grbl and it would be the same with Marlin.  The codes really used are G0 G1 G2 G3 G4 G38.2 G93 G94 M3 M4 M5 M7 M8 M9 .  Everything else can be handled in my interface.  I see that Marlin has an M42 and I think I can use that for the M3 M4 M5 M7 M8 M9 commands.  Basically have the gui intercept say an M3 command and instead of sending M3 to Marlin, send something like M42PXXS255. Then there would be no change to Marlin making updates easy.

Actually, my plan for the extruder and bed are to have cam locks to attach them, The Spindle would remain in place and the extruder would cam lock to the side.of the head. I figure if I do it right the head goes on or off in about 10 seconds each way. The bed would be similar.

Bob-the-Kuhn, 2017-05-12:
PWM output for M3 & M4?

109JB, 2017-05-12:
The way I read the Marlin M42 command, as long as you are using a PWM capable pin you can do PWM with M42 Pxx S### where ### can set PWM between 0 and 255

Bob-the-Kuhn, 2017-05-12:
The trick is finding an available hardware PWM.

Here's my notes on PWMs. 

```
 *
 *  There are 16 PWM ports assigned to 15 physical pins.
 *     Pin 13 has two ports assigned to it.
 *
 *      Timer   Digital Normally    Used by         Optional
 *      & Port   Pin    assigned    system          users
 *      TIMER3B   2     X_MAX
 *      TIMER3C   3     X_MIN
 *      TIMER0B   4     HEATER_4    #temp & milli ISR
 *      TIMER3A   5     HEATER_5                    *servo 0-11 ISR
 *      TIMER4A   6     HEATER_6                    *servo 12-23 ISR
 *      TIMER4B   7     LCD
 *      TIMER4C   8     HOTBED
 *      TIMER2B   9     HEATER_1
 *      TIMER2A   10    HEATER_0
 *      TIMER1A   11    HEATER_7    *stepper ISR     *E axis waveform generator
 *      TIMER1B   12    PS_ON_PIN   *stepper ISR
 *      TIMER0A   13    LED         LED PWM & #step adv ISR
 *      TIMER1C   13                *stepper ISR
 *      TIMER5C   44    LCD                         stepper motor current XY PWM
 *      TIMER5B   45    LCD                         stepper motor current Z PWM
 *      TIMER5A   46    Z_STEP                      stepper motor current E PWM or *servo 24-35 ISR
 *
 *   * - These hardware PWMs are not available.  The pin can still be used as a general purpose
 *       digital I/O.
 *
 *   # - still can be used a hardware PWM even though it's also used by an ISR
 *
 *  In addition to the above, fans can be assigned to PWM pins.  If you pick a pin that's
 *  already assigned to a fan then you'll need to delete the fan or change its pin assignment.
 *  This needs to be done even if FAN_FAST_PWM is disabled.
 *
 *  NOTE: Most pins that are hardwired to a heater or a fan usually are driven by a power driver
 *        MOSFET which has on it's output a pull up through an LED to +12V/+24V. This will
 *        probably damage your spindle controller unless you add some circuitry.  If there isn't
 *        a +12V/+24V pull up you'll need an external 1k-10k pull up resistor to the pin.
 ```

109JB, 2017-05-12:
Sorry, I think I misunderstood your last post.  I actually don't need PWM for spindle for my purposes. I actually only need a pin to control a relay as my spindle on that machine is only on-off.  What i was I guess trying to say is that my interpretation is that if someone needs PWM that the M42 command can do it.

Bob-the-Kuhn, 2017-05-12:
Trying to figute out if I can be of any help.  Looks like you've got it figured out.

Happy printing.

closed 2018-02-06 by thinkyhead