MarlinFirmware/Marlin #6690: WATCH_BED_TEMP_INCREASE  with 1 is dangerous

trivalik, 2017-05-11:
Yesterday I had strange effect that my bed throws a heater error. So I changed it from:

```
#define WATCH_BED_TEMP_PERIOD 120                // Seconds
#define WATCH_BED_TEMP_INCREASE 1               // Degrees Celsius
```
to:
```
#define WATCH_BED_TEMP_PERIOD 120                // Seconds
#define WATCH_BED_TEMP_INCREASE 1               // Degrees Celsius
```

on heat up to 90 °C it was jumping between 85 and 86°C and never reached the 90°C. So the observation function is not useless.

I would suggest to forbid values which are smaller than 2.

The cause of bed failure was a lose power supply to board, that board could not reach the 90°C.

trivalik, 2017-05-11:
[image]

As you can  see there is no error, also if 90°C is not reached.

mperdue, 2017-05-12:
Well, the comments do say you shouldn't set it below 2 but enforcing a minimum is probably a good idea.

trivalik, 2017-05-16:
Today, I had also with this settings a temperature failure. But I can only guess that the bed was the cause, printed without PI. Anyway I believe it is somehow different when I was tuning the bed temp. The gcode file was setting 80. I don't want to wait, and switch it to 40. Then nozzle warm up and I switch direct back to 90 degree. I believe that here the time needs longer. Is here another watch time used than in the gcode command for heating up?

mperdue, 2017-05-16:
This looks like the issue I was having. I needed to get and insulator to put under my heated bed to allow the 12V heater to git above 90c. FWIW, I used this... 

https://www.amazon.com/gp/product/B00029KCW0/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

I put the shiny side up to reflect the heat back toward the bed. After doing this I was able to get beyond 100c and print ABS with no problems.

trivalik, 2017-05-17:
I get also 90 degree. It was occure just in my described case.

closed 2018-02-06 by thinkyhead