MarlinFirmware/Marlin #6692: G29 P2 B Issue

bgort, 2017-05-11:
Latest bugfix is raising by 2mm (my SIZE_OF_LITTLE_RAISE) between measurement points instead of the size specified by the H code, which makes manually filling in the mesh take a lot longer.

Relevent code section:
```
      if (HYPOT(dx, dy) < BIG_RAISE_NOT_NEEDED)
        do_blocking_move_to_z(current_position[Z_AXIS] + SIZE_OF_LITTLE_RAISE);
      else
        do_blocking_move_to_z(z_clearance);
```

Is there ever a case where we actually need a raise other than what's specified by the H code while manually filling in mesh points?  I'm not seeing a need for anything other than the H value if one is specified, but it's possible I'm missing something.

bgort, 2017-05-11:
Ah, I see.  We're updating `height` (which becomes z_clearance) in `mesaure_business_card_thickness(height)` but we're updating it to Z_CLEARANCE_BETWEEN_PROBES above 'zero' / bed rather than the thickness of the business card + some small margin.

```
    const float z2 = use_encoder_wheel_to_measure_point();
    do_blocking_move_to_z(current_position[Z_AXIS] + Z_CLEARANCE_BETWEEN_PROBES);

    if (g29_verbose_level > 1) {
      SERIAL_PROTOCOLPGM("Business Card is: ");
      SERIAL_PROTOCOL_F(abs(z1 - z2), 6);
      SERIAL_PROTOCOLLNPGM("mm thick.");
    }
    in_height = current_position[Z_AXIS]; // do manual probing at lower height
```

Perhaps should be?:

```
    const float z2 = use_encoder_wheel_to_measure_point();
    do_blocking_move_to_z(current_position[Z_AXIS] + Z_CLEARANCE_BETWEEN_PROBES);

    if (g29_verbose_level > 1) {
      SERIAL_PROTOCOLPGM("Business Card is: ");
      SERIAL_PROTOCOL_F(abs(z1 - z2), 6);
      SERIAL_PROTOCOLLNPGM("mm thick.");
    }
    in_height = abs(z1-z2)+0.3; // do manual probing at height of business card + 300um margin
```

This okay with everyone?  

Or is it better to not change the height if H is specified?  Doing it that way makes things more 'you get what you asked for', I think.

Roxy-3D, 2017-05-11:
The BIG_RAISE and LITTLE_RAISE was paranoid thinking.   I was worried that if the nozzle was going very far to get to the next point, there might be unknown bed flaws in the way.   So the BIG_RAISE would be safer.  But UBL was developed on a machine with a very very slow Z-Axis.  So, it was nicer to just do a LITTLE_RAISE when possible.     

I'm not sure we need to keep both around any more.   But we probably do need to raise some 'safe' amount as the nozzle runs around the bed looking for the next location it is going to deal with.

bgort, 2017-05-11:
Makes sense.  Why not raise to z_clearance between_probes as we move, then lower to the H specified, or the business card height + some small margin if H isn't specified?

Roxy-3D, 2017-05-11:
I'm good with that!    Also remember...  When UBL development started...   Z_CLEARANCE_BETWEEN_PROBES did not exist.   But in hindsight...  That is probably the 'right' answer.

bgort, 2017-05-11:
Ah, that makes sense!  

Okay, I'll make these changes now and do a PR.

Bob-the-Kuhn, 2017-05-11:
I have about a 1.5mm tilt on my bed.  I'll need a higher raise.

If this is the first time probing a bed you won't know what a "safe" raise is.

It's a tradeoff between worn out encoders & sore thumbs for most people vs. hitting the bed for a few.

bgort, 2017-05-11:
Hmm.  If we just let H override whatever we 'learn' with B (plus a margin), would that work for you?

bgort, 2017-05-11:
Or shall we just leave the B code the way it is, and allow H to override it for those of us who want a small raise?

Bob-the-Kuhn, 2017-05-11:
Lots of reasonable ways to implement it.

I'll be happy with whatever you & Roxy are happy with.

bgort, 2017-05-11:
Okay, so I'm going to leave the current B the way it is, but just allow H to override whatever B comes up with (the bed/zero + Z_CLEARANCE_BETWEEN_PROBES, currently).  

That keeps things pretty 'safe', generally, but allows users who want to specify a height to do so (at their own risk, of course) ...

closed 2017-11-01 by thinkyhead