MarlinFirmware/Marlin #6696: Printrbot RevF5 bluetoothserial not defined

dragoonmc, 2017-05-12:
OK guys. I have been trying to get a firmware to compile using platformio for my Printrbot simplemetal with an e3dv6 hotend with a non-standard thermistor.

Doing this I am getting an error about Bluetoothserial not being defined yet I think. I am out of ideas as to how to get round this now. I am using a copy of 1.1.x from today.

Any help is appreciated.
`C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\serial.h:31:22: error: 'bluetoothSerial' was not declared in this scope
#define MYSERIAL bluetoothSerial
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\serial.h:43:25: note: in expansion of macro 'MYSERIAL'
#define SERIAL_CHAR(x) (MYSERIAL.write(x))
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\serial.h:44:20: note: in expansion of macro 'SERIAL_CHAR'
#define SERIAL_EOL SERIAL_CHAR('\n')
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\serial.h:53:83: note: in expansion of macro 'SERIAL_EOL'
#define SERIAL_PROTOCOLLNPAIR(name, value)  do{ SERIAL_PROTOCOLPAIR(name, value); SERIAL_EOL; }while(0)
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\serial.h:61:40: note: in expansion of macro 'SERIAL_PROTOCOLLNPAIR'
#define SERIAL_ECHOLNPAIR(name, value) SERIAL_PROTOCOLLNPAIR(name, value)
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\Marlin_main.cpp:11900:5: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
SERIAL_ECHOLNPAIR(MSG_KILL_INACTIVE_TIME, current_command);
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\Marlin_main.cpp: In function 'void setup()':
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\serial.h:31:22: error: 'bluetoothSerial' was not declared in this scope
#define MYSERIAL bluetoothSerial
^
C:\Users\cline\Desktop\Marlin-1.1.x\Marlin\Marlin_main.cpp:12180:3: note: in expansion of macro 'MYSERIAL'
MYSERIAL.begin(BAUDRATE);
^
*** [.pioenvs\printrboard\src\Marlin_main.o] Error 1`

fiveangle, 2017-05-13:
See last comment on #6550.  Try adding #define MarlinConfig.h into serial.h and see what happens.

dragoonmc, 2017-05-13:
Ok I swear I tried that last night before posting this issue but it actually compiled when I did it just now with that addition

fiveangle, 2017-05-13:
I think the solution to #6550 (and this one) is likely to simply include MarlinConfig.h into serial.h, but for the life of me I can't figure out why.  Marlin.h (where USBCON test first occurs to set these parameters) includes only these:

```
 #include "types.h"
  #include "fastio.h"
  #include "utility.h"
```

Which don't help including into serial.h after the unmarriage from Marlin.h.  Only including both macros.h and Arduino.h resolve (both part of MarlinConfig.h).

But I'm not a seasoned programmer, nor do I play one on television

dragoonmc, 2017-05-13:
Now I can talk to the printer but it refuses to anything at all.
Endstops report incorrectly and it seems that the temp probes aren't happy either as they report as off

Bob-the-Kuhn, 2017-05-13:
Check your temp probe settings.  If Marlin thinks the temperature is out of bounds then it won't do any movement commands.

You could also set your TEMP_SENSOR_xx to 998 or 999 just to see if that's the problem.

dragoonmc, 2017-05-13:
I did try 999 and it registered the temp. I now know that my hot end fan auto on is the correct pin. But something else is clearly wrong. I will have to try to change all the pin outs to the teensy call out tomorrow and see if that is the fix instead of the marlin pin call outs

Bob-the-Kuhn, 2017-05-14:
The AT90USB boards are an unusual combination of two pin mappings
 so changing the pin numbers can be challenging.

If you can't get it running you could consider the following.

Attached is something I proposed months ago but never made it into Marlin.  It has the current pin mappings, Teensy only mappings & Marlin "standard" mappings.  
[pins_PRINTRBOARD.zip](https://github.com/MarlinFirmware/Marlin/files/999013/pins_PRINTRBOARD.zip)

I need to re-word that file.  You can make a useable image with the current system.  It's just that 1) it's not guaranteed to do so in the future & 2) changing pin numbers can be challenging.

If you use the  [JSON](https://rawgit.com/Bob-the-Kuhn/Marlin_AT90USB/master/package_MARLIN_AT90USB_index.json) then you can use the M43 commands to verify that a function is mapped to the desired pin or find the pin number of an unknown pin..

dragoonmc, 2017-05-14:
It seems that once I set platformIO to use board definition as 811 instead of 84 it all started to work as it would allow me to disable to teensy pin assignment definition in fastio

So looks like platformIO needs to have a new definition added in marlin to use MB 811 and have printrboard revF instead of just normal printrboard

Thank you all for your help. I am happy to be able to make my own firmware now

whosawhatsis, 2017-05-16:
Looks like MYSERIAL still isn't defined after adding those includes.

dragoonmc, 2017-05-16:
@whosawhatsis I'm not sure what you are you are saying. It is pretty clear something is not defining properly but no one in this thread has figured out why it not working correctly but I am fine with this work around as it does compile

closed 2017-05-16 by dragoonmc

Bob-the-Kuhn, 2017-05-16:
I put some code up on PR #6737 that is intended to make Marlin compatible with the Teensyduino type IDE extensions.   You can try that to see if the problems go away.

The FAST_FAN option is still apparently not working.  I'll have to investigate & fix.

fiveangle, 2017-05-17:
I could not get PIO to work at all but that could be a Deviot plug-in issue with Sublime Text 3, but using Arduino 1.8.2 with Teensyduino extension 1.36 (both latest release versions available), with`#include "MarlinConfig.h"` added to serial.h, bugfix-1.1.x (pulled as of about an hour ago) compiles and works on otherwise stock PBSM 1403 RevF with factory heated bed.

Config:
https://github.com/fiveangle/Marlin/blob/printrbot/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/printrbot/Marlin/Configuration_adv.h

fiveangle, 2017-05-17:
FYI... leaving both `//#define FAST_PWM_FAN` and `//#define FAN_SOFT_PWM` disabled triggers use of Teensy pin mapping to use pin 16 instead of 22.  I'm not sure why this check is in here because it looks as if PIN 22 on Teensyduino is ground, but perhaps I do not understand the pin numbering:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/pins_PRINTRBOARD_REVF.h#L93-L99

This works.  Teensy HW PWM is in audible range so creates quite the racket but since I'm using the anaemic stock fan, I use S255 only so this works for me.

Bob-the-Kuhn, 2017-05-18:
@thinkyhead has merged the serial.h change into rcBugfix1.1.x.  

I've put corrected code back on the PR.  That should make FAN0 operate the same as it always has.

FYI - the pins_Printrboard files are wrong when it comes to FAN0.  Right now if FAST_FAN_PWM is enabled the pin that is selected is NOT a hardware PWM.  If you disable the FAST_FAN_PWM option and the FAN_SOFT_PWM options **THEN** FAN0 is on a hardware PWM.

Please see if the new code works on your system.